espressif/esp-idf #12544: NVS encryption with factory NVS data

MacWyznawca, 2023-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Does enabling NVS encryption (at the same time as enabling encryption of all flash memory) allow you to encrypt data already stored in NVS (using standard functions)?

If so, is it necessary to take a different approach to the data saved before the encryption was activated.

I mean the data saved before the first start of the software (factory data).

### What is the actual behavior?

In my case, the data stored in NVS and factory_nvs after encryption could not be read. Perhaps the problem was due to not checking nvs_keys to see if it is cleared.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

rrtandler, 2023-11-08:
Hi @MacWyznawca,
Could you share more details how was the NVS partition encrypted in your case ? Have you provided the extra partition for holding NVS encryption keys ? Have you provided your own NVS encryption keys or have you left chip to generate keys during flash encryption ? You can use [this part](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_encryption.html) of documentation for reference.

mahavirj, 2023-11-08:
Related https://github.com/espressif/esp-idf/issues/12350#issuecomment-1746534607

MacWyznawca, 2023-11-08:
Hi @mahavirj  @rrtandler 
it's not the previously encrypted NVS data, but the unencrypted data stored in NVS and factory_nvs partitions that are correctly handled before flash and NVS encryption is enabled. 
In the factory_nvs partition, are the data used by ESP-HomeKit-SDK (for example to verify HomeKit code)

mahavirj, 2023-11-09:
@MacWyznawca 

If you would like to store the factory data in the NVS then you must create a pre-encrypted NVS partition and flash it on the device during manufacturing stage. Once the NVS encryption feature is enabled, it does not encrypt the pre-flashed contents, it assumes that contents are already encrypted.

Please refer to tools and docs pointers supplied in my earlier comment for more details.

closed 2023-11-09 by MacWyznawca