espressif/esp-idf #12547: Virtual Ethernet over USB

diplfranzhoepfinger, 2023-11-08:
### Is your feature request related to a problem?

Hi, 

ESP32 S3 has no Ethernet integrated, only SPI Ethernet, 
but has USB. 

is is possible to make a Ethernet on USB, so if you plug into PC it has a IP Connection ?

igrr, 2023-11-08:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_ncm

diplfranzhoepfinger, 2023-11-08:
cooooooooooooooool. want to test it !

diplfranzhoepfinger, 2024-01-17:
> https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_ncm

had a look into this. 

My next Question is: 

if i have a internal File Server like this one: 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving

and i want to access this one over this Ethernet over USB, 
how to do it ? 

i do not want to brige Wifi to USB so that the host could access the Internet, but let the Host get a IP over DHCP and access the Internal Webserver. 

Next Question would be if i want this Ethernet over USB and a Wifi AP on the ESP32. 

so i can access the File Server either over USB or over Wifi AP

diplfranzhoepfinger, 2024-01-17:
found this: 
https://github.com/kaspernyhus/esp_tinyusb_test/blob/master/main/usb_ncm.c

diplfranzhoepfinger, 2024-01-17:
found this:
https://github.com/buglloc/BoundBoxESP/blob/main/components/hardware/src/net_usb.cc

diplfranzhoepfinger, 2024-01-23:
> found this: https://github.com/kaspernyhus/esp_tinyusb_test/blob/master/main/usb_ncm.c

this helped. 
working. 
even with DHCP

diplfranzhoepfinger, 2024-01-25:
so, tested on a Windows today, and get a unknown Device in Device Manager, 

but was running fine on Linux. 

where to get the Driver ? 

Thanks !

diplfranzhoepfinger, 2024-01-27:
@igrr any Help getting this als work under Windows ? 
or any way i could ping someone who has such driver ?

diplfranzhoepfinger, 2024-01-27:
working:

[https://learn.microsoft.com/en-us/answers/questions/52386/usb-cdc-ncm-devices-not-assigned-to-usbncm-driver](https://learn.microsoft.com/en-us/answers/questions/52386/usb-cdc-ncm-devices-not-assigned-to-usbncm-driver)

https://learn-attachment.microsoft.com/api/attachments/91824-win10-native-ncm.pdf?platform=QnA

closed 2024-01-27 by diplfranzhoepfinger