espressif/esp-idf #12548: btsnd_hcic_ble_sync_sem_init compile error in esp-idf v5.0 release version

lida2003, 2023-11-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-285-g7493b907ef

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

like esp32 target, can build and run.

### What is the actual behavior?

strange when switch from esp32 target to esp32c3target.

### Steps to reproduce.

1. build esp32 target OK
2. build esp32c3 target failed

### Build or installation Logs.

```plain
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c: In function 'hci_layer_init_env':
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c:189:5: error: implicit declaration of function 'btsnd_hcic_ble_sync_sem_init'; did you mean 'btsnd_hcic_ble_set_phy'? [-Werror=implicit-function-declaration]
  189 |     btsnd_hcic_ble_sync_sem_init();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     btsnd_hcic_ble_set_phy
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c: In function 'hci_layer_deinit_env':
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c:215:5: error: implicit declaration of function 'btsnd_hcic_ble_sync_sem_deinit' [-Werror=implicit-function-declaration]
  215 |     btsnd_hcic_ble_sync_sem_deinit();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c: In function 'filter_incoming_event':
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c:452:36: error: implicit declaration of function 'btsnd_hcic_ble_get_sync_info'; did you mean 'btsnd_hcic_ble_ext_scan_enable'? [-Werror=implicit-function-declaration]
  452 |             BlE_SYNC *sync_info =  btsnd_hcic_ble_get_sync_info();
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                    btsnd_hcic_ble_ext_scan_enable
/home/daniel/SnapDragonfly/esp-idf-v5.0/components/bt/host/bluedroid/hci/hci_layer.c:452:36: warning: initialization of 'BlE_SYNC *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
cc1: some warnings being treated as errors
[878/1163] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/main/bte_main.c.objninja: build stopped: subcommand failed.
HINT: Please check that the function name is correct. Also it is possible that you've forgot to import btsnd_hcic_ble_sync_sem_init library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /home/daniel/SnapDragonfly/SnapAirUnit/build/log/idf_py_stderr_output_22338 and /home/daniel/SnapDragonfly/SnapAirUnit/build/log/idf_py_stdout_output_22338
```

zhp0406, 2023-11-09:
@lida2003 
Hi,how can I reproduce the issue?

lida2003, 2023-11-09:
> @lida2003 Hi,how can I reproduce the issue?

Follow this guide(https://github.com/SnapDragonfly/SnapAirUnit/blob/main/docs/Build_and_Flash.md)  with latest git version.

zhp0406, 2023-11-09:
@lida2003 
Is the Bluetooth Low Energy configuration option turned on?
Location: (Top) → Component config → Bluetooth → Bluedroid Option

zhp0406, 2023-11-09:
I reproduced the issue when I disable "Bluetooth Low Energy configuration" in menuconfig:
LATFORM -D_POSIX_READER_WRITER_LOCKS -Wno-implicit-fallthrough -Wno-unused-const-variable -Wno-format -MD -MT esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/hci/hci_layer.c.obj -MF esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/hci/hci_layer.c.obj.d -o esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/hci/hci_layer.c.obj   -c /home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c: In function 'hci_layer_init_env':
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:189:5: error: implicit declaration of function 'btsnd_hcic_ble_sync_sem_init'; did you mean 'btsnd_hcic_ble_set_phy'? [-Werror=implicit-function-declaration]
  189 |     btsnd_hcic_ble_sync_sem_init();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     btsnd_hcic_ble_set_phy
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c: In function 'hci_layer_deinit_env':
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:215:5: error: implicit declaration of function 'btsnd_hcic_ble_sync_sem_deinit' [-Werror=implicit-function-declaration]
  215 |     btsnd_hcic_ble_sync_sem_deinit();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c: In function 'filter_incoming_event':
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:452:36: error: implicit declaration of function 'btsnd_hcic_ble_get_sync_info'; did you mean 'btsnd_hcic_ble_ext_scan_enable'? [-Werror=implicit-function-declaration]
  452 |             BlE_SYNC *sync_info =  btsnd_hcic_ble_get_sync_info();
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                    btsnd_hcic_ble_ext_scan_enable
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:452:36: warning: initialization of 'BlE_SYNC *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
cc1: some warnings being treated as errors
[897/1162] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/stack/avrc/avrc_sdp.c.objninja: build stopped: subcommand failed.

--- 
but when I enable it, the issue gone:
used-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0.2-285-g7493b907ef\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -Wno-format -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/btspp.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/btspp.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/btspp.c.obj   -c ../main/btspp.c
In file included from ../main/handle.h:4,
                 from ../main/btspp.c:34:
../main/blink.h:48:5: error: unknown type name 'led_strip_handle_t'
   48 |     led_strip_handle_t strip;
      |     ^~~~~~~~~~~~~~~~~~
../main/btspp.c: In function 'task_bt_start_spp':
../main/btspp.c:413:5: warning: 'esp_spp_init' is deprecated: Please use esp_spp_enhanced_init [-Wdeprecated-declarations]
  413 |     if ((ret = esp_spp_init(g_esp_spp_mode)) != ESP_OK) {
      |     ^~
In file included from ../main/btspp.c:21:
/home/zhanghaipeng/esp_v5/esp-idf/components/bt/host/bluedroid/api/include/api/esp_spp_api.h:261:11: note: declared here
  261 | esp_err_t esp_spp_init(esp_spp_mode_t mode) __attribute__((deprecated("Please use esp_spp_enhanced_init")));
      |           ^~~~~~~~~~~~
[1105/1138] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.objninja: build stopped: subcommand failed.

lida2003, 2023-11-09:
> Is the Bluetooth Low Energy configuration option turned on?
> Location: (Top) → Component config → Bluetooth → Bluedroid Option

Yes, but when enable "Bluetooth Low Energy configuration", managed component led_strip can't be used.

zhp0406, 2023-11-14:
@lida2003 
You need to declare the dependency on the led_strip component.

for example,incomponents/command/cmd_udp/CMakeLists.txt

```
idf_component_register(SRCS "cmd_udp.c"
                    INCLUDE_DIRS . ../../../ ../../../main ../../button/include
                    PRIV_REQUIRES driver esp_timer esp_event led_strip)

target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
```

and add 

```
#include "led_strip.h"
```

in main/blink.h

lida2003, 2023-11-14:
@zhp0406 

We have declared the dependency on the led_strip component.

When enable "Bluetooth Low Energy configuration", then it fails.

zhp0406, 2023-11-14:
@lida2003 
I understand. Please test the following fix：
[0001-fix-bt-bluedroid-Fix-bluedroid-menuconfig-error.patch](https://github.com/espressif/esp-idf/files/13348353/0001-fix-bt-bluedroid-Fix-bluedroid-menuconfig-error.patch)

zhp0406, 2023-11-14:
@lida2003 
It should be noted that the ESP32 only supports Classic Bluetooth and does not support BLE. If you require BLE features, please use ESP32c2, ESP32c6, ESP32c3, or ESP32s3.

zhp0406, 2023-11-14:
@lida2003 
After you have applied my patch, if there are still failures, please provide me with the latest failure log so that I can analyze the reasons.thanks.

lida2003, 2023-11-14:
> If you require BLE features, please use ESP32c2, ESP32c6, ESP32c3, or ESP32s3.

No, we don't need BLE.

zhp0406, 2023-11-14:
> > If you require BLE features, please use ESP32c2, ESP32c6, ESP32c3, or ESP32s3.
> 
> No, we don't need BLE.

If you only use Classic Bluetooth, then only ESP32 is supported.

lida2003, 2023-11-14:
> After you have applied my patch, if there are still failures, please provide me with the latest failure log so that I can analyze the reasons.thanks.

```
daniel@palfocus:~/SnapDragonfly/esp-idf-v5.0$ git apply 0001-fix-bt-bluedroid-Fix-bluedroid-menuconfig-error.patch
error: patch failed: components/bt/host/bluedroid/Kconfig.in:1133
error: components/bt/host/bluedroid/Kconfig.in: patch does not apply
```

```
daniel@palfocus:~/SnapDragonfly/esp-idf-v5.0$ git remote -v
origin  git@github.com:espressif/esp-idf.git (fetch)
origin  git@github.com:espressif/esp-idf.git (push)
daniel@palfocus:~/SnapDragonfly/esp-idf-v5.0$ git log -n 1
commit 7493b907efd081332c1f1ef3167181727fbef922 (HEAD -> release/v5.0, origin/release/v5.0)
Merge: 96c4ec8df2 9156bbb55c
Author: Roland Dobai <roland@espressif.com>
Date:   Fri Jun 9 00:52:47 2023 +0800

    Merge branch 'feature/extend_proj_desc_json_v5.0' into 'release/v5.0'

    extend information in project_description.json (v5.0)

    See merge request espressif/esp-idf!24126

```

lida2003, 2023-11-14:
> If you use Classic Bluetooth, then only ESP32 is supported.

We need this code to work on ESP32C3 :)

zhp0406, 2023-11-14:
@lida2003 
If you are using ESP32c3, do you still need to use Classic Bluetooth? SPP is a classic feature. I will provide you with the latest patch based on your version shortly.

lida2003, 2023-11-14:
> SPP is a classic feature. I will provide you with the latest patch based on your version shortly.

Yes, we need spp and wifi features.

zhp0406, 2023-11-14:
I have created the following patch based on the version you provided (commit hash: 6e8eb2a059ae11bed0edcd8fa7d9c62d142cc1d9). This patch addresses some issues in the bluedroid menuconfig.
This issue may cause the BLE code to be invoked even after BLE is turned off, resulting in a compilation failure.
[0001-fix-bt-bluedroid-Fix-bluedroid-menuconfig.patch](https://github.com/espressif/esp-idf/files/13348590/0001-fix-bt-bluedroid-Fix-bluedroid-menuconfig.patch)

zhp0406, 2023-11-14:
> Yes, we need spp and wifi features.

If you want to use Classic Bluetooth and Wi-Fi, only the ESP32 chip supports it; currently, other chips do not support this feature.

lida2003, 2023-11-14:
> If you want to use Classic Bluetooth and Wi-Fi, only the ESP32 chip supports it; currently, other chips do not support this feature.

ESP32-C3 doesn't support Classic Bluetooth SPP ?

zhp0406, 2023-11-14:
> > If you want to use Classic Bluetooth and Wi-Fi, only the ESP32 chip supports it; currently, other chips do not support this feature.
> 
> ESP32-C3 doesn't support Classic Bluetooth SPP ?

yes,ESP32-C3 doesn't support Classic Bluetooth SPP.ESP32-C3 only support BLE.

ESP32 support BLE and Classic Bluetooth.

lida2003, 2023-11-14:
What I have missed here. The spec says it support Bluetooth 5.0, no spp?

Is there any SPP over BLE features, like https://docs.silabs.com/bluetooth/2.13/code-examples/applications/spp-serial-port-profile-over-ble ?

[image]

zhp0406, 2023-11-14:
@lida2003 
You may have some misunderstandings about Bluetooth LE: Bluetooth 5 in the document. This explanation is for BLE.

SPP is originally a profile for classic Bluetooth, and there is no corresponding profile in BLE.

However, BLE simulates SPP and implements transparent transmission. We have two examples based on BLE for transparent transmission.I'm not sure if I can meet your requirements in all cases.

BLE SPP example：
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_server
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_client

lida2003, 2023-11-14:
I have found https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/api-reference/bluetooth/esp_spp.html, which states esp32c3 supports classic bt spp profile.

The above ble spp demo (server/client) is NOT like classic spp profile, where pc already has the driver. And pc doesn't have ble spp driver.

zhp0406, 2023-11-14:
@lida2003 
Yes, BLE's SPP belongs to a custom profile, not a profile that is recognized as a standard by the Bluetooth Special Interest Group (SIG).

zhp0406, 2023-11-14:
@lida2003 
If you insist on using the ESP32-C3 chip, you will need to give up using classic Bluetooth and explore other methods to implement your functionality.

lida2003, 2023-11-14:
How to explain the description below:  esp-idf v4.3 esp32c3 bluetooth/esp_spp

https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/api-reference/bluetooth/esp_spp.html

zhp0406, 2023-11-14:
@lida2003 
I sincerely apologize for the confusion. There is a misguidance in the document. This is an error in the document, and ESP32-C3 does not support Classic Bluetooth. The document will be updated accordingly.

lida2003, 2023-11-14:
OK, if we implement a ble spp. Is there any application avaliable out there on PC(linux, windows)?

zhp0406, 2023-11-14:
@lida2003 
Currently, I haven't come across any relevant applications. maybe you need to make it yourself.

lida2003, 2023-11-14:
OK

closed 2023-11-14 by lida2003