espressif/esp-idf #12549: v5.2 select NVS_ENCRYPTION in config SECURE_FLASH_ENC_ENABLED breaks existing application

AxelLin, 2023-11-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3903-g66992aca7a

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

Application that was working fine when build with older esp-idf version should also work in esp-idf-v5.2.

### What is the actual behavior?

I have application built with SECURE_FLASH_ENC_ENABLED=y but NVS_ENCRYPTION=n.
It was working when compile with esp-idf-v4.x and esp-idf-v5.1.
However, after switching to esp-idf-v5.2, the application hit panic'ed (LoadProhibited) during boot.
I think the issue is due to select NVS_ENCRYPTION in config SECURE_FLASH_ENC_ENABLED.
For backward compatible, you should not select NVS_ENCRYPTION in config SECURE_FLASH_ENC_ENABLED.
You should allow developers to decide enable NVS_ENCRYPTION or not.

### Steps to reproduce.

Just build application with SECURE_FLASH_ENC_ENABLED=y but NVS_ENCRYPTION=n using older esp-idf versions.
Test again with esp-idf-5.2, it cannot correctly read the existing settings in older firmware (which was built with older esp-idf version)

AxelLin, 2023-11-09:
@laukik-hase 
I think c1bed366ba733426825e98f36f090fbed142bf57 causes this regression.

laukik-hase, 2023-11-17:
Hello, @AxelLin!

Thank you for the issue report. From the ESP-IDF `v5.2` release, we do not recommend turning off NVS encryption, as the default NVS partition stores the WiFi and other credentials.

Could you try out this [patch](https://github.com/espressif/esp-idf/files/13390600/nvs_encr_disable.zip) and tell me if it works for you? You'll need to turn off NVS encryption by turning on the newly added config option `CONFIG_NVS_ENCRYPTION_FORCEFUL_DISABLE`.

We'll add a note in the migration guide for the same if this is acceptable.

CC: @mahavirj

AxelLin, 2023-11-17:
@laukik-hase 
No, this patch does not work well.
I still cannot find the "Enable NVS encryption" config option even when NVS_ENCRYPTION_FORCEFUL_DISABLE=y.
And honestly, this NVS_ENCRYPTION_FORCEFUL_DISABLE looks very confusing.

I think just remove "select NVS_ENCRYPTION" from config SECURE_FLASH_ENC_ENABLED is enough.
The config NVS_ENCRYPTION is default y if SECURE_FLASH_ENC_ENABLED.
(It already implies to enable NVS encryption, You can just add note in SECURE_FLASH_ENC_ENABLED help text if necessary.) This would be much simpler.

AxelLin, 2023-11-24:
@laukik-hase @mahavirj 
Do you agree my comment in https://github.com/espressif/esp-idf/issues/12549#issuecomment-1816212292 ?
Hope to merge the fix soon.

AxelLin, 2023-12-05:
@laukik-hase @mahavirj Just remind I'm still waiting for this regression fix.

laukik-hase, 2023-12-05:
Hello, @AxelLin!

The fix is in the internal review process, you'll get notified once it syncs to GitHub.

Sorry for the delay.

AxelLin, 2023-12-28:
@laukik-hase Just remind this issue was reported against v5.2 branch, but v5.2 branch still does not include this fix.

AxelLin, 2024-01-27:
@laukik-hase 
After waiting for yet another month, the v5.2 still does not include this fix. What's wrong?

laukik-hase, 2024-01-29:
Hello, @AxelLin!

The change has been backported to the `release/v5.2` branch internally, but it has not synced to GitHub for a long time now - I will check with our CI team. The fix shall appear in the `v5.2-rc` release. Sorry for the inconvenience.