espressif/esp-idf #12550: [E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for (There's a net in the back that can't be recovered)

hzhh110, 2023-11-09:
### Is your feature request related to a problem?

The problem can be easily repeated when trying wifi(password error) and network cable connection, as long as the problem occurs, it is basically impossible to recover

#define ETH_TYPE ETH_PHY_LAN8720
ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);

hzhh110, 2023-11-09:
The network cable of the Internet connection router, after unplugging, many routers will also appear after this cannot be restored

kostaond, 2023-11-10:
@hzhh110 hi, thanks for reporting potential issue. However, you didn't specify your development environment and versions. It looks like Arduino so it would probably be better to report at [Arduino ESP32 repo](https://github.com/espressif/arduino-esp32).

Please also consider to rephrase the steps how to reproduce the issue. I'm note sure if I fully understand it. It could be just simple bullets like:
* Connect Ethernet cable
* Disconnect Ethernet cable
* Repeat X times
* Observe error "XY"

me-no-dev, 2023-11-10:
@kostaond feel free to close this issue. The user has opened a few on our end too. His problem is that DNS servers are global for client interfaces.

closed 2023-11-13 by kostaond