espressif/esp-idf #12552: hid example gets stuck connecting Zebra Symbol Code Scanner DS6708

TRYOKETHEPEN, 2023-11-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.1.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.1

### Power Supply used.

External 5V

### What is the expected behavior?

With the hid example, when I plug in the Code Scanner into the USB-OTG jack,  the ESP-IDF monitor should print as follows:
_2023-11-09 15:53:34 I (261586) example: HID Device, protocol 'KEYBOARD' CONNECTED 
2023-11-09 15:53:42 Keyboard_

and when I scan a code, the ESP-IDF monitor should print the code such as:
_2023-11-09 15:53:42 BJS-2232475009_

and when I plug out the Code Scanner, the ESP-IDF monitor should print:
_2023-11-09 15:56:30 E (438116) USBH: Device 1 gone
2023-11-09 15:56:30 I (438116) example: HID Device, protocol 'KEYBOARD' DISCONNECTED
2023-11-09 15:56:30 I (438136) example: USB Event flags: ALL_FREE_

### What is the actual behavior?

【1】When I change nothing with the hid example, and the project's Maximum log verbosity is "Same as default"(=info):

When I plug in the Code Scanner, the ESP-IDF monitor print nothing. And when I plug out, it sometimes print ERROR as follows, othertimes print nothing:
_2023-11-09 15:35:02 E (22406) HUB: Bad transfer status 1: CHECK_FULL_CONFIG_DESC
2023-11-09 15:35:02 E (22406) HUB: Stage failed: CHECK_FULL_CONFIG_DESC_

【2】When I add "esp_log_level_set("HUB", ESP_LOG_VERBOSE); esp_log_level_set("USBH", ESP_LOG_VERBOSE);" on the top of app_main(), and the Maximum log verbosity is Verbose：

Everything goes allright.

I've repeated re-plug/reset/re-flash/re-build many times, the running result is steadily affected by whether the log level is set as DEBUG/VERBOSE.

### Steps to reproduce.

1. In VSCODE, select ESP-IDF: Show Examples Projects. Select hid, create the example project in a local folder.
2. Connect my board with PC by the USB-to-UART Port.
3. A 5VDC/0.6A adapter directly connect to the board's 5V and GND pin.
4. Zebra Symbol Code Scanner DS6708's USB-A plug in a USB-OTG wire, and the wire plug in the board's USB Port.
5. Select COM and chip target.
6. Click ESP-IDF Build, Flash and Monitor.
[... 338 lines cut ...]
                           6E 00 63 00 2C 00 20 00 32 00 30 00 30 00 32 00   n.c.,. .2.0.0.2.
             ------ String Descriptor 2 ------
bLength                  : 0x30 (48 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Symbol Bar Code Scanner"
Data (HexDump)           : 30 03 53 00 79 00 6D 00 62 00 6F 00 6C 00 20 00   0.S.y.m.b.o.l. .
                           42 00 61 00 72 00 20 00 43 00 6F 00 64 00 65 00   B.a.r. .C.o.d.e.
                           20 00 53 00 63 00 61 00 6E 00 6E 00 65 00 72 00    .S.c.a.n.n.e.r.
             ------ String Descriptor 3 ------
bLength                  : 0x66 (102 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "S/N:C8820020FD115541B46BDB1CBA3D5E16 Rev:NBRPPAAQ3"
Data (HexDump)           : 66 03 53 00 2F 00 4E 00 3A 00 43 00 38 00 38 00   f.S./.N.:.C.8.8.
                           32 00 30 00 30 00 32 00 30 00 46 00 44 00 31 00   2.0.0.2.0.F.D.1.
                           31 00 35 00 35 00 34 00 31 00 42 00 34 00 36 00   1.5.5.4.1.B.4.6.
                           42 00 44 00 42 00 31 00 43 00 42 00 41 00 33 00   B.D.B.1.C.B.A.3.
                           44 00 35 00 45 00 31 00 36 00 20 00 52 00 65 00   D.5.E.1.6. .R.e.
                           76 00 3A 00 4E 00 42 00 52 00 50 00 50 00 41 00   v.:.N.B.R.P.P.A.
                           41 00 51 00 33 00                                 A.Q.3.
             ------ String Descriptor 4 ------
bLength                  : 0x18 (24 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Bus Powered"
Data (HexDump)           : 18 03 42 00 75 00 73 00 20 00 50 00 6F 00 77 00   ..B.u.s. .P.o.w.
                           65 00 72 00 65 00 64 00                           e.r.e.d.

TRYOKETHEPEN, 2023-11-09:
More information: 
the same hid example with the same board, acts OK when plug in typical keyboard or mouse.

roma-jam, 2023-11-10:
Hi @TRYOKETHEPEN, 

Thanks for the such detailed description and all the logs you had provided! 
Regarding all the information is available from the logs, there are several steps that can help or provide a necessary  information to be able to solve the problem.

May I ask you to try the following things: 
1. `usb_host_lib` example. Depending on the problem, it should works the same way as HID example.
2. Demanded Current  of the device is pretty high (400 mA). That means that power supply should be enough to power both devices. 
3. There are several delays during the enumeration process, that can be configured. Please, refer the [ESP_IDF Documentation: USB Host stack configuration](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_host.html#host-stack-configuration) and try to increase the reset recovery interval. 

If there will be anything new after, feel free to share the updates. 
Thanks.

TRYOKETHEPEN, 2023-11-13:
> Hi @TRYOKETHEPEN,
> 
> Thanks for the such detailed description and all the logs you had provided! Regarding all the information is available from the logs, there are several steps that can help or provide a necessary information to be able to solve the problem.
> 
> May I ask you to try the following things:
> 
> 1. `usb_host_lib` example. Depending on the problem, it should works the same way as HID example.
> 2. Demanded Current  of the device is pretty high (400 mA). That means that power supply should be enough to power both devices.
> 3. There are several delays during the enumeration process, that can be configured. Please, refer the [ESP_IDF Documentation: USB Host stack configuration](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_host.html#host-stack-configuration) and try to increase the reset recovery 间隔.
> 
> If there will be anything new after, feel free to share the updates. Thanks.

(1) **usb_host_lib** example looks like the same:
【change nothing or add code that sets HUB&USBH‘s log level to info】
2023-11-13 08:49:32 I (316) main_task: Started on CPU0
2023-11-13 08:49:32 I (326) main_task: Calling app_main()
2023-11-13 08:49:32 I (326) DAEMON: Installing USB Host Library
2023-11-13 08:49:32 I (366) CLASS: Registering Client
2023-11-13 08:50:48 E (76046) HUB: Bad transfer status 1: CHECK_FULL_CONFIG_DESC
2023-11-13 08:50:48 E (76046) HUB: Stage failed: CHECK_FULL_CONFIG_DESC

【 add code that sets HUB&USBH‘s log level to debug or verbose】
2023-11-13 08:47:03 I (316) main_task: Started on CPU0
2023-11-13 08:47:03 I (326) main_task: Calling app_main()
2023-11-13 08:47:03 I (326) DAEMON: Installing USB Host Library
2023-11-13 08:47:03 I (366) CLASS: Registering Client
2023-11-13 08:47:04 D (776) HUB: Root port reset
2023-11-13 08:47:04 D (776) HUB: Stage done: START
2023-11-13 08:47:04 D (776) HUB: Stage done: GET_SHORT_DEV_DESC
2023-11-13 08:47:04 D (776) HUB: Stage done: CHECK_SHORT_DEV_DESC
2023-11-13 08:47:04 D (836) HUB: Stage done: SECOND_RESET
2023-11-13 08:47:04 D (836) HUB: Stage done: SET_ADDR
2023-11-13 08:47:04 D (836) HUB: Stage done: CHECK_ADDR
2023-11-13 08:47:04 D (846) HUB: Stage done: SET_ADDR_RECOVERY
2023-11-13 08:47:04 D (846) HUB: Stage done: GET_FULL_DEV_DESC
2023-11-13 08:47:04 D (846) HUB: Stage done: CHECK_FULL_DEV_DESC
2023-11-13 08:47:04 D (846) HUB: Stage done: GET_SHORT_CONFIG_DESC
2023-11-13 08:47:04 D (846) HUB: Stage done: CHECK_SHORT_CONFIG_DESC
2023-11-13 08:47:04 D (856) HUB: Stage done: GET_FULL_CONFIG_DESC
2023-11-13 08:47:04 D (856) HUB: Stage done: CHECK_FULL_CONFIG_DESC
2023-11-13 08:47:04 D (866) HUB: Stage done: SET_CONFIG
2023-11-13 08:47:04 D (866) HUB: Stage done: CHECK_CONFIG
2023-11-13 08:47:04 D (866) HUB: Stage done: GET_SHORT_LANGID_TABLE
2023-11-13 08:47:04 D (876) HUB: Stage done: CHECK_SHORT_LANGID_TABLE
2023-11-13 08:47:04 D (876) HUB: Stage done: GET_FULL_LANGID_TABLE
2023-11-13 08:47:04 D (886) HUB: Stage done: CHECK_FULL_LANGID_TABLE
2023-11-13 08:47:04 D (886) HUB: Stage done: GET_SHORT_MANU_STR_DESC
2023-11-13 08:47:04 D (896) HUB: Stage done: CHECK_SHORT_MANU_STR_DESC
2023-11-13 08:47:04 D (896) HUB: Stage done: GET_FULL_MANU_STR_DESC
2023-11-13 08:47:04 D (906) HUB: Stage done: CHECK_FULL_MANU_STR_DESC
2023-11-13 08:47:04 D (906) HUB: Stage done: GET_SHORT_PROD_STR_DESC
2023-11-13 08:47:04 D (916) HUB: Stage done: CHECK_SHORT_PROD_STR_DESC
2023-11-13 08:47:04 D (916) HUB: Stage done: GET_FULL_PROD_STR_DESC
2023-11-13 08:47:04 D (916) HUB: Stage done: CHECK_FULL_PROD_STR_DESC
2023-11-13 08:47:04 D (926) HUB: Stage done: GET_SHORT_SER_STR_DESC
2023-11-13 08:47:04 D (926) HUB: Stage done: CHECK_SHORT_SER_STR_DESC
2023-11-13 08:47:04 D (936) HUB: Stage done: GET_FULL_SER_STR_DESC
2023-11-13 08:47:04 D (936) HUB: Stage done: CHECK_FULL_SER_STR_DESC
2023-11-13 08:47:04 D (946) HUB: Stage done: CLEANUP
2023-11-13 08:47:04 D (946) USBH: Processing actions 0x100
2023-11-13 08:47:04 D (956) USBH: New device 1
2023-11-13 08:47:04 I (956) CLASS: Opening device at address 1
2023-11-13 08:47:04 I (956) CLASS: Getting device information
2023-11-13 08:47:04 I (966) CLASS:      Full speed
2023-11-13 08:47:04 I (966) CLASS:      bConfigurationValue 1
2023-11-13 08:47:04 I (976) CLASS: Getting device descriptor
2023-11-13 08:47:04 *** Device descriptor ***
2023-11-13 08:47:04 bLength 18
2023-11-13 08:47:04 bDescriptorType 1
2023-11-13 08:47:04 bcdUSB 2.00
[... 29 lines cut ...]
2023-11-13 08:47:04     bInterfaceProtocol 0x1
2023-11-13 08:47:04     iInterface 0
2023-11-13 08:47:04             *** Endpoint descriptor ***
2023-11-13 08:47:04             bLength 7
2023-11-13 08:47:04             bDescriptorType 5
2023-11-13 08:47:04             bEndpointAddress 0x81   EP 1 IN
2023-11-13 08:47:04             bmAttributes 0x3        INT
2023-11-13 08:47:04             wMaxPacketSize 8
2023-11-13 08:47:04             bInterval 10
2023-11-13 08:47:04 I (1046) CLASS: Getting Manufacturer string descriptor
2023-11-13 08:47:04 Symbol Technologies, Inc, 20020
2023-11-13 08:47:04 I (1056) CLASS: Getting Product string descriptor
2023-11-13 08:47:04 Symbol Bar Code Scannerh
2023-11-13 08:47:04 I (1066) CLASS: Getting Serial Number string descriptor
2023-11-13 08:47:04 S/N:C8820020FD115541B46BDB1CBA3D5E16 Rev:NBRPPAAQ3

[... cut ...]
(3.2) 100ms: **worked OK once** (not on the first try), and then I tried a dozen times, can't recurrence.
(3.3) 200ms: the plug out error (mentioned above, "HUB: Bad transfer status 1: CHECK_FULL_CONFIG_DESC")’s probability of occurrence  goes down (about 1/2 to 1/10).
(3.4) 250/300/350/500ms: the plug out error changed to "CHECK_FULL_MANU_STR_DESC".

Sorry I don't know much about the Universal Serial Bus 2.0 specification, and have no plan to study it at recent. Except for the **reset recovery delay**, should I try to increase or decrease other delay?

roma-jam, 2023-11-13:
@TRYOKETHEPEN ,

Thanks for the reply, the information was very helpful! 

> should I try to increase or decrease other delay?

It is hard to say without seeing a data on the bus line, but in case the there is obviously that debug output change the situation means that scanner needs a little bit more time at some stage. 

You could try the following: 
1. Try to keep the USB scanner already plugged into the USB port (i.e., already powered), then reset the ESP32-S3 (either via the EN button on the development board, or via idf.py monitor).
2. Try to increase delays (debounce, reset and recovery) from 30 to 100 ms. Decreasing them definitely doesn't make any sense. 

Have you ever had an experience with Wireshark USB traffic capture?

Thanks.

TRYOKETHEPEN, 2023-11-14:
> @TRYOKETHEPEN ,
>
> Thanks for the reply, the information was very helpful!
>
>
> It is hard to say without seeing a data on the bus line, but in case the there is obviously that debug output change the [...]

(1) I've tried _keep the USB scanner already plugged_ before.
(2) Wireshark USB capture log: Device 1.7.0&1.7.1 is the scanner, and the action is plug in -> scann code "9787302484387" -> plug out.
[wiresharkusbcap.zip](https://github.com/espressif/esp-idf/files/13343719/wiresharkusbcap.zip)

TRYOKETHEPEN, 2023-11-14:
> (1) I've tried _keep the USB scanner already plugged_ before. (2) Wireshark USB capture log: Device 1.7.0&1.7.1 is the scanner, and the action is plug in -> scann code "9787302484387" -> plug out. [...]

[wiresharkusbcap.zip](https://github.com/espressif/esp-idf/files/13343748/wiresharkusbcap.zip)

roma-jam, 2023-11-15:
Hi @TRYOKETHEPEN, 

Thanks for the log. Seems like a regular HID keyboard, nothing specific.

Enabling which of the logs helps? HUB or USBH? Have you tried with enabling HUB verbose log only? 

In hub.c:674 ([link](https://github.com/espressif/esp-idf/blob/release/v5.1/components/usb/hub.c#L674)), could you try inserting a delay for the state machine and try it without a debug output for HUB&USBH‘s levels? Something like ` vTaskDelay(pdMS_TO_TICKS(10));` and see if that resolves your issue?

Thanks

TRYOKETHEPEN, 2023-11-15:
> Hi @TRYOKETHEPEN,
>
> Thanks for the log. Seems like a regular HID keyboard, nothing specific.
>
> Enabling which of the logs helps? HUB or USBH? Have you tried with enabling HUB verbose log only?
>
> In [...]

(1) only HUB logs affect this issue.
(2) add delay10ms in **enum_set_next_stage()** can resolve it. 
Thanks for your help, I think this issue can be closed.

closed 2023-11-17 by TRYOKETHEPEN

roma-jam, 2023-11-21:
Hi @TRYOKETHEPEN , 

Thanks for your help and for the very detailed description of every step. 
The problem is technically not closed, because the nature of it unfortunately not so obvious. Adding/Increasing the timeout could only be treated as a workaround here.

Anyway, we will proceed to analyze this issue more and probably come up with some other parameters which could help configure the usb_host stack to support devices, which are not to strict to the timing values (but it is hard to say right now).

If someone run into the same problem with any other device - please, feel free to re-open this issue again. 

Thanks.