espressif/esp-idf #12555: tcp_client example on esp32h2 with constant restarts

KEEPHEADU, 2023-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-222-gb90dfe0759

### Espressif SoC revision.

ESP32-H2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-H2-DevKitM-1 V1.2

### Power Supply used.

USB

### What is the expected behavior?

The chip is operating normally.

### What is the actual behavior?

The chip keeps restarting when running tcp-client demo.

### Steps to reproduce.

1. idf.py flash monitor
[log.txt](https://github.com/espressif/esp-idf/files/13314770/log.txt)

suda-morris, 2023-11-10:
esp32h2 doesn't support wifi, so you need to use the ethernet link.

Check you configuration:

[image]

KEEPHEADU, 2023-11-10:
I'm using an Ethernet link, but it's still rebooting.
[image]

> esp32h2 doesn't support wifi, so you need to use the ethernet link.
> 
> Check you configuration:
> 
> [image]

closed 2023-11-20 by KEEPHEADU