espressif/esp-idf #12556: light_slee failed to connect to ot_cli correctly

KEEPHEADU, 2023-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

light_slee failed to connect to ot_cli correctly, report the following log again;
I've followed the instructions in the readme;
```
I (627) ieee802154: ieee802154 mac sleep retention initialization
I(638) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I (645) OPENTHREAD: OpenThread attached to netif
I(648) OPENTHREAD:[N] Mle-----------: Mode 0x0f -> 0x04 [rx-on:no ftd:no full-net:no]
I (615) main_task: Returned from app_main()
I(662) OPENTHREAD:[N] Mle-----------: Role disabled -> detached
I (672) OT_STATE: netif up
I(1230) OPENTHREAD:[N] Mle-----------: Attach attempt 1, AnyPartition reattaching with Active Dataset
I(7838) OPENTHREAD:[N] Mle-----------: Attach attempt 1 unsuccessful, will try again in 0.295 seconds
I(8145) OPENTHREAD:[N] Mle-----------: Attach attempt 2, AnyPartition 
I(11785) OPENTHREAD:[N] Mle-----------: Attach attempt 2 unsuccessful, will try again in 0.531 seconds
I(12328) OPENTHREAD:[N] Mle-----------: Attach attempt 3, AnyPartition 
I(15957) OPENTHREAD:[N] Mle-----------: Attach attempt 3 unsuccessful, will try again in 1.005 seconds
```
light_sleep dataset:
[image]
ot_cli dataset:
[image]

zwx1995esp, 2023-11-13:
@KEEPHEADU Hi, have you chaecked the active dataset on the `ot_cli` device? Actually we have a logic that we prefer to used a stored network info in the flash rather than use the pre-configured network info. So I guess that your `ot_cli` device might be joined other Thread network and stored that network info to the flash. 

A quick test I suggested is that you firstly set the command `factoryreset` to the `ot_cli` device, after the role changed to Leader, then you start the sleepy-end-device.

KEEPHEADU, 2023-11-15:
> @KEEPHEADU Hi, have you chaecked the active dataset on the `ot_cli` device? Actually we have a logic that we prefer to used a stored network info in the flash rather than use the pre-configured [...]

Ok, thank you for your answer. I will test whether this is feasible later. For now, I will temporarily close the issue.

closed 2023-11-15 by KEEPHEADU