espressif/esp-idf #12557: Ethernet fails with "wrong chip OUI"

webcompas, 2023-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1, v5.3-dev-262-g294c2bd70d

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board (ESP32-POE-ISO Rev. L)

### Power Supply used.

USB

### What is the expected behavior?

When using the example "ethernet_basic" the ESP32 should initialize Ethernet and issue a DHCP request to get an IP address.

### What is the actual behavior?

The ESP32 immediately fails with a panic and the following output:
```
[...]
I (375) main_task: Started on CPU0
I (385) main_task: Calling app_main()
E (405) lan87xx: lan87xx_init(299): wrong chip OUI
E (405) esp_eth: esp_eth_driver_install(229): init phy failed
E (405) example_eth_init: eth_init_internal(89): Ethernet driver install failed
E (405) example_eth_init: example_eth_init(244): internal Ethernet init failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d69ee
[...]
```

### Steps to reproduce.

1. Create a new project from the example "ethernet_basic".
2. Configure the project according to the [board vendor's Ethernet example](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv4.4/sdkconfig).
 ```
CONFIG_EXAMPLE_USE_INTERNAL_ETHERNET=y
CONFIG_EXAMPLE_ETH_PHY_LAN87XX=y
CONFIG_EXAMPLE_ETH_MDC_GPIO=23
CONFIG_EXAMPLE_ETH_MDIO_GPIO=18
CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=-1
CONFIG_EXAMPLE_ETH_PHY_ADDR=0
CONFIG_ETH_PHY_INTERFACE_RMII=y
CONFIG_ETH_RMII_CLK_OUTPUT=y
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
```
3. Build and flash the project.

### More Information.

I did some debugging and came to the result that in `esp_eth_phy_802_3_read_oui()` the retrieved register values of `id1` and `id2` are both 0x0 and hence also `oui` is set to 0x0.
The Ethernet PHY is a LAN8710A-EZC which is supported by the driver.

kostaond, 2023-11-24:
Could you please try to configure `CONFIG_EXAMPLE_ETH_PHY_ADDR=-1` with the latest IDF version (master)?

webcompas, 2023-11-27:
> Could you please try to configure `CONFIG_EXAMPLE_ETH_PHY_ADDR=-1` with the latest IDF version (master)?

I just tried it but without success. It still fails with the same error message (also having set the config value to "0" and "1"). I did the tests with "v5.3-dev-491-g30eb2918c6".

fenderle, 2023-11-27:
The OLIMEX board uses a dedicated FET (FET3 in their schematics) to power the PHY (U4, LAN8710A) using GPIO12. This is something the ethernet/basic example has to be extended for. Otherwise the PHY is not powered, which leads to the invalid OUI. Just add the required commands to set GPIO12 to HIGH and it should work:

**esp-idf/examples/ethernet/basic/main/ethernet_example_main.c:**

```
...
#include "driver/gpio.h"
...
void app_main(void)
{
  // Enable Power to PHY
  const gpio_num_t phy_power_pin = 12;
  gpio_config_t phy_power_conf = {0};
  phy_power_conf.mode = GPIO_MODE_OUTPUT;
  phy_power_conf.pin_bit_mask = (1ULL << phy_power_pin);
  ESP_ERROR_CHECK(gpio_config(&phy_power_conf));
  ESP_ERROR_CHECK(gpio_set_level(phy_power_pin, 1));
  
  // Initialize Ethernet driver
...
}
```

Make sure you also adjust the clock configuration for the PHY since OLIMEX uses GPIO17 for the clock signal. Here are the relevant sdkconfig sections which should be adjusted:
```
CONFIG_ETH_RMII_CLK_OUTPUT=y
CONFIG_ETH_RMII_CLK_OUT_GPIO=17

CONFIG_EXAMPLE_ETH_PHY_LAN87XX=y
CONFIG_EXAMPLE_ETH_MDC_GPIO=23
CONFIG_EXAMPLE_ETH_MDIO_GPIO=18
CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=-1
CONFIG_EXAMPLE_ETH_PHY_ADDR=0
```

kostaond, 2023-11-27:
@fenderle thank you very much for help!

I'm closing the issue since this is not ESP-IDF problem.

sauttefk, 2023-11-27:
@webcompas you could just configure this:
`CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=12`

fenderle, 2023-11-27:
> @webcompas you could just configure this:
> 
> `CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=12`

I tried that and it didn't work for me.

webcompas, 2023-11-27:
Thanks for your help.
> > @webcompas you could just configure this:
> > `CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=12`
> 
> I tried that and it didn't work for me.

For me it's working. So it's not necessary to manually set it to HIGH.