espressif/esp-idf #12563: Generating ld/sections.ld FAILED Error.

stellna, 2023-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I just want to build my own deeplearning esp32 camera (ESP32, OV2640) project but there is an error as a following one.

### What is the actual behavior?

[image]

### Steps to reproduce.

1. download esp-dl and esp-who which is version 5, and Add those components to my new esp-idf project. 
2. Run build
3. Error occured as following one.
    Generating ld/sections.ld FAILED:

### Build or installation Logs.

```plain
*  Executing task: ninja  

[1/8] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\whgPd\OneDrive\Desktop\esp-test\camera_dl_test_3\build\bootloader\esp-idf\esptool_py && C:\Users\whgPd\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/whgPd/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_dl_test_3/build/bootloader/bootloader.bin"
Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
[2/6] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_dl_test_3/build/esp-idf/esp_system/ld/sections.ld 
cmd.exe /C "cd /D C:\Users\whgPd\OneDrive\Desktop\esp-test\camera_dl_test_3\build\esp-idf\esp_system && C:\Users\whgPd\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/whgPd/esp/esp-idf/tools/ldgen/ldgen.py --config C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_dl_test_3/sdkconfig --fragments-list C:/Users/whgPd/esp/esp-idf/components/xtensa/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_ringbuf/linker.lf;C:/Users/whgPd/esp/esp-idf/components/driver/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_pm/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_mm/linker.lf;C:/Users/whgPd/esp/esp-idf/components/spi_flash/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_system/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_system/app.lf;C:/Users/whgPd/esp/esp-idf/components/esp_rom/linker.lf;C:/Users/whgPd/esp/esp-idf/components/hal/linker.lf;C:/Users/whgPd/esp/esp-idf/components/log/linker.lf;C:/Users/whgPd/esp/esp-idf/components/heap/linker.lf;C:/Users/whgPd/esp/esp-idf/components/soc/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_hw_support/linker.lf;C:/Users/whgPd/esp/esp-idf/components/freertos/linker.lf;C:/Users/whgPd/esp/esp-idf/components/freertos/linker_common.lf;C:/Users/whgPd/esp/esp-idf/components/newlib/newlib.lf;C:/Users/whgPd/esp/esp-idf/components/newlib/system_libs.lf;C:/Users/whgPd/esp/esp-idf/components/esp_common/common.lf;C:/Users/whgPd/esp/esp-idf/components/esp_common/soc.lf;C:/Users/whgPd/esp/esp-idf/components/app_trace/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_event/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_phy/linker.lf;C:/Users/whgPd/esp/esp-idf/components/lwip/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_netif/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_wifi/linker.lf;C:/Users/whgPd/esp/esp-idf/components/bt/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_adc/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_adc_cal/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_gdbstub/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_psram/linker.lf;C:/Users/whgPd/esp/esp-idf/components/esp_lcd/linker.lf;C:/Users/whgPd/esp/esp-idf/components/espcoredump/linker.lf;C:/Users/whgPd/esp/esp-idf/components/ieee802154/linker.lf;C:/Users/whgPd/esp/esp-idf/components/openthread/linker.lf --input C:/Users/whgPd/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in --output C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_dl_test_3/build/esp-idf/esp_system/ld/sections.ld --kconfig C:/Users/whgPd/esp/esp-idf/Kconfig --env-file C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_dl_test_3/build/config.env --libraries-file C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_dl_test_3/build/ldgen_libraries --objdump C:/Users/whgPd/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump.exe"
linker script generation failed for C:/Users/whgPd/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
ERROR: Duplicate definition of fragment 'esp_adc' found in C:/Users/whgPd/esp/esp-idf/components/esp_adc/linker.lf and C:/Users/whgPd/esp/esp-idf/components/esp_adc_cal/linker.lf.
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

stellna, 2023-11-11:
I have tried this code on my terminal but it still can't solved the problem.
$ pip install 'pyparsing<3.1.0'

I think I have the same problem as this one, but I really can't understand what is 4020b0d. Can anyone tell me how to solve this problem? 
https://github.com/espressif/esp-idf/issues/11746

Thank you.

igrr, 2023-11-11:
> C:/Users/whgPd/esp/esp-idf/components/esp_adc/linker.lf and
> C:/Users/whgPd/esp/esp-idf/components/esp_adc_cal/linker.lf.

These two components should not exist at the same time... `esp_adc_cal` exists in IDF v4.x and `esp_adc` exists in v5.x. Please check if your IDF copy is somehow corrupted.

stellna, 2023-11-12:
Thank you so much! I don't know a reason but as soon as I delete esp_adc_cal it build successful. But I had a same problem that I needed to make 2 folder so that it can build. And I have to add this code to my Cmakelist in order to build.

if(CONFIG_BOARD_TARGET_PATH)
    set(boards_dir ${CONFIG_BOARD_TARGET_PATH} )
else()
    message(FATAL_ERROR, "Board path not exists")
endif()

if("${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}" VERSION_GREATER_EQUAL "5.0")
    list(APPEND PRIVREQ esp_adc)
else() 
    list(APPEND PRIVREQ esp_adc_cal)
endif()

set(user_src_dir "")
if(DEFINED boards_dir)
message(STATUS "-----------Board Info---------")
message(STATUS "IDF_TARGET = ${IDF_TARGET}")
message(STATUS "Board DIR = ${boards_dir}")
message(STATUS "---------Board Info End---------")
get_filename_component(abs_dir ${boards_dir} ABSOLUTE)
if(IS_DIRECTORY ${abs_dir})
file(GLOB dir_sources "${abs_dir}/*.c")
if(dir_sources)
    set(user_src_dir ${abs_dir})
    message(STATUS "source files found for '${boards_dir}'.")
else()
    message(STATUS "No source files found for '${boards_dir}'.")
endif()
endif()
endif()

#The SRC_DIRS order cannot be changed, otherwise weak functions may not be overridden
idf_component_register( SRC_DIRS "${user_src_dir}" "."
                        INCLUDE_DIRS "${boards_dir}" "."
                        REQUIRES ${PRIVREQ} nvs_flash esp_wifi)
   
   But now I don't understand it was able to solve the problem, But thank you so much for helping me.

sudeep-mohanty, 2023-11-13:
@stellna It looks like you have correctly excluded either the `esp_adc` or the `esp_adc_cal` component based on the IDF version, therefore avoiding the build error. [Here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/peripherals.html#adc-calibration-driver) is some more information from our migration guide about moving the ADC calibration driver from 4.4 to 5.0.

I'm closing this ticket and marking it resolved. Please feel free to open a new ticket in case of further issues. Thanks!

closed 2023-11-13 by sudeep-mohanty