espressif/esp-idf #12568: Incorrect reception in RS485 half-duplex one-wire mode when there is no delay between TX and RX

neoxic, 2023-11-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3 Super Mini board

### Power Supply used.

USB

### What is the expected behavior?

I run ESP32-C3 and ESP32-S2 boards for communication with another board via UART working in half-duplex mode over a single wire (a diode is installed between RX and TX). The link to the project:
https://github.com/neoxic/ESCape32-WiFi-Link

Please take a look here for schematics:
https://github.com/neoxic/ESCape32/wiki/WiFiLink

Expected behaviour is properly received characters no matter if there's a delay or not between TX and RX.

### What is the actual behavior?

I've noticed that the UART driver receives garbage when there is no delay between transmission and reception. In other words, having received a request, the remote board instantly starts transmitting a response which is not received properly by the ESP32-C3. If I add a little delay between reception and transmission on the remote board, all begins to work fine.

### Steps to reproduce.

1. Flash an STM32F05x/STM32F07x board with:
https://github.com/neoxic/ESCape32/releases/download/rel8/BOOT1_PA2-rev2.bin

2. Install a diode between RX and TX on the ESP32-C3 board.

3. Connect RX to PA2 on the STM32 board.

4. Send two characters `0x00 0xFF` using the ESP32-C3. You will receive one character `0x80` back instead of expected two characters `0x00 0xFF`.

To check that a delayed response work:

1. Flash the STM32F05x/STM32F07x board with:
https://github.com/neoxic/ESCape32/releases/download/rel8/BOOT1_PB4-rev2.bin

2. Connect RX to PB4 on the STM32 board.

3. Send two characters `0x00 0xFF` using the ESP32-C3. You will receive two characters `0x00 0xFF`.

### More Information.

The problem stems from inconsistency in `uart_ll_set_mode_rs485_half_duplex()` when `dl0_en=1` and `dl1_en=1` bits are enabled for the ESP32-C2, ESP32-C3, ESP32-C6 and ESP32-H2 and disabled for the ESP32, ESP32-S2 and ESP32-S3. Hence all works as expected on the ESP32-S2 in my case.

Take a look how it's done for the ESP32-C3 for example:
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c3/include/hal/uart_ll.h#L740-L741

And this is how it's done for the ESP32-S2:
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32s2/include/hal/uart_ll.h#L678

neoxic, 2023-11-12:
I created a PR that addresses the issue: https://github.com/espressif/esp-idf/pull/12569

alisitsyn, 2023-11-15:
Hi @neoxic,

Thank you for this issue. Before start progress I would like to get more information and clarify some aspects related to the issue.

> I run ESP32-C3 and ESP32-S2 boards for communication with another board via UART working in half-duplex mode over a single wire (a diode is installed between RX and TX).

According to the schematic the communication bus is just one wire signal SIG = RX/TX with diode + the GND + pullup resistor on RX-SIG (do not see on your connection diagram but it is required). This can be treated as simple bus with collisions allowed.

> The problem stems from inconsistency in uart_ll_set_mode_rs485_half_duplex() when dl0_en=1 and dl1_en=1 bits are enabled for the ESP32-C2, ESP32-C3, ESP32-C6 and ESP32-H2

The bits `dl0_en=1 and dl1_en=1` add additional turnaround delay between bytes (logic one on the bus during one or two baud rate cycles accordingly)  by transmitter of esp32xxx chip and due to logic of your bus should not apply to the output signal in your project. The receiver should be able to get the answer right after sending the stop bit by transmitter. What can apply to the logic according to my expectation is the bit `hw->rs485_conf.rx_busy_tx_en = 1` changed in your code. This behavior applies to bus snooping and collisions avoidance functions implemented differently in some chips and should be tested intentionally to not break existing functionality of the driver. 
I would agree and don't see any reason to add the turnaround delays by default to other chips and I believe they were added by mistake. The turnaround delays could be useful for protocols, and they should be controllable through the appropriate API.

Unfortunately, I can not repeat your results with your reference STM32F05x/STM32F07x board. Using PC with python script and TTL serial adapter could be useful but can prevent to reproduce the same results. Please consider to send the logic analyzer diagram or picture from oscilloscope (with and without your code modification) which would be very useful. 

Thanks.

neoxic, 2023-11-16:
> do not see on your connection diagram but it is required

Internal pull-up is used.

> This can be treated as simple bus with collisions allowed.

Correct. There are no collisions in my case because it's always a request-response sequence.

> The bits dl0_en=1 and dl1_en=1 add additional turnaround delay between bytes (logic one on the bus during one or two baud rate cycles accordingly) by transmitter of esp32xxx chip and due to logic of [...]

Well, the reality is different. Please note that `tx_rx_en=0` (no local echo) in half-duplex mode which prevents the receiver from receiving while the transmitter is transmitting. And so it appears that `dl1_en=1` makes the transmitter busy for one more bit. Hence, the receiver is essentially blocked. When it gets unblocked, it starts receiving one bit into the first character of a response which leads to corrupted data.

> What can apply to the logic according to my expectation is the bit hw->rs485_conf.rx_busy_tx_en = 1 changed in your code. This behavior applies to bus snooping and collisions avoidance functions [...]

`rx_busy_tx_en=1` is irrelevant in my case because there are no collisions on the line. I also tested with it set and cleared, and it has no effect on the issue. In the PR that I submitted, I kept this bit set just to be consistent with the other chips where it's already set, i.e. ESP32/S2/S3. If you know for sure that it needs to be cleared for the ESP32-C3/C6/H2/P4, please feel free to revert that change. Again, it's irrelevant to the issue.

> I would agree and don't see any reason to add the turnaround delays by default to other chips and I believe they were added by mistake. The turnaround delays could be useful for protocols, and they should be controllable through the appropriate API.

100% agree!

> Unfortunately, I will not be able to repeat your results with your reference STM32F05x/STM32F07x board. Using PC with python script and TTL serial adapter could be reproducible but can prevent to get the same results. Please consider to send the logic analyzer diagram or picture from oscilloscope (with and without your code modification) which would be very useful.

Taking into account that you agree that the turnaround delays must not be enabled by default and need to be made controllable by the user instead, I don't see how logic diagrams are relevant here, but if you insist, I will make such diagrams for sure. There's nothing outstanding in there - just four characters `0x00 0xFF 0x00 0xFF` (the first two are a request, the last two are a response) with or without a delay between the pairs.

alisitsyn, 2023-11-16:
> Internal pull-up is used.

Ok. Good.

> No collisions allowed on such a bus unless it's open-drain output. There are no collisions in my case because it's always a request-response sequence, and the output is push-pull.

No collisions in your case with client - server communication. I meant this is analog of bus with dominant and recessive bits and possible connection of several devices to the same bus (i.e CAN, J1708). In this case the collisions are possible and monitoring of the collisions is required.

> Well, the reality is different. Please note that tx_rx_en=0 (no local echo) in half-duplex mode which prevents the receiver from receiving while the transmitter is transmitting. And so it appears [...]

> rx_busy_tx_en=1 is irrelevant in my case because there are no collisions on the line. I tested with it set and cleared, and it has no effect on the issue. Please also note that uart_ll_set_mode_rs485_half_duplex() doesn't have collision detection function, thus it's largely irrelevant what this bit holds. In the PR that I submitted, I kept this bit set to be consistent with the other chips where it's already set, i.e. ESP32/S2/S3.

Ah, yes. Your note with the transmitter FSM being busy for `dl1_en=1` bit looks correct to me but needs to be verified carefully on all targets with the option `tx_rx_en=0`.  As I said before the bus snooping and collisions avoidance functions for RS485 mode may work differently on other targets. In real the behavior with `tx_rx_en=0` works differently on ESP32 due to a HW issue and does not block the RX FSM during sending and then incorrectly generates event `UART_BRK_DET_INT`.

> Taking into account that you agree that the turnaround delays must not be enabled by default and need to be made controllable by the user instead, I don't see how logic diagrams are relevant here, [...]

Well, the diagram is optional. The timings on the diagram may give more information.

neoxic, 2023-11-16:
> In this case the collisions are possible and monitoring of the collisions is required.

I understand that collision detection is possible in certain driver modes like `UART_MODE_RS485_COLLISION_DETECT`, but I use `UART_MODE_RS485_HALF_DUPLEX` mode (`uart_ll_set_mode_rs485_half_duplex()`) which doesn't imply collisions. Please take a look at this comment:
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c3/include/hal/uart_ll.h#L737

> In real the behavior with tx_rx_en=0 works differently on ESP32 due to a HW issue and does not block the RX FSM during sending and then incorrectly generates event UART_BRK_DET_INT.

This is unfortunate, but I don't see how this can be fixed by the `dl0_en=1` and `dl1_en=1` bits. Please also note that no ESP32 code has been modified by my PR (only ESP32-C2/C3/C6/H2/P4).

The `rx_busy_tx_en = 1` fix was just meant for consistency with the similar ESP32-S2/S3 code. It can be reverted if you would rather such inconsistency remain, but personally, I don't see why one would because it seems it was simply copy-pasted along with the set `dl0_en/dl1_en` bits. Again, it doesn't matter what value this bit holds there because `UART_MODE_RS485_HALF_DUPLEX` doesn't imply collisions and collision detection.

alisitsyn, 2023-11-18:
@neoxic ,

Your PR/issue will go through the formal process and verification. The PR and the issue will be updated with the status information.

neoxic, 2024-04-01:
Just wondering if there's any progress with this....

alisitsyn, 2024-04-02:
@neoxic,

Sorry for the delay. The MR for this issue is created and manually tested. The implementation of the automatic test for all targets is not yet completed because of the low priority of this issue. I will let you know when it will be ready to be merged.

neoxic, 2025-04-16:
Bump. Still hoping this will merge one day because for now, I'm still forced to release a dirty (modified) ESP-IDF build each time and migrate the fixes over to each new ESP-IDF release. Doesn't look great...🙁

Massimiliano-solutiontech, 2026-10-03:
The ESP32-C6 half-duplex helper still enables both delay bits in IDF v6.1 (`fff9895c82d744c7237be8847347bdd1b07c6643`) and current master (`4d59230ddff16327812782151ef0afef202dc6d7`). The current C6 path is `components/esp_hal_uart/esp32c6/include/hal/uart_ll.h`.

A fresh C6 bench comparison used an external RS-485 transceiver, two populated eight-node Modbus buses at 9600 baud / 8N1, and simultaneous GetState, set-state and raw-serial requests. Two diagnostic images differed only in the `dl0_en` / `dl1_en` assignments after UART mode setup. Both retained one bit of DE setup before each frame, the driver's TX_DONE release, and the same application inter-frame gap. `rs485rxby_tx_en` remained at the IDF half-duplex default of zero.

| Delay bits | COM1 transactions/s | COM2 transactions/s | Cloud calls with no errors |
|---|---:|---:|---:|
| `dl0_en=1`, `dl1_en=1` | 35.66 | 33.66 | 38/38 |
| `dl0_en=0`, `dl1_en=0` | 35.49 | 35.43 | 38/38 |

Each measured load window lasted 120 seconds. Both had 16 healthy nodes before and after, no bad request CRCs, and no panic or watchdog event. This single application-level comparison is too variable to establish a throughput gain. It also does not reproduce the original immediate-response, diode-based one-wire setup; no TX/DE waveform was captured.

The source behavior and the successful zero-delay C6 run support keeping the delay-setting discussion here and in #12569. A narrow change can explicitly clear the two delay bits while leaving the separate receive-busy setting alone. Removing assignments would not clear a previous mode's register values. A supported delay-configuration API would also avoid application code accessing target-specific registers.

Could the current implementation be reviewed with these changes separated, including the additional targets mentioned in #12569? Transceiver DE setup and protocol frame silence need their own timing requirements and should remain separate from per-byte delay configuration.