espressif/esp-idf #12571: Long Time Run Create websocket and destory Crash

Shaco-Ma, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

esp32s3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I Create websocket in wss url and send and receive some data,then destory the websocket connect, long time run in this mode ,had crash in ssl handshake

### What is the actual behavior?

some heap in tls_prf_generic memory had over in ctx->hmac_ctx

### Steps to reproduce.

1. create websocket with wss url
2. send some data and receive some data
3. about 10S destory connection

### Debug Logs.

```plain
CORRUPT HEAP: Bad tail at 0x3c24a6d4-0x3c24a5d4. Expected 0xbaad5678 got 0x3c1a6560, size:256

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40375ea2:0x3fcaf830 0x40382a79:0x3fcaf850 0x4038a3b1:0x3fcaf870 0x40388b43:0x3fcaf990 0x403769fa:0x3fcaf9b0 0x4037d215:0x3fcaf9d0 0x420e7d56:0x3fcaf9f0 0x420e6449:0x3fcafa10 0x420de81b:0x3fcafa30 0x420de875:0x3fcafac0 0x420deef1:0x3fcafaf0 0x420dfad4:0x3fcafc40 0x420e10ba:0x3fcafcb0 0x420e147a:0x3fcafd20 0x420df2c6:0x3fcafd40 0x420df331:0x3fcafd60 0x420d83da:0x3fcafd80 0x420d7b01:0x3fcafda0 0x420d7d36:0x3fcafdc0 0x420d7e68:0x3fcafe00 0x420d9092:0x3fcafe40 0x42138761:0x3fcafe70 0x420d99d4:0x3fcafe90 0x42138761:0x3fcaff90 0x4201bc88:0x3fcaffb0 0x40385566:0x3fcafff0
CORRUPT HEAP: Bad tail at 0x3c24a6d4-0x3c24a5d4. Expected 0xbaad5678 got 0x3c1a6560, size:256

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40375ea2:0x3fcaf830 0x40382a79:0x3fcaf850 0x4038a3b1:0x3fcaf870 0x40388b43:0x3fcaf990 0x403769fa:0x3fcaf9b0 0x4037d215:0x3fcaf9d0 0x420e7d56:0x3fcaf9f0 0x420e6449:0x3fcafa10 0x420de81b:0x3fcafa30 0x420de875:0x3fcafac0 0x420deef1:0x3fcafaf0 0x420dfad4:0x3fcafc40 0x420e10ba:0x3fcafcb0 0x420e147a:0x3fcafd20 0x420df2c6:0x3fcafd40 0x420df331:0x3fcafd60 0x420d83da:0x3fcafd80 0x420d7b01:0x3fcafda0 0x420d7d36:0x3fcafdc0 0x420d7e68:0x3fcafe00 0x420d9092:0x3fcafe40 0x42138761:0x3fcafe70 0x420d99d4:0x3fcafe90 0x42138761:0x3fcaff90 0x4201bc88:0x3fcaffb0 0x40385566:0x3fcafff0
0x40375ea2: panic_abort at /home/MyData/esp/esp-idf-5.1/components/esp_system/panic.c:452

0x40382a79: esp_system_abort at /home/MyData/esp/esp-idf-5.1/components/esp_system/port/esp_system_chip.c:84

0x4038a3b1: __assert_func at /home/MyData/esp/esp-idf-5.1/components/newlib/assert.c:81

0x40388b43: multi_heap_free at /home/MyData/esp/esp-idf-5.1/components/heap/multi_heap_poisoning.c:276 (discriminator 1)

0x403769fa: heap_caps_free at /home/MyData/esp/esp-idf-5.1/components/heap/heap_caps.c:388

0x4037d215: esp_mbedtls_mem_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/port/esp_mem.c:38

0x420e7d56: mbedtls_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/platform.c:66

0x420e6449: mbedtls_md_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/md.c:294

0x420de81b: tls_prf_generic at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:6265

0x420de875: tls_prf_sha384 at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:6299

0x420deef1: ssl_tls12_populate_transform at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:8365

0x420dfad4: mbedtls_ssl_derive_keys at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:6586

0x420e10ba: ssl_write_certificate_verify at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3260

0x420e147a: mbedtls_ssl_handshake_client_step at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3553

0x420df2c6: mbedtls_ssl_handshake_step at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:3880

0x420df331: mbedtls_ssl_handshake at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:3943

0x420d83da: esp_mbedtls_handshake at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls_mbedtls.c:192

0x420d7b01: esp_tls_handshake at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls.c:114

0x420d7d36: esp_tls_low_level_conn at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls.c:493 (discriminator 15)

0x420d7e68: esp_tls_conn_new_sync at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls.c:522

0x420d9092: ssl_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport_ssl.c:111

0x42138761: esp_transport_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport.c:123

0x420d99d4: ws_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport_ws.c:167

0x42138761: esp_transport_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport.c:123

0x4201bc88: esp_websocket_client_task at /home/shaco/Work/Code/smartlock_esp32s3_1550/components/esp_websocket_client/esp_websocket_client.c:848

0x40385566: vPortTaskWrapper at /home/MyData/esp/esp-idf-5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

the sdkconfig
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13331552/sdkconfig.zip)

the websocket config:
static const esp_websocket_client_config_t wss_conf = 
{   
     .uri = WSS_URL,
     .disable_auto_reconnect = true,
     .task_stack = 6<<10,
     .buffer_size = 9216,
     //.pingpong_timeout_sec = WSS_NO_PINGPONGTIMEOUT,
     .skip_cert_common_name_check =true,
     .keep_alive_enable = true,
 #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 1, 0)
     .reconnect_timeout_ms = 100,
     .network_timeout_ms = 5000,
#endif
};--->WSS_URL:wss://*****

Create websocket connection
taskhandle->client = esp_websocket_client_init(&wss_conf);
if(taskhandle->client != NULL)
{
         ESP_ERROR_CHECK(esp_websocket_register_events(taskhandle->client, WEBSOCKET_EVENT_ANY, websocket_event_handler, NULL));
         ESP_ERROR_CHECK(esp_websocket_client_start(taskhandle->client));
         ESP_LOGI(TAG, "OVER!");
}

Destory websocket connection
if(taskhandle->client != NULL)
{
          hal_system_clear_state(WSS_HAD_LOGIN_BIT);
          websocket_common_set_events(WEBSOCKET_BIT);
          esp_websocket_client_close(taskhandle->client, portMAX_DELAY);
          esp_websocket_client_destroy(taskhandle->client);
          taskhandle->client = NULL;
}

also, I Had usb heap_caps_check_integrity_all before esp_websocket_client_init and after destory websocket connection

and I Can't Find how to fixed this crash,Please Help me,Thanks!

Shaco-Ma, 2023-11-22:
any one can help me?I got other log ,and I create a task every 50ms call heap_caps_check_integrity_all,BUT Can't Check any wrong heap info.how can I solve this?
[0;32mI (19:38:24.219) mbedtls: ssl_msg.c:2124 => flush output
[0m
[0;32mI (19:38:24.225) mbedtls: ssl_msg.c:2133 <= flush output
[0m
[0;32mI (19:38:24.232) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_HANDSHAKE_WRAPUP
[0m

assert failed: block_merge_prev tlsf.c:469 (block_is_free(prev) && "prev block is not free though marked as such")

Backtrace: 0x40375e66:0x3fce5e40 0x4038235d:0x3fce5e60 0x40389ce1:0x3fce5e80 0x40388815:0x3fce5fa0 0x403883b4:0x3fce5fc0 0x4038848e:0x3fce5fe0 0x403769be:0x3fce6000 0x4037d291:0x3fce6020 0x420fb4ca:0x3fce6040 0x420f29d3:0x3fce6060 0x420f6b90:0x3fce6080 0x420f5b08:0x3fce60a0 0x420f5d23:0x3fce60c0 0x420ea9c6:0x3fce60e0 0x420eba58:0x3fce6100 0x420ebb08:0x3fce6120 0x420ee6de:0x3fce6140 0x420ea8ee:0x3fce6160 0x420ea96f:0x3fce6180 0x420e2a4a:0x3fce61a0 0x420e210d:0x3fce61c0 0x420e2354:0x3fce61e0 0x420e248d:0x3fce6220 0x420e379b:0x3fce6260 0x42154a71:0x3fce6290 0x420e4112:0x3fce62b0 0x42154a71:0x3fce63b0 0x4201c124:0x3fce63d0 0x40384e9a:0x3fce6410

assert failed: block_merge_prev tlsf.c:469 (block_is_free(prev) && "prev block is not free though marked as such")

Backtrace: 0x40375e66:0x3fce5e40 0x4038235d:0x3fce5e60 0x40389ce1:0x3fce5e80 0x40388815:0x3fce5fa0 0x403883b4:0x3fce5fc0 0x4038848e:0x3fce5fe0 0x403769be:0x3fce6000 0x4037d291:0x3fce6020 0x420fb4ca:0x3fce6040 0x420f29d3:0x3fce6060 0x420f6b90:0x3fce6080 0x420f5b08:0x3fce60a0 0x420f5d23:0x3fce60c0 0x420ea9c6:0x3fce60e0 0x420eba58:0x3fce6100 0x420ebb08:0x3fce6120 0x420ee6de:0x3fce6140 0x420ea8ee:0x3fce6160 0x420ea96f:0x3fce6180 0x420e2a4a:0x3fce61a0 0x420e210d:0x3fce61c0 0x420e2354:0x3fce61e0 0x420e248d:0x3fce6220 0x420e379b:0x3fce6260 0x42154a71:0x3fce6290 0x420e4112:0x3fce62b0 0x42154a71:0x3fce63b0 0x4201c124:0x3fce63d0 0x40384e9a:0x3fce6410
0x40375e66: panic_abort at /home/MyData/esp/esp-idf-5.1/components/esp_system/panic.c:452

0x4038235d: esp_system_abort at /home/MyData/esp/esp-idf-5.1/components/esp_system/port/esp_system_chip.c:84

0x40389ce1: __assert_func at /home/MyData/esp/esp-idf-5.1/components/newlib/assert.c:81

0x40388815: block_merge_prev at /home/MyData/esp/esp-idf-5.1/components/heap/tlsf/tlsf.c:469
 (inlined by) tlsf_free at /home/MyData/esp/esp-idf-5.1/components/heap/tlsf/tlsf.c:1121

0x403883b4: multi_heap_free_impl at /home/MyData/esp/esp-idf-5.1/components/heap/multi_heap.c:231

0x4038848e: multi_heap_free at /home/MyData/esp/esp-idf-5.1/components/heap/multi_heap_poisoning.c:283

0x403769be: heap_caps_free at /home/MyData/esp/esp-idf-5.1/components/heap/heap_caps.c:388

0x4037d291: esp_mbedtls_mem_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/port/esp_mem.c:38

0x420fb4ca: mbedtls_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/platform.c:66

0x420f29d3: mbedtls_mpi_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/bignum.c:90

0x420f6b90: mbedtls_ecp_point_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ecp.c:555

0x420f5b08: ecdh_free_internal at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ecdh.c:223

0x420f5d23: mbedtls_ecdh_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ecdh.c:264

0x420ea9c6: mbedtls_ssl_handshake_free at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:4140

0x420eba58: mbedtls_ssl_handshake_wrapup_free_hs_transform at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:7881

0x420ebb08: mbedtls_ssl_handshake_wrapup at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:7952

0x420ee6de: mbedtls_ssl_handshake_client_step at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3589

0x420ea8ee: mbedtls_ssl_handshake_step at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:3880

0x420ea96f: mbedtls_ssl_handshake at /home/MyData/esp/esp-idf-5.1/components/mbedtls/mbedtls/library/ssl_tls.c:3943

0x420e2a4a: esp_mbedtls_handshake at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls_mbedtls.c:192

0x420e210d: esp_tls_handshake at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls.c:114

0x420e2354: esp_tls_low_level_conn at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls.c:493 (discriminator 15)

0x420e248d: esp_tls_conn_new_sync at /home/MyData/esp/esp-idf-5.1/components/esp-tls/esp_tls.c:522

0x420e379b: ssl_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport_ssl.c:111

0x42154a71: esp_transport_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport.c:123

0x420e4112: ws_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport_ws.c:167

0x42154a71: esp_transport_connect at /home/MyData/esp/esp-idf-5.1/components/tcp_transport/transport.c:123

0x4201c124: esp_websocket_client_task at /home/shaco/Work/Code/smartlock_esp32s3_1550/components/esp_websocket_client/esp_websocket_client.c:848

0x40384e9a: vPortTaskWrapper at /home/MyData/esp/esp-idf-5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

and What mean prev block is not free though marked as such?it's heap corruption?but I check my code all I use heap,it's ok, or in lvgl lib?

gabsuren, 2023-12-14:
@Shaco-Ma Thank you for bringing this issue to our attention. 
I have attempted to replicate the problem using the logic outlined in the example below
```
#include <stdio.h>
#include "esp_wifi.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "protocol_examples_common.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_tls.h"
#include "freertos/semphr.h"
#include "freertos/event_groups.h"

#include "esp_tls.h"
#include "esp_log.h"
#include "esp_websocket_client.h"
#include "esp_event.h"
#include <cJSON.h>

#define NO_DATA_TIMEOUT_SEC 5

static const char *TAG = "websocket";

static void log_error_if_nonzero(const char *message, int error_code)
{
    if (error_code != 0) {
        ESP_LOGE(TAG, "Last error %s: 0x%x", message, error_code);
    }
}

static void websocket_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    esp_websocket_event_data_t *data = (esp_websocket_event_data_t *)event_data;
    switch (event_id) {
    case WEBSOCKET_EVENT_CONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_CONNECTED");
        break;
    case WEBSOCKET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_DISCONNECTED");
        log_error_if_nonzero("HTTP status code",  data->error_handle.esp_ws_handshake_status_code);
        if (data->error_handle.error_type == WEBSOCKET_ERROR_TYPE_TCP_TRANSPORT) {
            log_error_if_nonzero("reported from esp-tls", data->error_handle.esp_tls_last_esp_err);
            log_error_if_nonzero("reported from tls stack", data->error_handle.esp_tls_stack_err);
            log_error_if_nonzero("captured as transport's socket errno",  data->error_handle.esp_transport_sock_errno);
        }
        break;
    case WEBSOCKET_EVENT_DATA:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_DATA");
        ESP_LOGI(TAG, "Received opcode=%d", data->op_code);
        if (data->op_code == 0x08 && data->data_len == 2) {
            ESP_LOGW(TAG, "Received closed message with code=%d", 256 * data->data_ptr[0] + data->data_ptr[1]);
        } else {
            ESP_LOGW(TAG, "Received=%.*s", data->data_len, (char *)data->data_ptr);
        }

        ESP_LOGW(TAG, "Total payload length=%d, data_len=%d, current payload offset=%d\r\n", data->payload_len, data->data_len, data->payload_offset);

        break;
    case WEBSOCKET_EVENT_ERROR:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_ERROR");
        log_error_if_nonzero("HTTP status code",  data->error_handle.esp_ws_handshake_status_code);
        if (data->error_handle.error_type == WEBSOCKET_ERROR_TYPE_TCP_TRANSPORT) {
            log_error_if_nonzero("reported from esp-tls", data->error_handle.esp_tls_last_esp_err);
            log_error_if_nonzero("reported from tls stack", data->error_handle.esp_tls_stack_err);
            log_error_if_nonzero("captured as transport's socket errno",  data->error_handle.esp_transport_sock_errno);
        }
        break;
    }
}

static void websocket_app_start(void)
{
    esp_websocket_client_config_t websocket_cfg = {};

    websocket_cfg.skip_cert_common_name_check = true;
    websocket_cfg.reconnect_timeout_ms = 100;
    websocket_cfg.network_timeout_ms = 5000;
    websocket_cfg.keep_alive_enable = true;
    websocket_cfg.disable_auto_reconnect = true;
    websocket_cfg.task_stack = 6<<10;
    websocket_cfg.buffer_size = 9216; 
    websocket_cfg.uri = CONFIG_WEBSOCKET_URI;

    ESP_LOGI(TAG, "Connecting to %s...", websocket_cfg.uri);

    esp_websocket_client_handle_t client = esp_websocket_client_init(&websocket_cfg);
    esp_websocket_register_events(client, WEBSOCKET_EVENT_ANY, websocket_event_handler, (void *)client);

    esp_websocket_client_start(client);
    char data[32];
    int i = 0;
    while (i < 5) {
        if (esp_websocket_client_is_connected(client)) {
            int len = sprintf(data, "hello %04d", i++);
            ESP_LOGI(TAG, "Sending %s", data);
            esp_websocket_client_send_text(client, data, len, portMAX_DELAY);
        }
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

    ESP_LOGI(TAG, "Sending fragmented message");
    vTaskDelay(1000 / portTICK_PERIOD_MS);

    esp_websocket_client_close(client, portMAX_DELAY);
    ESP_LOGI(TAG, "Websocket Stopped");
    esp_websocket_client_destroy(client);
    client = NULL;
}

void app_main(void)
{
    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %" PRIu32 " bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("websocket_client", ESP_LOG_DEBUG);
    esp_log_level_set("transport_ws", ESP_LOG_DEBUG);
    esp_log_level_set("trans_tcp", ESP_LOG_DEBUG);

    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());

    websocket_app_start();
}
```

But couldn't find any crashes. 

Could you please mention which version of `esp_websocket_client` do you use, is it the latest in [esp-protocols](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client)?

Otherwise please check the article below for more debugging information
https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/api-reference/system/heap_debug.html
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html?highlight=debugging%20tools