espressif/esp-idf #12572: esp_mesh_stop() don't return

LonesomeUnixCowboy, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6-0-g3572900934

### Espressif SoC revision.

Chip is ESP32-D0WD (revision v1.0) 

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom HW

### Power Supply used.

Battery

### What is the expected behavior?

When `esp_mesh_stop()` is called I expect it to return nicely, not lock up. 

### What is the actual behavior?

Sometimes `esp_mesh_stop()` never returns. It happens more often when there are many ESP32 wrooms communicating on the mesh network.

### Steps to reproduce.

1. Set up a mesh network with 10 ESP32 wrooms according to the documentation/examples.
2. Let them run for 30 s - 60 s after they all have connected so the can begin sending data on the network
3. Shut them down one by one. 

### Debug Logs.

```plain
I have not taken any logs because I don't know what I'm looking for. Therefore I would appreciate some tips from a esp_mesh guru on what log messages that are of interest and then I will happily provide them for you.

But I can say that after entering `esp_mesh_stop()` no esp_mesh logs are printed out. The system hangs and a reset is required.
```

### More Information.

On the mesh network I run a TCP-IP stack with the code in  [https://github.com/espressif/esp-idf/blob/v4.4.6/examples/mesh/ip_internal_network/main/mesh_netif.c](url) if this gives any clue on whats wrong.

I have found an issue by @MisterScience5 here [https://github.com/espressif/esp-idf/issues/12280](url) that describes the same problem that I'm experiencing allthough on v5.1. But I don't want to rely on an watchdog to trigger a reset in our application. The app should go to sleep after 60 s of operation and then wake up every 15 minutes to form a network with other ESP32 wrooms and send data to an off mesh AP. So a watchdog reset would upset the sleeping part of the app.

zhangyanjiaoesp, 2023-11-14:
1.  Is this what your application scenario does?
> The app should go to sleep after 60 s of operation and then wake up every 15 minutes to form a network with other ESP32 wrooms and send data to an off mesh AP.

2. the app got to sleep, is modem sleep ? light sleep ? or deep sleep?
3. Does the `ip_internal_network` example can reproduce your problem?
4.  When you say `Shut them down one by one`, do you mean to turn off a device randomly, or turn off the root node first, or turn off the child node first ？
5. When calling `esp_mesh_stop()`, do all nodes fail to return ? If some can succeed, and some fail, which layer are they on？
6. You can provide a log of any failed device, and maybe we can find some useful information in the log before the lock.

LonesomeUnixCowboy, 2023-11-20:
Thank you for your respons @zhangyanjiaoesp and I apologize that im late with my reply.

1. Is this what your application scenario does? 
	The app wakes up every 15 mins to do a measuremnt and the send this data over the mesh. When this is done it should go to sleep.
	
2 	the app got to sleep, is modem sleep ? light sleep ? or deep sleep?	
	The mode is deep sleep.
	
3. Does the ip_internal_network example can reproduce your problem?
	Im not sure because i have not run the example by it self.
	
4. When you say Shut them down one by one, do you mean to turn off a device randomly, or turn off the root node first, or turn off the child node first ？
	A devices shutsdown by its selfe depending on if they have any child nodes connected to its self or not. If a device has a child node, it will wait
	until the child node has disconnected and gone to sleep before its self goes to sleep.

5. When calling esp_mesh_stop(), do all nodes fail to return ? If some can succeed, and some fail, which layer are they on？
	Its usually only one device that esp_mesh_stop() fails at. I the provided logs there the layers are as follow
	
	Device (a) that esp_mesh_stop() fail is on layer 3
	Device (b) that (a) first connected to is on layer 2
	Device (c) that (a) switched to from (b) is also on layer 2
	Device (d) that (c) and (b) is connected to is on layer 1 and is root.
	
	
6. You can provide a log of any failed device, and maybe we can find some useful information in the log before the lock.
  Yes, i will also provide logs for the other 3 devices that the fail node connected to. The failing node connects to a parent and
  also recives a ip adress, but immedeatly switches parent for some reason. 
After 63 seconds MESH_EVENT_PARRENT_DISCONNECTED drops and at 94 seconds the app puts the node to deep sleep but hangs on esp_mesh_stop(). I have put a debug message 5 and 6 before and after esp_mesh_stop() respectilly and messege 6 is not printed out.

	
	Device (b) prints out [revc]invalid child e8:9f:6d:49:72:38 (which is device (a) adress) numerous times before MESH_EVENT_CHILD_CONNECTED is raising that (a) is connected ag 20 s in. At 58 s (a) is disconnected.
	
	Device (c) then raises at 59 s that device (a) is connected. And at 70 s (a) is disconneted.

LonesomeUnixCowboy, 2023-11-20:
Here are the logs.

[device_d.txt](https://github.com/espressif/esp-idf/files/13408104/device_d.txt)
[device_c.txt](https://github.com/espressif/esp-idf/files/13408105/device_c.txt)
[device_b.txt](https://github.com/espressif/esp-idf/files/13408106/device_b.txt)
[device_a.txt](https://github.com/espressif/esp-idf/files/13408107/device_a.txt)

zhangyanjiaoesp, 2023-11-21:
According to your logs, when the mesh stop fail, it only show `E (94168) ui: sleep 5` log , but don't have the following logs:
```
I (101944) wifi:Set ps type: 0
I (101944) mesh: 452<stop>
```
It is uncertain whether the `esp_mesh_stop()` function has been entered or not. We add some debug log, please replace the wifi lib in `idf/components/esp_wifi/lib/esp32` with the attachment and retest, thanks.
[wifi_lib_1121.zip](https://github.com/espressif/esp-idf/files/13423156/wifi_lib_1121.zip)

LonesomeUnixCowboy, 2023-11-22:
I have runned another test with your provided lib. It took longer time to get a device to fail and I also noticed that sometimes devices didn't get its data sent. But when those devices woke up again at an later stage they managed to send their data. I have not studied those logs yet to see on whats going.

Here is the log from the device that failed.
[device_a.txt](https://github.com/espressif/esp-idf/files/13435053/device_a.txt)

A side note, I have runned `esp_mesh_deinit()` before `esp_mesh_stop()` this time. I have also add more detailed log messages according to this picture:

[image]

As you can see in the log the app never prints out `nodenet: 4` so  `esp_mesh_deinit()` never returns.

hejpadigminvan, 2023-11-23:
I have now runned the ip_internal_network example. I have only added my on coap transmisson to our off mesh AP,  devices go to sleep for 5 minutes so i can call `esp_mesh_deinit() / esp_mesh_stop()` and also set the protocol to `WIFI_PROTOCOL_LR`. I also use your provided wifi lib.

I see the same issue as in my app. The program enters `esp_mesh_deinit()` and never returns.

Here is some logs from two devices. They connect to the same parent and layer.

[ip_internal_network_device_b.txt](https://github.com/espressif/esp-idf/files/13447965/ip_internal_network_device_b.txt)
[ip_internal_network_device_a.txt](https://github.com/espressif/esp-idf/files/13447966/ip_internal_network_device_a.txt)

LonesomeUnixCowboy, 2023-11-23:
I have now runned the ip_internal_network example. I have only added my on coap transmisson to our off mesh AP, devices go to sleep for 5 minutes so i can call esp_mesh_deinit() / esp_mesh_stop() and also set the protocol to WIFI_PROTOCOL_LR. I also use your provided wifi lib.

I see the same issue as in my app. The program enters esp_mesh_deinit() and never returns.

Here is some logs from two devices. They connect to the same parent and layer.

[ip_internal_network_device_b.txt](https://github.com/espressif/esp-idf/files/13450466/ip_internal_network_device_b.txt)
[ip_internal_network_device_a.txt](https://github.com/espressif/esp-idf/files/13450467/ip_internal_network_device_a.txt)

zhangyanjiaoesp, 2024-03-18:
@LonesomeUnixCowboy Do you still have this problem?

LonesomeUnixCowboy, 2024-03-18:
I apologize for not responding sooner.

No, the issue is solved. The error was that `esp_mesh_recv()` in `receive_task()` in `mesh_netif.c` would sometimes never return. Which made my code waiting forever when calling `esp_mesh_stop()`. I modified the call to  `esp_mesh_recv()` so it timesout after a couple of seconds which then makes  `esp_mesh_stop()` return.

Again I apologize for not coming back to you. Thank you @zhangyanjiaoesp for you help.

zhangyanjiaoesp, 2024-03-19:
@LonesomeUnixCowboy It's okay. I'm glad to hear that your problem has been resolved.