espressif/esp-idf #12573: Nimble / bss in SPIRAM, non-deterministic result - sometimes in SPIRAM, sometimes not.

JamieDriver, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1 (and I think many prior)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

With SPIRAM enabled, with 'allow bss in spiram' and nimble to use external ram (as below) I expect Nimble to use the .ext_ram.bss section.
At the very least the build is expected to be deterministic, and rebuilding to produce the same output in this regard.
```
CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
```

### What is the actual behavior?

The built is not not deterministic - sometimes the `.ext_ram.bss` is used, sometimes not.
```
            Archive File DRAM .data & 0.bss .ext_ram.bss IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .appdesc flash_total
           libbtdm_app.a        870    2767            0       30475           0        34112       81344      5274          0      117963
                  libc.a         45       0            0        7336           0         7381       83383      4739          0       95503
**->             libbt.a       5078       0         6398        5214           0        16690       68898      8026          0       87216
```
vs:
```
            Archive File DRAM .data & 0.bss IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .appdesc flash_total
           libbtdm_app.a        870    2767       30475           0        34112       81344      5274          0      117963
                  libc.a         45       0        7336           0         7381       83383      4739          0       95503
**->             libbt.a       5078    6398        5214           0        16690       68898      8026          0       87216
```

Repeated rebuilding reveals `build/esp-idf/esp_system/ld/sections.ld` is not deterministic, and sometimes contains:
```
   /* external memory bss, from any global variable with EXT_RAM_BSS_ATTR attribute*/
  .ext_ram.bss (NOLOAD) :
  {
    _ext_ram_bss_start = ABSOLUTE(.);

    *(.ext_ram.bss .ext_ram.bss.*)
    *libbt.a:(.bss .bss.*)
    *libbt.a:(COMMON)
    *liblwip.a:(.bss .bss.*)
    *liblwip.a:(COMMON)
    *libnet80211.a:(.bss .bss.*)
    *libnet80211.a:(COMMON)
    *libpp.a:(.bss .bss.*)
    *libpp.a:(COMMON)

    . = ALIGN(4);
    _ext_ram_bss_end = ABSOLUTE(.);
  } > extern_ram_seg
```
[... 216 lines cut ...]
libnimble_peripheral_uti         20       8           0           0           28         646       159          0         825
                  libm.a          0       0           0           0            0         602         0          0         602
     libesp_app_format.a          1       8         211           0          220          27         4        256         499
           libesp_coex.a         88       0         166           0          254         224         0          0         478
             libxt_hal.a          0       0         443           0          443           0        32          0         475
         libapp_update.a          0       4           0           0            4         158       125          0         283
                libgcc.a          0       0           0           0            0          89       160          0         249
            libesp_rom.a          0       0         215           0          215           0         0          0         215
           libesp_wifi.a          0       4           0           0            4          54        39          0          93
                libcxx.a          0       0           0           0            0          44         0          0          44
                   (exe)          0       0           0           3            3           3        12          0          18
               libcore.a          0       0           0           0            0           0         0          0           0
          libesp_event.a          0       0           0           0            0           0         0          0           0
            libmbedtls.a          0       0           0           0            0           0         0          0           0
           libmbedx509.a          0       0           0           0            0           0         0          0           0
               libmesh.a          0       0           0           0            0           0         0          0           0
           libnet80211.a          0       0           0           0            0           0         0          0           0
                 libpp.a          0       0           0           0            0           0         0          0           0
     libwpa_supplicant.a          0       0           0           0            0           0         0          0           0
```
```

### More Information.

May need to re-run multiple times to see both states.

igrr, 2023-11-13:
This should be fixed, i think, in be88d11a818f291bedeaa3bcaa61de26bde4b057. The backport commit in release/v5.1 branch is de93968670b68e7a3b7fed02b8493c12d5fb29e5.

JamieDriver, 2023-11-13:
Wow that was quick work!  ;-)
Cherry-picked that commit locally and agree that appears to resolve the issue.  
Many thanks.

JamieDriver, 2024-01-10:
We've bumped to release v5.1.2 and the issue appears resolved I think.
Closing - many thanks.

closed 2024-01-10 by JamieDriver

andreaslundeen, 2024-04-18:
I have the exact issue using ESP-IDF 4.4.6.  When will this be part of official 4.4.x release?