espressif/esp-idf #12574: malloc_internal_wrapper in blufi return NULL

Shaco-Ma, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

esp32s3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I set CONFIG_SPIRAM_USE_MALLOC and CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=512 or CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=1024
malloc_internal_wrapper will be return NULL,because in malloc_internal_wrapper
void *p = heap_caps_malloc(size, MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);
if set CONFIG_SPIRAM_USE_MALLOC,the MALLOC_CAP_DEFAULT and the memory bigger CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL will be in psram,but the size malloc is only 7,when blufi connect like this
I (17:23:22.343) BLE_BLUFI: blufi_event_callback:405: BLUFI ble connect
I (17:23:22.344) BLE_BLUFI: >>>>>>>>>>hal_system_setDevicestate :1
blufi_security_init:206:0x3c267410--456
I (17:23:22.351) APP_BLE_CLIENT: esp_gap_cb:1138: stop adv successfully
Mem alloc fail. size 0x00000007 caps 0x00001808   -------> the size is 0x00000007

Backtrace: 0x40375eb6:0x3fcda370 0x40382b51:0x3fcda390 0x403767fd:0x3fcda3b0 0x40376852:0x3fcda400 0x42055e6b:0x3fcda420 0x4000a671:0x3fcda440 0x400074ee:0x3fcda460 0x4000ccc9:0x3fcda480 0x4000c984:0x3fcda4a0 0x40378421:0x3fcda4c0 0x4038568e:0x3fcda4f0
0x40375eb6: panic_abort at /home/MyData/esp/esp-idf-5.1/components/esp_system/panic.c:452

0x40382b51: esp_system_abort at /home/MyData/esp/esp-idf-5.1/components/esp_system/port/esp_system_chip.c:84

0x403767fd: heap_caps_alloc_failed at /home/MyData/esp/esp-idf-5.1/components/heap/heap_caps.c:96

0x40376852: heap_caps_malloc at /home/MyData/esp/esp-idf-5.1/components/heap/heap_caps.c:201

0x42055e6b: malloc_internal_wrapper at /home/MyData/esp/esp-idf-5.1/components/bt/controller/esp32c3/bt.c:649

0x4000a671: r_emi_alloc_em_mapping_by_offset in ROM

0x400074ee: r_ble_util_buf_acl_tx_alloc in ROM

0x4000ccc9: r_hci_acl_tx_data_alloc in ROM

0x4000c984: r_hci_tl_hci_pkt_handler in ROM

0x40378421: btdm_controller_task at ??:?

0x4038568e: vPortTaskWrapper at /home/MyData/esp/esp-idf-5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

### What is the actual behavior?

crash in malloc_internal_wrapper when set malloc fail abort
this maybe a bug?

### Steps to reproduce.

1. set CONFIG_SPIRAM_USE_MALLOC
2. set CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL = 512 or 1024
3. blufi connected

### More Information.

the sdkconfig
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13334260/sdkconfig.zip)

zhp0406, 2023-11-16:
@Shaco-Ma 
if change 

```
void *p = heap_caps_malloc(size, MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);
```

to

```
void *p = heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);
```

Will the problem disappear?

Shaco-Ma, 2023-11-20:
@zhp0406 I Change to void *p = heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA); is OK
the complete log?U mean all of the blufi log?with ble verbose?

zhp0406, 2023-11-20:
@Shaco-Ma 
Please provide me with all the logs after the chip has been activated. I need them for problem analysis.

bryghtlabs-richard, 2023-11-28:
@zhp0406 , I've also seen this when using NimBLE and PSRAM. For me it occurs the moment phone establishes BLE connection. I attach my logs. I also tried removing MALLOC_CAP_DEFAULT, and that seems to work and allows the connection to establish and operate. I had been trying to trace this out as well, so logs contain a lot of heap stats printouts
[ESP_IDF_issues_12574_DEFAULT_OR_INTERNAL_OR_DMA_NOTWORKING.txt](https://github.com/espressif/esp-idf/files/13494172/ESP_IDF_issues_12574_DEFAULT_OR_INTERNAL_OR_DMA_NOTWORKING.txt)
[ESP_IDF_issues_12574_INTERNAL_OR_DMA_WORKING.txt](https://github.com/espressif/esp-idf/files/13494173/ESP_IDF_issues_12574_INTERNAL_OR_DMA_WORKING.txt)
.

zhp0406, 2023-11-29:
@bryghtlabs-richard 
which chip?

zhp0406, 2023-11-29:
@bryghtlabs-richard @Shaco-Ma 
In the previous approach, the memory used by BLE needs to satisfy the capabilities of MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA. If we eliminate the MALLOC_CAP_DEFAULT capability, more memory becomes available.
Thank you for your feedback; we will evaluate whether to disable the MALLOC_CAP_DEFAULT capability.
Please disable MALLOC_CAP_DEFAULT in your current usage. If you encounter any issues with memory allocation again, please let me know.

zhp0406, 2023-11-29:
@bryghtlabs-richard 
Do you use WIFI? If you can provide your code to reproduce the issue, I would greatly appreciate.

bryghtlabs-richard, 2023-11-29:
Thanks @zhp0406 ,

0) Chip is ESP32-S3
1) It can be reproduced on v5.1.1, but so far not on v5.1.2. Even on v5.1.1, it is intermittent and can disappear with unrelated code changes. I'm not sure how to build against one newer than 5.1.x. Mine is only dirty from adding some logging.
2) We are not using WiFi on this build.
3) I don't think I can share the code with you, but, thanks to your help, I think I can explain what's occurring, at least on my board.

I did extend my logging to include those combinations of heap caps on v5.1.1, to show the issue you've explained:

```
I (140285) example: Free Heap:
  MALLOC_CAP_8BIT                                      |8405051 bytes
  MALLOC_CAP_DMA                                       |  12311 bytes
  MALLOC_CAP_SPIRAM                                    |8385108 bytes
  MALLOC_CAP_INTERNAL                                  |  19943 bytes
  MALLOC_CAP_DEFAULT                                   |8392764 bytes
  MALLOC_CAP_IRAM_8BIT                                 |      0 bytes
  MALLOC_CAP_RETENTION                                 |      8 bytes
  MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA|     24 bytes
  MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA                   |  12311 bytes

I (140825) NimBLE: connection established; status=0
I (140825) NimBLE: handle=1 our_ota_addr_type=0 our_ota_addr=
I (140825) NimBLE: f4:12:fa:e6:4e:22
I (140825) NimBLE:  our_id_addr_type=0 our_id_addr=
I (140835) NimBLE: f4:12:fa:e6:4e:22
I (140835) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (140845) NimBLE: 55:b6:25:94:36:6e
I (140845) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (140855) NimBLE: 55:b6:25:94:36:6e
I (140855) NimBLE:  conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

I (140865) NimBLE:

Mem alloc fail. size 0x00000010 caps 0x00001808
```

Indeed, the specified DEFAULT|INTERNAL|DMA memory is down to 24B, so a 16B allocation with those caps fails. I have some debug/tracing options enabled, so per-allocation may be a little higher for me than default. I also do have CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768, but I believe MALLOC_CAP_DEFAULT would prevent usage of the reserved internal/dma pool.

> Thank you for your feedback; we will evaluate whether to disable the MALLOC_CAP_DEFAULT capability. Please disable MALLOC_CAP_DEFAULT in your current usage. If you encounter any issues with memory allocation again, please let me know.

Got it. By disabling MALLOC_CAP_DEFAULT (v5.1.1 with only MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA), BLE connection memory now comes from the larger INTERNAL|DMA memory, and I have not seen the allocation failure yet today.

I also noticed that when the out-of-memory condition occurs, my MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA is very small(like 24B), but sometimes it is much larger(like 7888B). I think separately from the choice of using DEFAULT flag for BLE allocator, I may have some allocation that sometimes is placed in INTERNAL and sometimes is placed in PSRAM. If it fits in INTERNAL, then DEFAULT|INTERNAL|DMA is much smaller, and the risk increases. But if it barely doesn't fit in INTERNAL, then DEFAULT|INTERNAL|DMA has enough space to support BLE allocations, which is the case when BLE works on unpatched v5.1.1.

Alvin1Zhang, 2024-02-20:
Thanks for reporting, fix on release/5.1 is available https://github.com/espressif/esp-idf/commit/4fd3de1ce8551c94ec4cb01c1725d135a94700d1, feel free to reopen if the issue still happens, thanks.

closed 2024-02-20 by Alvin1Zhang