espressif/esp-idf #12575: HTTP post of unsigned char binary data is not decoded properly

chaabanemalki, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I'm using esp-idf v4.3.6 on ESP32 to post data 
I'm sending 255 bytes that goes from 0x00 to 0xFE
This packet is posted to the test webhook.site 

This is what I'm trying to send
[image]

When I use postman, I get the data as expected ( from 0x00 to 0xFE)

[image]

When I send it using esp-idf http component ( see code below)

I only get from 0x00 to 0x7F and the rest is 0x3F 

[image]

```
      char data2send[256];
      for (size_t i = 0; i < 255; i++) {
          data2send[i] = i;
      } 
    esp_http_client_set_url(client, c_data_url);
    esp_http_client_set_authtype(client, HTTP_AUTH_TYPE_NONE);
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "x-api-key", X_API_KEY_VALUE);
    esp_http_client_set_header(client, "Content-Type", "text/plain");
    esp_http_client_set_header(client, "accept", "application/json");
    esp_http_client_set_post_field(client, data2send, len);
    ESP_LOG_BUFFER_HEX_LEVEL(TAG, data2send, len, ESP_LOG_INFO);    

    error = esp_http_client_perform(client);
    int status = esp_http_client_get_status_code(client);
```

Can you please advice how to fix this issue ?

hmalpani, 2023-11-20:
Hello @chaabanemalki 

I tried this on both master as well as v4.3.6
It works fine for me on both master and v4.3.6. I am sharing a wireshark screenshot for reference

[image]

Here is my code snippet:
```
    char data2send[256];
      for (size_t i = 0; i < 255; i++) {
          data2send[i] = i;
      } 
    
    esp_http_client_set_url(client, "http://192.168.20.198:8000/");
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "Content-Type", "application/json");
    esp_http_client_set_post_field(client, data2send, 256);
    esp_err_t err = esp_http_client_perform(client);
    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %"PRId64,
                 esp_http_client_get_status_code(client),
                 esp_http_client_get_content_length(client));
    } else {
        ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
    }
```

hmalpani, 2023-12-06:
Hello @chaabanemalki 

I have tested the code you shared above with both webhook.site and a local HTTPS server. With the webhook.site, I saw similar result as you shared. But with the local server, it was working fine. I suspect that webhook.site is unable to render the ASCII characters greater then 0x7F and thus just prints `?` instead. What you are doing is copying the received data as seen on webhook.site and converting it into hex. That is the reason you are seeing `0x3F` which is ASCII for `?`.

Hope this helps.