espressif/esp-idf #12576: Bootloader encrypt_flash_contents: add means to encrypt application image only

boborjan2, 2023-11-13:
### Is your feature request related to a problem?

When enabling flash encryption, bootloader will set relevant fuses and encrypt partitions on next boot. Partition encryption is very slow btw: it takes ~30s to encrypt a 1.5M partition. It becomes even more relevant when we first flash a small app for eol testing. It basically doubles our eol times (compared to the non-encrypted process).

### Describe the solution you'd like.

App partition encryption is performed by encrypt_partition() in bootloader_support/src/flash_encryption/flash_encrypt.c. Here all the information is available to encrypt the image only (instead of the whole partition). An example solution is as follows:
```
static esp_err_t encrypt_partition(int index, const esp_partition_info_t *partition)
{
    esp_err_t err;
    bool should_encrypt = (partition->flags & PART_FLAG_ENCRYPTED);
    uint32_t size = partition->pos.size;

    if (partition->type == PART_TYPE_APP) {
        /* check if the partition holds a valid unencrypted app */
        esp_image_metadata_t metadata;
        err = esp_image_verify(ESP_IMAGE_VERIFY,
                               &partition->pos,
                               &metadata);
        if((should_encrypt = (err == ESP_OK))) {
            size = metadata.image_len;
        }
    } else if ((partition->type == PART_TYPE_DATA && partition->subtype == PART_SUBTYPE_DATA_OTA)
                || (partition->type == PART_TYPE_DATA && partition->subtype == PART_SUBTYPE_DATA_NVS_KEYS)) {
        /* check if we have ota data partition and the partition should be encrypted unconditionally */
        should_encrypt = true;
    }

    if (!should_encrypt) {
        return ESP_OK;
    } else {
        /* should_encrypt */
        ESP_LOGI(TAG, "Encrypting partition %d at offset 0x%x (length 0x%x)...", index, partition->pos.offset, size);

        err = esp_flash_encrypt_region(partition->pos.offset, size);
        ESP_LOGI(TAG, "Done encrypting");
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "Failed to encrypt partition %d", index);
        }
        return err;
    }
}
```
The change may also be configurable via menuconfig.

mahavirj, 2023-11-15:
@boborjan2 

Thanks for the suggestion, looks quite useful!

However, the actual image length may vary based on the secure boot enabled case. We will have to consider the padding, signature block into calculations. We will analyze this further and share the update on this tracker.

closed 2023-12-23 by espressif-bot

Harshal5, 2023-12-26:
Feature implemented in https://github.com/espressif/esp-idf/commit/42943845e49a2e4f39a79a8a1f75a91f71e16fc0, thus closing this issue.