espressif/esp-idf #12577: esp_pm_lock_create() with invalid lock type causes abort

xyzzy42, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-277-gc8243465e4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitC

### Power Supply used.

USB

### What is the expected behavior?

If an invalid lock type is used, per ESP-IDF documentation for `esp_pm_lock_create()`, expected behavior is:

> **`ESP_ERR_INVALID_ARG`** if out_handle is NULL or type argument is not valid

### What is the actual behavior?

Lock is created without checking type.  Invalid lock then immediately causes an abort().

### Steps to reproduce.

Call `esp_pm_lock_handle_t lock; esp_pm_lock_create(42, 0, "test", &lock);`

### Debug Logs.

```plain
abort() was called at PC 0x42110ff6 on core 1

Backtrace: 0x40375a62:0x3fcaeae0 0x403869c5:0x3fcaeb00 0x4038cff2:0x3fcaeb20 0x42110ff6:0x3fcaeb90 0x420783be:0x3fcaebb0 0x4202ed1c:0x3fcaebd0 0x4201471d:0x3fcaebf0 0x420163cd:0x3fcaec10 0x42016491:0x3fcaec30 0x40378e3a:0x3fcaec50 0x4200eee7:0x3fcaecf0 0x420163cd:0x3fcaed50 0x420163e2:0x3fcaed70 0x420253b6:0x3fcaed90 0x420256a4:0x3fcaee20 0x420071dc:0x3fcaee60

ELF file SHA256: 5b06583c1

Rebooting...
```
```

10086loutianhao, 2023-11-14:
hello, Why did you choose 42 as the parameter? Please refer to the available types in this location by checking esp_pm_lock_type_t

xyzzy42, 2023-11-14:
Because it is not a valid value.

The point is that the documentation says invalid values will return `ESP_ERR_INVALID_ARG`, but they do not.  Instead they cause a crash.

10086loutianhao, 2023-11-14:
Thank you for your feedback. After internal discussions, we plan to update the function description, while keeping the functionality of abort unchanged.

xyzzy42, 2023-11-14:
Could a enum at the end of `esp_pm_lock_type_t` like `ESP_PM_LOCK_MAX` be added?  This way it would be possible to check if the type is valid by comparing it to `lock < ESP_PM_LOCK_MAX`?  Similar to `WIFI_AUTH_MAX` and many other enums.

I implemented support for pm locks in Micropython.  The lock type is user Python code, so it might an invalid value.  If `esp_pm_lock_create()` doesn't check, then I need to validate first in Micropython.  Which is certainly possible.  But the only way to do that is to compare against every possible lock type, and then update this every time a new lock type is added.  Which is hard to maintain.  If there was an enum for the last lock type, it would be easier to validate.

10086loutianhao, 2023-11-14:
Okay, I will add ESP_PM_LOCK_MAX for you, but please note that this change will only be effective on the master branch

10086loutianhao, 2023-11-21:
hello You can pull the master branch for validation

xyzzy42, 2023-11-28:
Thank you, I've tested this and it works as expected.  Code using it in Micropython is in commit xyzzy42/micropython@5c5dbba61818da2fe9ce1fe3bb542a2fa0edc549 in my branch [xyzzy42/micropython/low-power](/xyzzy42/micropython/tree/low-power).

```
>>> import esp32
>>> cpu = esp32.Pmlock(0, "CPU")
>>> apb = esp32.Pmlock(1, "APB")
>>> sleep = esp32.Pmlock(2, "SLEEP")
>>> fail = esp32.Pmlock(3, "fail")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid lock type
>>> esp32.pm_dump_locks()
Time since bootup: 11955048 us
Lock stats:
Name            Type            Arg    Active    Total_count    Time(us)        Time(%)
SLEEP           NO_LIGHT_SLEEP  0      0         0              0               0  %
APB             APB_FREQ_MAX    0      0         0              0               0  %
CPU             CPU_FREQ_MAX    0      0         0              0               0  %
stdin           NO_LIGHT_SLEEP  0      1         1              10994331        92 %
rtos1           CPU_FREQ_MAX    0      1         11806          1498064         13 %
rtos0           CPU_FREQ_MAX    0      0         12264          1457671         13 %
```

closed 2023-11-29 by Alvin1Zhang