espressif/esp-idf #12578: esp_http_client_perform(client) slow and is_async causing non connect

ThatBigPrint, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1 release

### Espressif SoC revision.

Esp32-devkit

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Esp32-devkit

### Power Supply used.

USB

### What is the expected behavior?

I want the stream of data to be responsive its only responding to changes every 5 seconds... So I tried to use is_async but the client can't connect... I added certificates for https which was supposed to be a prerequisite 

### What is the actual behavior?

It's just looping through every 5 seconds ~ then giving me the new data if there is any... Or the keep alive 

If I use the is_async I get the can't connect error 

### Steps to reproduce.

Same as the code for HTTP client example

ThatBigPrint, 2023-11-14:
Updates ?

suryasid09, 2023-11-27:
hello, did you find what was causing the issue? 
I too am facing the same issue when I try to do http stream using http_client_open(). When I use http_client_perform(), it works fine, but speed is around 400KB/sec which I want to improve using http streamer as mentioned in the esp documentation. 

Funny thing! when I use http_client_open(), it throws me the same error what you got. And after that if I modify my program to use http_client_perform(), it gives the same error and stops working all together!

suryasid09, 2023-11-27:
@hmalpani I tried with the patch you provided, it didn't fix the issue. A more context in what I am doing.

I am reading from the USB 4KB of data until end of the file. After each 4KB read operation, I am uploading it to the locally created XAMP server using HTTP post. I was already getting good speed of around 400KB/sec but I want to improve it. So, I decided to use HTTP in streamer mode as mentioned in the documentation. However, I always get the error as above pointed. 

This is the screenshot from Wireshark and esp32. I would appreciate any help in this regard.
[image]
[image]

ThatBigPrint, 2023-12-09:
@hmalpani the patch worked but why isn't it in the main repo ?

hmalpani, 2023-12-11:
Hello @ThatBigPrint 
The fix has been merged into master https://github.com/espressif/esp-idf/commit/61d4775027d142d7ea116164a1bae8d344bf73db

I didn't quite understand the second question. Can you please elaborate?

Thanks!

ThatBigPrint, 2023-12-11:
so this is the part of my code the sse stream is fine but when i call the post on a separate client and handler it kicks the stream off am i missing something? i want to be able to post to the database whilst the main sse client is running i have tried the perform on the same client etc but this is the closest I've had it to working. do i need to interrupt the perform in the sse stream task whilst the post is happening?

idealy i want to post using te same client and handler but im not sure how i can call the post within the inbound connection client ?
`static void post_rest_function()
{
    char full_url[256];
    char mac_str[20];
    uint8_t mac[6] = {0};
    esp_wifi_get_mac(ESP_IF_WIFI_STA, mac);
    snprintf(mac_str, sizeof(mac_str), "%02x%02x%02x%02x%02x%02x", MAC2STR(mac));
    ESP_LOGI(TAG, "MAC: %s", mac_str);

    // snprintf(full_url, sizeof(full_url), "%s%s?auth=%s", FIREBASE_HOST, FIREBASE_PATH, FIREBASE_AUTH);
    snprintf(full_url, sizeof(full_url), "%s/%s%s.json?auth=%s", FIREBASE_HOST, rvd_data_str, FIREBASE_PATH, FIREBASE_AUTH);
    //snprintf(full_url, sizeof(full_url), "%s/%s%s%s.json?auth=%s", FIREBASE_HOST, rvd_data_str, FIREBASE_PATH, mac_str, FIREBASE_AUTH);

    esp_http_client_config_t config_outbound = {
        .url = full_url,
        .method = HTTP_METHOD_PUT,
        .crt_bundle_attach = esp_crt_bundle_attach,
        .event_handler = outbound_http_event_handler,
        };

    outbound_client = esp_http_client_init(&config_outbound);

char* outbound_data = "{}";
    ESP_LOGI(TAG, "Outbound data: %s", outbound_data);

    esp_http_client_set_header(outbound_client, "Content-Type", "application/json");
    esp_http_client_set_header(outbound_client, "Accept", "application/json");
    esp_http_client_set_post_field(outbound_client, outbound_data, strlen(outbound_data));

    esp_err_t err = esp_http_client_perform(outbound_client);
    if (err == ESP_OK)
    {
        ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %" PRId64,
                 esp_http_client_get_status_code(outbound_client),
                 esp_http_client_get_content_length(outbound_client));
    }
    else
    {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(outbound_client);
}

void sse_stream_task(void *pvParameters)
{
    start_config = true;
    char *buffer = malloc(MAX_HTTP_RECV_BUFFER + 1);
    if (buffer == NULL)
    {
        ESP_LOGE(TAG, "Cannot malloc http receive buffer");
        return;
    }

    char full_url[256];

    // snprintf(full_url, sizeof(full_url), "%s%s?auth=%s", FIREBASE_HOST, FIREBASE_PATH, FIREBASE_AUTH);
    snprintf(full_url, sizeof(full_url), "%s/%s%s.json?auth=%s", FIREBASE_HOST, rvd_data_str, FIREBASE_PATH, FIREBASE_AUTH);

    ESP_LOGI(TAG, "Connecting to Firebase URL: %s", full_url);

    esp_http_client_config_t config_inbound = {
        .url = full_url,
        .crt_bundle_attach = esp_crt_bundle_attach,
        .event_handler = inbound_http_event_handler,
        .is_async = true,
        .user_data = buffer,
    };
    inbound_client = esp_http_client_init(&config_inbound);
    esp_http_client_set_header(inbound_client, "Accept", "text/event-stream");
    esp_err_t err;
    while (1)
    {
        err = esp_http_client_perform(inbound_client);
        if (err != ESP_ERR_HTTP_EAGAIN)
        {
            //break;
        }
    }
    if (err == ESP_OK)
    {
        ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %" PRId64,
                 esp_http_client_get_status_code(inbound_client),
                 esp_http_client_get_content_length(inbound_client));
    }
    else
    {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(inbound_client);
}`

just to clarify the functions work indipendantly its just the posting whilst the sse is running blocks the sse as per the docs ive set up a second client and handler because the sse is in async mode ?

hmalpani, 2023-12-11:
@ThatBigPrint 
I think you should start the stream task using the workflow described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_client.html#http-stream)

I hope this helps. If not can you share a sample application with which I can reproduce this issue?

Thanks!

ThatBigPrint, 2023-12-11:
ive also tried this but i dont think its not putting...

`void sse_stream_task(void *pvParameters)
{
    start_config = true;
    char *buffer = malloc(MAX_HTTP_RECV_BUFFER + 1);
    if (buffer == NULL)
    {
        ESP_LOGE(TAG, "Cannot malloc http receive buffer");
        return;
    }

    char full_url[256];

    // snprintf(full_url, sizeof(full_url), "%s%s?auth=%s", FIREBASE_HOST, FIREBASE_PATH, FIREBASE_AUTH);
    snprintf(full_url, sizeof(full_url), "%s/%s%s.json?auth=%s", FIREBASE_HOST, rvd_data_str, FIREBASE_PATH, FIREBASE_AUTH);

    ESP_LOGI(TAG, "Connecting to Firebase URL: %s", full_url);

    esp_http_client_config_t config = {
        .url = full_url,
        .method = HTTP_METHOD_PUT,
        .crt_bundle_attach = esp_crt_bundle_attach,
        .event_handler = _http_event_handler,
        .is_async = true,
        .user_data = buffer,
    };
    client = esp_http_client_init(&config);

    esp_http_client_set_header(client, "Accept", "text/event-stream");

    esp_err_t err;
    while (1)
    {
        if (data_to_send)
        {
            char outbound_data[256]; // Increase the size of the buffer to accommodate the formatted string
            snprintf(outbound_data, sizeof(outbound_data), "{\"brightness\": \"%s\", \"keep_alive\": %d}", global_brightness, keep_alive);
            ESP_LOGI(TAG, "Outbound data: %s", outbound_data);

            esp_http_client_set_post_field(client, outbound_data, strlen(outbound_data));

            err = esp_http_client_perform(client);
            if (err == ESP_OK)
            {
                ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %" PRId64,
                         esp_http_client_get_status_code(client),
                         esp_http_client_get_content_length(client));
            }
            else
            {
                ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
            }

            data_to_send = false;
        }
        else
        {

            err = esp_http_client_perform(client);
            if (err != ESP_ERR_HTTP_EAGAIN)
            {
                break;
            }
        }
    }

    if (err == ESP_OK)
    {
        ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %" PRId64,
                 esp_http_client_get_status_code(client),
                 esp_http_client_get_content_length(client));
    }
    else
    {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);
}`

this is the log 

`I (12169) esp-x509-crt-bundle: Certificate validated
I (12964) Mesh/Firebase: DATA=event: put
data: {"path":"/","data":null}

I (21536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -60, free heap: 202400
I (21541) Mesh/Firebase: child node number: 0
I (31536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -58, free heap: 202400
I (31541) Mesh/Firebase: child node number: 0
I (41536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -57, free heap: 202400
I (41541) Mesh/Firebase: child node number: 0
I (43065) Mesh/Firebase: DATA=event: keep-alive
data: null

I (48068) Mesh/Firebase: Outbound data: {"brightness": "", "keep_alive": 1}
I (51536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -58, free heap: 202312
I (51541) Mesh/Firebase: child node number: 0
E (53070) Mesh/Firebase: Error perform http request ESP_ERR_HTTP_EAGAIN
I (61536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -57, free heap: 202312
I (61541) Mesh/Firebase: child node number: 0
I (71536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -58, free heap: 202312
I (71541) Mesh/Firebase: child node number: 0
I (72864) Mesh/Firebase: DATA=event: keep-alive
data: null

I (77867) Mesh/Firebase: Outbound data: {"brightness": "", "keep_alive": 2}
I (81536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -57, free heap: 202308
I (81541) Mesh/Firebase: child node number: 0
E (82869) Mesh/Firebase: Error perform http request ESP_ERR_HTTP_EAGAIN
I (91536) Mesh/Firebase: System information, channel: 1, layer: 1, self mac: 68:b6:b3:44:ee:fc, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -59, free heap: 202308
I (91541) Mesh/Firebase: child node number: 0`

this is not a stream reader?

hmalpani, 2023-12-11:
Hello @ThatBigPrint 
It would be helpful if you could share whole code which can be used to reproduce this.

ThatBigPrint, 2023-12-11:
`#include <inttypes.h>
#include <string.h>
#include <stdlib.h>
#include <sys/param.h>
#include <sys/socket.h>
#include <ctype.h>

#include "lwip/err.h"
#include "lwip/sys.h"

#include "nvs_flash.h"

#include "cJSON.h"

#include "esp_log.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_system.h"
#include "esp_netif.h"
#include "esp_smartconfig.h"
#include "esp_bridge.h"
#include "esp_mesh_lite.h"
#include "esp_netif_types.h"
#include "esp_event_base.h"
#include "esp_mac.h"
#include "esp_http_client.h"
#include "esp_crt_bundle.h"
#include "esp_tls.h"
#include "esp_bridge_wifi.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "freertos/event_groups.h"

#define MAX_HTTP_RECV_BUFFER 16384
#define MAX_HTTP_OUTPUT_BUFFER 16384
static esp_http_client_handle_t client;
static esp_http_client_handle_t outbound_client;

#define FIREBASE_HOST ""
#define FIREBASE_PATH ""
#define FIREBASE_AUTH ""

#define SMARTCONFIG_START_CONFIG() {   \
    .enable_log = false,               \
    .esp_touch_v2_enable_crypt = true, \
    .esp_touch_v2_key = ""};

static EventGroupHandle_t s_wifi_event_group;
const int WIFI_CONNECTED_BIT = BIT0;
const int BRIDGE_EVENT_STA_CONNECTED = BIT0;
static const int ESPTOUCH_DONE_BIT = BIT1;
static const char *TAG = "Mesh/Firebase";

static char *output_buffer;
static int output_len;

uint8_t ssid[33] = {0};
uint8_t password[65] = {0};
uint8_t rvd_data[65] = {0};

char ssid_str[34] = {0};
char password_str[66] = {0};
char rvd_data_str[66] = {0};

char global_brightness[100] = {0}; // Adjust the size as needed
int global_yoo = 0;

// char *ssid_str = NULL;
[... 546 lines cut ...]
void app_main()
{
    esp_log_level_set("*", ESP_LOG_INFO);

    esp_storage_init();

    initialise_wifi();

    if (wifi_config_done)
    {
        // wifi_config_t wifi_cfg;
        // esp_wifi_get_config(WIFI_IF_AP, &wifi_cfg);

        // wifi_cfg.ap.ssid_hidden = true;
        // wifi_cfg.ap.beacon_interval = 100;
        // esp_wifi_set_config(WIFI_IF_AP, &wifi_cfg);
        esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
        esp_mesh_lite_init(&mesh_lite_config);
        esp_mesh_lite_start();
        TimerHandle_t timer = xTimerCreate("print_system_info", 10000 / portTICK_PERIOD_MS,
                                           true, NULL, print_system_info_timercb);
        xTimerStart(timer, 0);
    }
}
`

ThatBigPrint, 2023-12-13:
Guys i still need help on this basicaly i want to have the incoming stream hapen and to be able to post out with the seperate function to the same endpoint do i need to interupt the client perform ? is this an issue with .is_async ?

KaeLL, 2023-12-13:
Would you please wrap this mess like this

````
```c
#include <stdio.h>

void app_main( void )
{
   printf( "Hello, World!" );
}
```
````
so it becomes
```c
#include <stdio.h>

void app_main( void )
{
   printf( "Hello, World!" );
}
```
?

ThatBigPrint, 2023-12-13:
@KaeLL mess resoved... would be pretty good if github could do that when you click the <>.... any help?

hmalpani, 2023-12-22:
Hello @ThatBigPrint 

This is my understanding of the problem you are facing. You want to use the same client handle for two purposes: first post data to the server and second read data from the server. I suppose these two are using POST and GET methods respectively. So I think, before calling esp_http_client_perform(), you need to set the method again. You can refer to the [esp_http_client_example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c#L142-L240)

ThatBigPrint, 2023-12-22:
No, i want the is_async to remain running for incoming get (that remains running continuously) and the random patch to be able to run every now and then with a cleanup to free resources but as soon as you call cleanup for the patch it disconnects the continuous incoming client as well

hmalpani, 2024-01-23:
If you are using the same client handle for both the requests, while cleaning up, you might have cleaned up the resources required for the GET request (that remains running continuously).
If you use two different handles for both the requests, does it works?

hmalpani, 2024-02-13:
Closing this issue. Please feel free to reopen if you have more updates. Thanks!

closed 2024-02-13 by hmalpani