espressif/esp-idf #12581: usb_host_install() abort at 0x4037d9df on core 0

TRYOKETHEPEN, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

My project create 3 freeRTOS tasks at app_main():
[task1] RC522 scann rf-id. 
[task2] LCD display & touch, with LVGL.
[task3] USB_HOST, which calls **usb_host_install()** that triggers this abortion.

My project file: 
[project1.zip](https://github.com/espressif/esp-idf/files/13345408/project1.zip)
task1  code based on https://github.com/abobija/esp-idf-rc522.
task2  code based on "spi_lcd_touch" example.
task3  code based on "hid" example.

When I build&flash&monitor, the expected monitor log should be:
_2023-11-14 14:04:32 I (902) main_task: Started on CPU0
2023-11-14 14:04:32 I (912) main_task: Calling app_main()
2023-11-14 14:04:32 W (912) 主程序: >>>开启RC522读卡任务(异步)
2023-11-14 14:04:32 I (922) rc522: versionReg: 0xb2
2023-11-14 14:04:32 I (922) rc522: Initialized (firmware v2.0)
2023-11-14 14:04:32 W (932) 主程序: >>>开启LCD
2023-11-14 14:04:32 I (932) lcd_touch_lvgl_init: 初始化SPI
2023-11-14 14:04:32 I (942) lcd_touch_lvgl_init: 初始化LCD panel
2023-11-14 14:04:32 I (942) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
2023-11-14 14:04:32 I (952) lcd_touch_lvgl_init: 安装panel驱动:ILI9341       
2023-11-14 14:04:32 I (962) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
2023-11-14 14:04:32 I (1092) lcd_touch_lvgl_init: 初始化LCD触控panel
2023-11-14 14:04:32 I (1092) lcd_touch_lvgl_init: 安装panel驱动:XPT2046      
2023-11-14 14:04:32 I (1092) lcd_touch_lvgl_init: 初始化LVGL绘制缓冲区       
2023-11-14 14:04:32 I (1092) lcd_touch_lvgl_init: 初始化&注册LVGL显示驱动器
2023-11-14 14:04:32 I (1102) lcd_touch_lvgl_init: 初始化LVGL定时器
2023-11-14 14:04:32 I (1112) lcd_touch_lvgl_init: 初始化LVGL触控驱动器       
2023-11-14 14:04:32 W (1112) 主程序: >>>开启USB HOST HID
2023-11-14 14:04:32 I (1122) usb_hid: 安装USB HOST驱动
2023-11-14 14:04:32 I (1122) usb_hid: 执行任务usb_events
2023-11-14 14:04:32 I (1162) usb_hid: 安装USB HOST驱动成功
2023-11-14 14:04:32 I (1162) usb_hid: 安装HID驱动
[... 150 lines cut ...]
2023-11-14 11:43:50 I (792) cpu_start: cpu freq: 240000000 Hz
2023-11-14 11:43:50 I (792) cpu_start: Application information:
2023-11-14 11:43:50 I (795) cpu_start: Project name:     project1
2023-11-14 11:43:50 I (800) cpu_start: App version:      1
2023-11-14 11:43:50 I (805) cpu_start: Compile time:     Nov 14 2023 11:33:412023-11-14 11:43:50 I (811) cpu_start: ELF file SHA256:  473c4fb0d95f68be... 
2023-11-14 11:43:50 I (817) cpu_start: ESP-IDF:          v5.1.1-dirty
2023-11-14 11:43:50 I (822) cpu_start: Min chip rev:     v0.0
2023-11-14 11:43:50 I (827) cpu_start: Max chip rev:     v0.99 
2023-11-14 11:43:50 I (832) cpu_start: Chip rev:         v0.1
2023-11-14 11:43:50 I (837) heap_init: Initializing. RAM available for dynamic allocation:
2023-11-14 11:43:50 I (844) heap_init: At 3FC9BCA8 len 0004DA68 (310 KiB): DRAM
2023-11-14 11:43:50 I (850) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
2023-11-14 11:43:50 I (857) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
2023-11-14 11:43:50 I (863) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
2023-11-14 11:43:51 I (869) spi_flash: detected chip: generic
2023-11-14 11:43:51 I (874) spi_flash: flash io: dio
2023-11-14 11:43:51 I (878) sleep: Configure to isolate all GPIO pins in sleep state
2023-11-14 11:43:51 I (884) sleep: Enable automatic switching of GPIO sleep configuration
2023-11-14 11:43:51 I (892) coexist: coex firmware version: 80b0d89
2023-11-14 11:43:51 I (897) coexist: coexist rom version e7ae62f
2023-11-14 11:43:51 I (902) app_start: Starting scheduler on CPU0
2023-11-14 11:43:51 I (907) app_start: Starting scheduler on CPU1
2023-11-14 11:43:51 I (907) main_task: Started on CPU0
2023-11-14 11:43:51 I (917) main_task: Calling app_main()
```

tore-espressif, 2023-11-16:
hi @TRYOKETHEPEN , thanks for the issue and log.

So far, the only case when we saw `usb_host_install` failing is because of failed interrupt allocation. Symptoms of your problem are the same. Here's a very good explanation of what is going on and how to debug it, from one of our colleagues https://github.com/espressif/esp-idf/issues/10903#issuecomment-1456176716

**TLDR:**
> When USB_HOST is set at CPU1, the project running well.

The easiest solution for you is to pin the USB task to CPU1

closed 2023-11-17 by TRYOKETHEPEN

acautomacaoecontrole, 2024-08-07:
I have experienced something similar to what was reported here.

A project that contains a webserver and a USB HID host was crashing at the very moment usb_host_install() was called. If webserver  code was removed, everything was ok. usb_host_install() was called from a task defined to run on CPU 0. After changing to CPU 1, problem is gone. Thanks to this issue registry, I could solve by now the problem, but it would be better to get a real solution for this. Hope you soon can do that.

DEBUG dump below:
-----------------------------------------------------------------------------------------------------------------------------
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4200a1af
0x4200a1af: usb_lib_task at D:/.../main/hid_host_example.c:923 (discriminator 1)

file: "./main/hid_host_example.c" line 923
func: usb_lib_task
expression: usb_host_install(&host_config)

abort() was called at PC 0x4037d0e7 on core 0
0x4037d0e7: _esp_error_check_failed at C:/Users/Admin/esp/v5.1.4/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x40375a8a:0x3fcb5ec0 0x4037d0f1:0x3fcb5ee0 0x403847ba:0x3fcb5f00 0x4037d0e7:0x3fcb5f70 0x4200a1af:0x3fcb5fa0 0x4037dc85:0x3fcb5fd0      
0x40375a8a: panic_abort at C:/Users/Admin/esp/v5.1.4/esp-idf/components/esp_system/panic.c:466
0x4037d0f1: esp_system_abort at C:/Users/Admin/esp/v5.1.4/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x403847ba: abort at C:/Users/Admin/esp/v5.1.4/esp-idf/components/newlib/abort.c:38
0x4037d0e7: _esp_error_check_failed at C:/Users/Admin/esp/v5.1.4/esp-idf/components/esp_system/esp_err.c:50
0x4200a1af: usb_lib_task at D:/.../main/hid_host_example.c:923 (discriminator 1)
0x4037dc85: vPortTaskWrapper at C:/Users/Admin/esp/v5.1.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 62590826

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037590c
0x4037590c: esp_restart_noos_dig at C:/Users/Admin/esp/v5.1.4/esp-idf/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e0c
entry 0x403c9908
-----------------------------------------------------------------------------------------------------------------------------