espressif/esp-idf #12582: PSRAM mapping fault?

i-e-b, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Espressif SoC revision.

Chip is ESP32D0WDQ5 (revision 3)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROVER-E (Lilygo T-SIM A7670E)

### Power Supply used.

External 3.3V

### What is the expected behavior?

I would expect mapped PSRAM to remain mapped

### What is the actual behavior?

PSRAM mapping seems to become corrupted intermittently. I have had this issue on multiple ESP32 modules, and with multiple revisions of ESP-IDF (master).

Before the issue triggers, `esp_get_free_heap_size();` will give about 4212k (on this module)
After the issue triggers, `esp_get_free_heap_size();` will give about 90k (just internal memory)

The results of these calls stay the same before and after the fault:
```
heap_caps_get_total_size(MALLOC_CAP_DEFAULT);
heap_caps_get_total_size(MALLOC_CAP_INTERNAL);
heap_caps_get_total_size(MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
```

Before the fault, calling `heap_caps_check_integrity_all(true)` returns true, and prints no output.
After the fault, the same call results in a panic:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40121593  PS      : 0x00060c30  A0      : 0x80121499  A1      : 0x3ffca440  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x007bdc24  A7      : 0x00000001  A8      : 0x00000803  A9      : 0x3ffb65e0  
A10     : 0x007be388  A11     : 0x3ffbe388  A12     : 0x00060021  A13     : 0x18000000  
A14     : 0x3ffc5c04  A15     : 0x3ffbdeac  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000f  LBEG    : 0x40082e21  LEND    : 0x40082e29  LCOUNT  : 0x00000027  

Backtrace: 0x40121590:0x3ffca440 0x40121496:0x3ffca470 0x400d336a:0x3ffca490 0x400d3384:0x3ffca4b0 0x400d787b:0x3ffca4d0 0x400d9159:0x3ffca500

0x40121590: tlsf_check at ~/esp/esp-idf/components/heap/tlsf/tlsf.c:687
0x40121496: multi_heap_check at ~/esp/esp-idf/components/heap/multi_heap.c:342
0x400d336a: heap_caps_check_integrity at ~/esp/esp-idf/components/heap/heap_caps.c:612
0x400d3384: heap_caps_check_integrity_all at ~/esp/esp-idf/components/heap/heap_caps.c:621
0x400d787b: --> my code, calling 
    if (!heap_caps_check_integrity_all(true)){
        ERR("Heap caps reports integrity errors")
    }
```

Before and after the fault, `esp_psram_is_initialized()` returns `true`;
After the fault, a call to `esp_psram_extram_add_to_heap_allocator();` returns a failure (-1)

### Steps to reproduce.

Run any program for a few days. Sometimes it fails after minutes, often not.

AxelLin, 2024-05-20:
@i-e-b Do you still hit this issue with recent esp-idf versions?

i-e-b, 2024-05-20:
My product needed to ship, so I turned PS-RAM off. It was still a problem with the most recent Git versions as of January 2024