espressif/esp-idf #12583: i80_controller example doesn't work.

So1aric, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board (Lilygo T-HMI)

### Power Supply used.

USB

### What is the expected behavior?

The example works.

### What is the actual behavior?

Nothing happened on the screen. It doesn't lit up.

### Steps to reproduce.

Git pull the 5.1.1 release.
Edit the pins.
Build and flash.

### Debug Logs.

```plain
--- esp-idf-monitor 1.3.3 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400490e9
0x400490e9: usb_uart_device_rx_one_char in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (27) boot: compile time Nov 15 2023 00:39:55
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=23348h (144200) map
I (126) esp_image: segment 1: paddr=00033370 vaddr=3fc91d00 size=02acch ( 10956) load
I (128) esp_image: segment 2: paddr=00035e44 vaddr=40374000 size=0a1d4h ( 41428) load
I (141) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=441e4h (279012) map
I (191) esp_image: segment 4: paddr=0008420c vaddr=4037e1d4 size=03a3ch ( 14908) load
I (201) boot: Loaded app from partition at offset 0x10000
I (201) boot: Disabling RNG early entropy source...
I (212) cpu_start: Multicore app
I (213) cpu_start: Pro cpu up.
I (213) cpu_start: Starting app cpu, entry point is 0x403752e4
0x403752e4: call_start_cpu1 at /home/so1aric/SDK/esp-idf-v5.1.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (231) cpu_start: Pro cpu start user code
I (231) cpu_start: cpu freq: 160000000 Hz
I (231) cpu_start: Application information:
I (234) cpu_start: Project name:     i80_controller
I (239) cpu_start: App version:      v5.1.1-dirty
I (245) cpu_start: Compile time:     Nov 15 2023 00:50:27
I (251) cpu_start: ELF file SHA256:  d968e3e36109b4da...
I (257) cpu_start: ESP-IDF:          v5.1.1-dirty
I (262) cpu_start: Min chip rev:     v0.0
I (267) cpu_start: Max chip rev:     v0.99
I (272) cpu_start: Chip rev:         v0.2
I (277) heap_init: Initializing. RAM available for dynamic allocation:
I (284) heap_init: At 3FC95530 len 000541E0 (336 KiB): DRAM
I (290) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (297) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (303) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (310) spi_flash: detected chip: winbond
I (314) spi_flash: flash io: dio
I (318) sleep: Configure to isolate all GPIO pins in sleep state
I (325) sleep: Enable automatic switching of GPIO sleep configuration
I (332) app_start: Starting scheduler on CPU0
I (337) app_start: Starting scheduler on CPU1
I (337) main_task: Started on CPU0
I (347) main_task: Calling app_main()
I (347) example: Turn off LCD backlight
I (357) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (367) example: Initialize Intel 8080 bus
I (367) example: Install LCD driver of st7789
I (497) example: Turn on LCD backlight
I (497) example: Initialize LVGL library
I (497) example: buf1@0x3fc9a8d8, buf2@0x3fca839c
I (497) example: Register display driver to LVGL
I (507) example: Install LVGL tick timer
I (507) example: Display LVGL animation
```
```

### More Information.

I tried with a simpler one, using i80 driver as the document told. It doesn't work either.
The board is not broken; Arduino with TFT_eSPI works.
I have checked the pins several times. And I have no idea what's wrong here.

suda-morris, 2023-11-14:
please attach your initialization code and the changes you made in the menuconfig.

So1aric, 2023-11-16:
Here is my `git diff` output:
```diff
diff --git a/examples/peripherals/lcd/i80_controller/main/Kconfig.projbuild b/examples/peripherals/lcd/i80_controller/main/Kconfig.projbuild
index 1edfd6df5f..9aae85c336 100644
--- a/examples/peripherals/lcd/i80_controller/main/Kconfig.projbuild
+++ b/examples/peripherals/lcd/i80_controller/main/Kconfig.projbuild
@@ -3,7 +3,7 @@ menu "Example Configuration"
     config EXAMPLE_LCD_I80_COLOR_IN_PSRAM
         bool "Allocate color data from PSRAM"
         depends on IDF_TARGET_ESP32S3
-        default y
+        default n
         help
             Enable this option if you wish to allocate the color buffer used by LVGL from PSRAM.
             Unmatched PSRAM band width with LCD requirement can lead to blurred image display.
diff --git a/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c b/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c
index 8c2090f4a9..bc6b9c3599 100644
--- a/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c
+++ b/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c
@@ -41,14 +41,14 @@ static const char *TAG = "example";
 
 #define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL  1
 #define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
-#define EXAMPLE_PIN_NUM_DATA0          6
-#define EXAMPLE_PIN_NUM_DATA1          7
-#define EXAMPLE_PIN_NUM_DATA2          8
-#define EXAMPLE_PIN_NUM_DATA3          9
-#define EXAMPLE_PIN_NUM_DATA4          10
-#define EXAMPLE_PIN_NUM_DATA5          11
-#define EXAMPLE_PIN_NUM_DATA6          12
-#define EXAMPLE_PIN_NUM_DATA7          13
+#define EXAMPLE_PIN_NUM_DATA0          48
+#define EXAMPLE_PIN_NUM_DATA1          47
+#define EXAMPLE_PIN_NUM_DATA2          39
+#define EXAMPLE_PIN_NUM_DATA3          40
+#define EXAMPLE_PIN_NUM_DATA4          41
+#define EXAMPLE_PIN_NUM_DATA5          42
+#define EXAMPLE_PIN_NUM_DATA6          45
+#define EXAMPLE_PIN_NUM_DATA7          46
 #if CONFIG_EXAMPLE_LCD_I80_BUS_WIDTH > 8
 #define EXAMPLE_PIN_NUM_DATA8          14
 #define EXAMPLE_PIN_NUM_DATA9          15
@@ -59,15 +59,15 @@ static const char *TAG = "example";
 #define EXAMPLE_PIN_NUM_DATA14         20
 #define EXAMPLE_PIN_NUM_DATA15         21
 #endif
-#define EXAMPLE_PIN_NUM_PCLK           5
-#define EXAMPLE_PIN_NUM_CS             3
-#define EXAMPLE_PIN_NUM_DC             4
-#define EXAMPLE_PIN_NUM_RST            2
-#define EXAMPLE_PIN_NUM_BK_LIGHT       1
+#define EXAMPLE_PIN_NUM_PCLK           8
+#define EXAMPLE_PIN_NUM_CS             6
+#define EXAMPLE_PIN_NUM_DC             7
+#define EXAMPLE_PIN_NUM_RST            -1
+#define EXAMPLE_PIN_NUM_BK_LIGHT       38
 
 // The pixel number in horizontal and vertical
-#define EXAMPLE_LCD_H_RES              240
-#define EXAMPLE_LCD_V_RES              280
+#define EXAMPLE_LCD_H_RES              280
+#define EXAMPLE_LCD_V_RES              320
 // Bit number used to represent command and parameter
 #if CONFIG_EXAMPLE_LCD_I80_CONTROLLER_ST7789
 #define EXAMPLE_LCD_CMD_BITS           8
```
I edit Kconfig because there's something wrong with the psram and I want to get rid of it.
For menuconfig, I only edit:
* Serial Flash Config: Flash Size 2 MB -> 16 MB
* Component ESP PSRAM: Disable
* Example Configuration: Disable PSRAM
And here is the board's pin layout.
[image]

So1aric, 2023-11-21:
@suda-morris Sorry to bother, but, am I doing something wrong?

crsrinidhi, 2023-12-04:
> @suda-morris Sorry to bother, but, am I doing something wrong?

I have the same display, and was facing the same issue. 
The display requires the PWR_EN (IO10) to be HIGH for the display to be ON. I added the below code and was able to get the example to work.

before app_main

`static void pwr_en(void)
{
    // Set the GPIO as a push/pull output
    gpio_set_direction(PWR_EN_PIN, GPIO_MODE_OUTPUT);
    gpio_set_level(PWR_EN_PIN, 1);
}`

in the app_main section on the first line added

`pwr_en();`

On compile and flash the example was displayed. 
Hope it helps

So1aric, 2023-12-11:
@crsrinidhi Thanks a lot! It does work.

closed 2023-12-11 by So1aric