espressif/esp-idf #12584: Error (ESP_ERR_NVS_NOT_FOUND) opening NVS handle! in encrypted nvs partition

hgwc, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I was trying to use pre-encrypted nvs flash & have initial data 
I don't know what's wrong and causing the error.
Please advise

using esp-idf v4.3.1-3-g954d44c7b9-dirty
Chip is ESP32-D0WD-V3 (revision 3)
SecureBoot & Flash Encryption is enabled

I executed it with the following procedure:

1. sdkconfig
CONFIG_NVS_ENCRYPTION=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y
CONFIG_SECURE_BOOT=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.2MB.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.2MB.csv"

2. CSV 파일
2.1 partitions.2MB.csv
Name,   Type, SubType, Offset,   Size,   Flags
stnv,     data, nvs,     ,        0x4000,
nvs,      data, nvs,     ,        0x4000,
info,data,nvs,,0x1000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
nvs_key,  data, nvs_keys, ,       0x1000, encrypted
ota_0,    app,  ota_0,   ,0x120000,
ota_1,    app,  ota_1,   ,0x120000,

2.2 stnv.csv
key,type,encoding,value
stdk,namespace,,
PublicKey,data,string,qZoy040gvbmDWbQmZELQc+pAWliAkljLUCVSI95l1/M=
PrivateKey,data,string,fszLdY2HUVL1Iv2QZ00KTo3iB1TUo887gSX+JX4Q/v4=

3. ESP-IDF Partition Table
Name, Type, SubType, Offset, Size, Flags
stnv,data,nvs,0x11000,16K,
nvs,data,nvs,0x15000,16K,
info,data,nvs,0x19000,4K,
otadata,data,ota,0x1a000,8K,
phy_init,data,phy,0x1c000,4K,
nvs_key,data,nvs_keys,0x1d000,4K,encrypted
ota_0,app,ota_0,0x20000,1152K,
ota_1,app,ota_1,0x140000,1152K,

4. Build & write Flash
1) python components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py encrypt stnv/stnv.csv  encrypted_nvs.bin 0x4000 --keygen --keyfile nvs_keys.bin
2) python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m --encrypt 0x10000 /build/partition_table/partition-table.bin 0x1a000 build/ota_data_initial.bin 0x20000 /build/SLBLDM-1000MOS-SCHOOL.bin 0x11000 build/keygen/encrypted_nvs.bin 0x1d0000 build/keygen/keys/nvs_keys.bin

5. program
void get_version_no(void)
{
    esp_err_t err;

    err = nvs_secure_initialize();
    if (err != ESP_OK) {
        ESP_LOGE("main", "Failed to initialize nvs (rc=0x%x). Halting.", err);
    }
    cert_get_data();
}

esp_err_t nvs_secure_initialize() {
    static const char *nvs_tag = "nvs";
    esp_err_t err = ESP_OK;
    // 1. find partition with nvs_keys
    const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
                                                                ESP_PARTITION_SUBTYPE_DATA_NVS_KEYS,
                                                                "nvs_key");
    if (partition == NULL) {
        ESP_LOGE(nvs_tag, "Could not locate nvs_key partition. Aborting.");
        return ESP_FAIL;
    }
    // 2. read nvs_keys from key partition
    nvs_sec_cfg_t cfg;
    if (ESP_OK != (err = nvs_flash_read_security_cfg(partition, &cfg))) {
        ESP_LOGE(nvs_tag, "Failed to read nvs keys (rc=0x%x)", err);
        return err;
    }
    // 3. initialize nvs partition
    if (ESP_OK != (err = nvs_flash_secure_init(&cfg))) {
        ESP_LOGE(nvs_tag, "failed to initialize nvs partition (err=0x%x). Aborting.", err);
        return err;
    };
    err = nvs_flash_secure_init_partition("stnv", &cfg);
    if (err != ESP_OK)
    {
        printf("failed to initialize stnv partition (err=0x%x). Aborting.", err);
    };
    return err;
}

static void cert_get_data()
{
    nvs_handle_t cert_handle;
   err = nvs_open_from_partition("stnv", "stdk", NVS_READONLY, &cert_handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG,"Error (%s) opening NVS handle!", esp_err_to_name(err));  **<-- An error occurred**
    }
}

hgwc, 2023-11-16:
The same problem has not been resolved yet.
Please advise.

laukik-hase, 2023-12-19:
Hello, @hgwc!

Thank you for the issue report. Sorry for not getting back to you sooner.

- Could you try reproducing this issue - 
  - on the ESP-IDF [`v4.3.6`]( https://github.com/espressif/esp-idf/tree/v4.3.6) tag
  - with security features (Flash Encryption and Secure Boot) disabled
- After running your example, could you dump and decrypt the `stnv` partition contents? You can use the `parttool` utility - `parttool.py read_partition -n stnv --output <output>.bin`. Also, decrypt the contents and verify them.

AxelLin, 2023-12-19:
> Hello, @hgwc!
> 
> Thank you for the issue report. Sorry for not getting back to you sooner.
> 
> * Could you try reproducing this issue -
>   
>   * on the ESP-IDF [`v4.3.6`](https://github.com/espressif/esp-idf/tree/v4.3.6) tag

Had better move to newer release branches, v4.3 branch is going to be EOL soon.

laukik-hase, 2024-01-15:
Hello, @hgwc!
Closing this issue due to lack of feedback, feel free to reopen with more updates. Thanks.

closed 2024-01-15 by laukik-hase