espressif/esp-idf #12585: "A fatal error occurred" error while flashing code to esp32s3

Abhay-2412, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello all of sudden iam getting below error while trying to flashing code

```
esptool.py v3.3.4-dev
Serial port COM13
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:42:dc:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800

A fatal error occurred: No serial data received.

 *  The terminal process "C:\Espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe 'C:\Espressif\frameworks\esp-idf-v4.4.5\components\esptool_py\esptool\esptool.py', '-p', 'COM13', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', 'detect', '0x0', 'bootloader/bootloader.bin', '0x10000', 'GAZZELE_MINI.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 2. 
```
i was able to flash my code this morning after that i did nothing and now iam getting above error while i trying to flash, and one more thing i observed when i restart my pc i was able to flash one time when i try to flash for 2nd time i get same error. and iam getting no issues while trying to flash code through arduino so i assume there is no hardware issue , please help me in debugging this

Additional information

- > iam using ESP-IDF of version 4.4.5 with VScode
-    operating system:windows
-    flash mode QIO
-    flash size 16mb
-    PSRAM: OPIPSRAM

dobairoland, 2023-11-14:
First of all, please try to flash it with the latest esptool and see if it helps. The simplest way is to use `esptool.exe` from https://github.com/espressif/esptool/releases/tag/untagged-09d18e9fc114822dd3cc and run it with the same arguments as above.

Abhay-2412, 2023-11-15:
i use vscode with ESP-IDF extension for building my code can you kindly mention how to implement your suggestion in that and one more thing i observed is its behaviour is very random some time iam able to uplaod flawlessly and sometime im unable to flash .

dobairoland, 2023-11-15:
Yes, I assumed that Arduino has a newer esptool than ESP-IDF v4.4. So no need to try my previous suggestion.

It could be a hardware issue (e.g. unstable power supply), or an operating system issue (e.g. driver). 

> OSError(22, 'A device which does not exist was specified.'

This error indicates that the connection between the operating system and the board was lost. Maybe the device disappeared for a second.

Abhay-2412, 2023-11-15:
While in the midst of uploading, I encountered a disconnection sound. After my last comment, I simply performed a restart. However, I'm facing the same issue again with the Arduino IDE. Oddly enough, when I encounter this problem, I can still upload code using ESP IDF, but my camera module isn't recognized. Strangely, after some time, when I can upload code through Arduino, the camera becomes recognizable. Notably, this issue is specific to ESP32-S3; I tested ESP32, and both Arduino and ESP IDF are functioning correctly with it.

Please suggest ways to troubleshoot it!

dobairoland, 2023-11-15:
I'm sorry but there are a lot of possible causes of this. You might try the same on a different computer, check if haven't received any recent upgrades and try to downgrade those drivers. On the hardware side, check the power source, possible short on the board.

But for sure this is not an ESP-IDF or esptool issue.

closed 2023-11-15 by dobairoland