espressif/esp-idf #12588: Error: app partition is too small for binary

stellna, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expected it to build properly. It was successful build the file before I add extra main.cpp and CMakeLists file into the main file. 
But right now, I encountered an error that I don't know how to solve. Sorry...

### What is the actual behavior?

[image]

### Steps to reproduce.

1. download code from my git website.
2. https://github.com/stellna/IOT-camera-learning
3. build the file

### Build or installation Logs.

```plain
[1/4] cmd.exe /C "cd /D C:\Users\whgPd\OneDrive\Deskt...sktop/esp-test/camera_test_4/build/camera_test_4.bin" 
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/build/esp-idf/esptool_py/CMakeFiles/app_check_size 
cmd.exe /C "cd /D C:\Users\whgPd\OneDrive\Desktop\esp-test\camera_test_4\build\esp-idf\esptool_py && C:\Users\whgPd\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/whgPd/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/build/partition_table/partition-table.bin C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/build/camera_test_4.bin"
Error: app partition is too small for binary camera_test_4.bin size 0x15a5b0:
  - Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0x5a5b0)
[2/4] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\whgPd\OneDrive\Desktop\esp-test\camera_test_4\build\bootloader\esp-idf\esptool_py && C:\Users\whgPd\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/whgPd/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/build/bootloader/bootloader.bin"
Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

sudeep-mohanty, 2023-11-15:
@stellna It looks like your application binary is larger than the app partition size in flash. You would have to adjust your partition table accordingly to fit your application on flash. Please check the [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html).

stellna, 2023-11-15:
It seem so. As I changed 2m to 4m it works. Thank you!

closed 2023-11-15 by stellna

uc-mani, 2024-11-24:
Create Custom partition table with the name partitions.csv in SDK config. Make sure remove the 'factory' row.
mine is like:
> nvs | data | nvs |   | 0x4000
> otadata | data | ota |   | 0x2000
> phy_init | data | phy |   | 0x1000
> ota_0 | app | ota_0 |   | 1984K
> ota_1 | app | ota_1 |   | 1984K