espressif/esp-idf #12590: Visual studio code extension installation bug

baldhead69, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am trying to update esp-idf version using visual studio code extension.
I did a clean installation, ie, deleted .espressif and esp-idf folder.

At the end of installation this error appear:
`Installing ESP-IDF extension python packages in C:\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe ...

O sistema n�o pode encontrar o caminho especificado.
Command failed: "C:\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\requirements.txt"
O sistema n�o pode encontrar o caminho especificado.
Command failed: "C:\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\requirements.txt"
O sistema n�o pode encontrar o caminho especificado.
` 
but at folder installation there in not  "C:\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe", only there is this path "C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\Scripts\python.exe".

brianignacio5, 2023-11-15:
Could you provide troubleshooting as specified [here](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) ?

brianignacio5, 2023-11-15:
For some reason you have conflicting python_env_path references. Not sure how but is probably due to some settings.json conflicts. Check your settings.json for existing `idf.pythonBinPath`

baldhead69, 2023-11-15:
@brianignacio5 ,

As soon as i have access to the computer that had the problem i will do this.

Should/can i delete the settings.json file every time i do a new installation ?

brianignacio5, 2023-11-16:
The thing is, if you update the esp-idf, new settings will be saved somewhere. Could be as global settings, workspace settings or workspace folder settings (project-specific). If old settings still remain then the project could break.

We are looking into ways to avoid and eliminate this problem in the extension. Sorry for the inconvenience

baldhead69, 2023-11-21:
Hi @brianignacio5,

I tried to install esp-idf v5.1.2 and when visual studio code extension was downloading esp-idf, the extension only downloaded the files but didnt created the folder and the extension returned to the home instalation screen.

So , i downloaded esp-idf v5.1.2 using command line with: ""

Error:
```
Installing ESP-IDF extension python packages in C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe ...

'"C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe"' n�o � reconhecido como um comando interno
ou externo, um programa oper�vel ou um arquivo em lotes.
Command failed: "C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\requirements.txt"
'"C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe"' n�o � reconhecido como um comando interno
ou externo, um programa oper�vel ou um arquivo em lotes.
Command failed: "C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\02373893983\.vscode\extensions\espressif.esp-idf-extension-1.6.5\requirements.txt"
'"C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe"' n�o � reconhecido como um comando interno
ou externo, um programa oper�vel ou um arquivo em lotes.
```

In my computer folder the path is this:
C:\.espressif\python_env\idf5.1_py3.11_env\Scripts\Scripts\Scripts\Scripts\python.exe

baldhead69, 2023-11-22:
????

packetkiller, 2023-11-23:
Same issue but already have ESP-IDF installed and working fine outside of VSC.  The ESP-IDF extension (using the advanced option) seems to pick up everything but ends with python version error (3.11.2). I have several versions of python installed on the workstation but for ESP-IDF I use the version it downloaded when it installed using the windows installer. This issue appears to be due to VSC using Windows path variables to id python version. This will definitely not go well for those with many versions running on same os.

baldhead69, 2023-11-23:
@packetkiller ,

Now worked.

steps:
- delete all environment variables related with esp-idf in windows.
- delete visual studio code folder intallation
- delete folder C:\Users\YourUserName\.vscode
- delete folder c:\\.espressif
- delete folder c:\esp-idf 
- restart computer
- install visual studio code ( i installed at c root partition and renamed, ie  C:\VSCode )
- install espressif visual studio code extension
- start esp-idf and tools installation at espressif visual studio code extension
- i used advanced installation and put c:\\.espressif and c:\esp-idf
- in my case the installation returning to vsc home screen after esp-idf download and not installed esp-idf at c:\esp-idf
- add "C:\.espressif\tools\idf-git\2.39.2\cmd" at environment variables of windows at Path in system variables, to git to work at cmd
- init cmd and select c root partition and type: "c:\git clone --depth 1 --recursive --branch v5.1.2 https://github.com/espressif/esp-idf.git

brianignacio5, 2023-11-27:
So what went wrong in the extension itself? 

Could you share the extension log found in :

Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

baldhead69, 2023-11-27:
> So what went wrong in the extension itself? 
> 
> Could you share the extension log found in :
> 
> Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
> Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

I dont know what went wrong with the extension.
I did the steps above and it worked.

I think these steps can help improve the extension and also help espressif team find the problem.

closed 2024-01-30 by Alvin1Zhang