espressif/esp-idf #12592: invalid conversion from 'int' to 'gpio_num_t'

stellna, 2023-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Project compiles successfully

### What is the actual behavior?

Error during the compilation

### Steps to reproduce.

1. This is my github pages.
    https://github.com/stellna/IOT-camera-learning
2. trying to take a c files from the main.cpp
[image]

3. building occurs the error.

### Build or installation Logs.

```plain
[1/10] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\whgPd\OneDrive\Desktop\esp-test\camera_test_4\build\bootloader\esp-idf\esptool_py && C:\Users\whgPd\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/whgPd/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/build/bootloader/bootloader.bin"
Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
[2/8] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj 
ccache C:\Users\whgPd\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/build/config -IC:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/main -IC:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/main/esp32/components -IC:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/main/esp32/main/include -IC:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/main/esp32/components/face-dl -IC:/Users/whgPd/OneDrive/Desktop/esp-test/camera_test_4/main/camera -IC:/Users/whgPd/esp/esp-idf/components/newlib/platform_include -IC:/Users/whgPd/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/whgPd/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/whgPd/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/whgPd/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/whgPd/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -IC:/Users/whgPd/esp/esp-idf/components/esp_hw_support/include -IC:/Users/whgPd/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/whgPd/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/Users/whgPd/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/whgPd/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -IC:/Users/whgPd/esp/esp-idf/components/heap/include -IC:/Users/whgPd/esp/esp-idf/components/log/include -IC:/Users/whgPd/esp/esp-idf/components/soc/include -IC:/Users/whgPd/esp/esp-idf/components/soc/esp32 -IC:/Users/whgPd/esp/esp-idf/components/soc/esp32/include -IC:/Users/whgPd/esp/esp-idf/components/hal/esp32/include -IC:/Users/whgPd/esp/esp-idf/components/hal/include -IC:/Users/whgPd/esp/esp-idf/components/hal/platform_port/include -IC:/Users/whgPd/esp/esp-idf/components/esp_rom/include -IC:/Users/whgPd/esp/esp-idf/components/esp_rom/include/esp32 -IC:/Users/whgPd/esp/esp-idf/components/esp_rom/esp32 -IC:/Users/whgPd/esp/esp-idf/components/esp_common/include -IC:/Users/whgPd/esp/esp-idf/components/esp_system/include -IC:/Users/whgPd/esp/esp-idf/components/esp_system/port/soc -IC:/Users/whgPd/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/whgPd/esp/esp-idf/components/xtensa/include -IC:/Users/whgPd/esp/esp-idf/components/xtensa/esp32/include -IC:/Users/whgPd/esp/esp-idf/components/lwip/include -IC:/Users/whgPd/esp/esp-idf/components/lwip/include/apps -IC:/Users/whgPd/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/whgPd/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/whgPd/esp/esp-idf/components/lwip/port/include -IC:/Users/whgPd/esp/esp-idf/components/lwip/port/freertos/include -IC:/Users/whgPd/esp/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/whgPd/esp/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/whgPd/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/whgPd/esp/esp-idf/components/efuse/include -IC:/Users/whgPd/esp/esp-idf/components/efuse/esp32/include -IC:/Users/whgPd/esp/esp-idf/components/driver/include -IC:/Users/whgPd/esp/esp-idf/components/driver/deprecated -IC:/Users/whgPd/esp/esp-idf/components/driver/analog_comparator/include -IC:/Users/whgPd/esp/esp-idf/components/driver/dac/include -IC:/Users/whgPd/esp/esp-idf/components/driver/gpio/include -IC:/Users/whgPd/esp/esp-idf/components/driver/gptimer/include -IC:/Users/whgPd/esp/esp-idf/components/driver/i2c/include -IC:/Users/whgPd/esp/esp-idf/components/driver/i2s/include -IC:/Users/whgPd/esp/esp-idf/components/driver/ledc/include -IC:/Users/whgPd/esp/esp-idf/components/driver/mcpwm/include -IC:/Users/whgPd/esp/esp-idf/components/driver/parlio/include -IC:/Users/whgPd/esp/esp-idf/components/driver/pcnt/include -IC:/Users/whgPd/esp/esp-idf/components/driver/rmt/include -IC:/Users/whgPd/esp/esp-idf/components/driver/sdio_slave/include -IC:/Users/whgPd/esp/esp-idf/components/driver/sdmmc/include -IC:/Users/whgPd/esp/esp-idf/components/driver/sigma_delta/include -IC:/Users/whgPd/esp/esp-idf/components/driver/spi/include -IC:/Users/whgPd/esp/esp-idf/components/driver/temperature_sensor/include -IC:/Users/whgPd/esp/esp-idf/components/driver/touch_sensor/include -IC:/Users/whgPd/esp/esp-idf/components/driver/twai/include -IC:/Users/whgPd/esp/esp-idf/components/driver/uart/include -IC:/Users/whgPd/esp/esp-idf/components/driver/usb_serial_jtag/include -IC:/Users/whgPd/esp/esp-idf/components/driver/touch_sensor/esp32/include -IC:/Users/whgPd/esp/esp-idf/components/esp_pm/include -IC:/Users/whgPd/esp/esp-idf/components/mbedtls/port/include -IC:/Users/whgPd/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/whgPd/esp/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/whgPd/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/whgPd/esp/esp-idf/components/esp_app_format/include -IC:/Users/whgPd/esp/esp-idf/components/bootloader_support/include -IC:/Users/whgPd/esp/esp-idf/components/bootloader_support/bootloader_flash/include -IC:/Users/whgPd/esp/esp-idf/components/esp_partition/include -IC:/Users/whgPd/esp/esp-idf/components/app_update/include -IC:/Users/whgPd/esp/esp-idf/components/esp_mm/include -IC:/Users/whgPd/esp/esp-idf/components/spi_flash/include -IC:/Users/whgPd/esp/esp-idf/components/pthread/include -IC:/Users/whgPd/esp/esp-idf/components/esp_timer/include -IC:/Users/whgPd/esp/esp-idf/components/app_trace/include -IC:/Users/whgPd/esp/esp-idf/components/esp_event/include -IC:/Users/whgPd/esp/esp-idf/components/nvs_flash/include -IC:/Users/whgPd/esp/esp-idf/components/esp_phy/include -IC:/Users/whgPd/esp/esp-idf/components/esp_phy/esp32/include -IC:/Users/whgPd/esp/esp-idf/components/vfs/include -IC:/Users/whgPd/esp/esp-idf/components/esp_netif/include -IC:/Users/whgPd/esp/esp-idf/components/wpa_supplicant/include -IC:/Users/whgPd/esp/esp-idf/components/wpa_supplicant/port/include -IC:/Users/whgPd/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -IC:/Users/whgPd/esp/esp-idf/components/esp_coex/include -IC:/Users/whgPd/esp/esp-idf/components/esp_wifi/include -IC:/Users/whgPd/esp/esp-idf/components/esp_wifi/wifi_apps/include -IC:/Users/whgPd/esp/esp-idf/components/unity/include -IC:/Users/whgPd/esp/esp-idf/components/unity/unity/src -IC:/Users/whgPd/esp/esp-idf/components/cmock/CMock/src -IC:/Users/whgPd/esp/esp-idf/components/console -IC:/Users/whgPd/esp/esp-idf/components/esp-dl/include -IC:/Users/whgPd/esp/esp-idf/components/esp-dl/include/tool -IC:/Users/whgPd/esp/esp-idf/components/esp-dl/include/typedef -IC:/Users/whgPd/esp/esp-idf/components/esp-dl/include/image -IC:/Users/whgPd/esp/esp-idf/components/esp-dl/include/math -IC:/Users/whgPd/esp/esp-idf/components/esp-dl/include/nn -I [... cut ...]
      |                        int
C:/Users/whgPd/esp/esp-idf/components/driver/gpio/include/driver/gpio.h:184:41: note:   initializing argument 1 of 'esp_err_t gpio_set_pull_mode(gpio_num_t, gpio_pull_mode_t)'
  184 | esp_err_t gpio_set_pull_mode(gpio_num_t gpio_num, gpio_pull_mode_t pull);
      |                              ~~~~~~~~~~~^~~~~~~~
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

nopnop2002, 2023-11-18:
gpio_set_pull_mode((gpio_num_t)4, GPIO_PULLUP_ONLY);

ScumCoder, 2024-02-13:
> Since you are using main.cpp, you need to do it like this.
> 
> gpio_set_pull_mode((gpio_num_t)4, GPIO_PULLUP_ONLY);
> 
> esp-idf supports C++, but it's a bit cumbersome.

No one should ever use C-style casts in C++.
If the function expects an enumeration value as an argument, you should give that function an enumeration value as an argument:
```
gpio_set_pull_mode(GPIO_NUM_4, GPIO_PULLUP_ONLY);
```

closed 2024-07-25 by suda-morris