espressif/esp-idf #12595: LEDC fade semaphore dead-lock on ESP32 with ESP-IDF 5.1

shreyuk23, 2023-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I expected it to continuously keep running and keep switching colour. 

### What is the actual behavior?

The code locks for esp32 after running the loop successfully for 3 time at `xSemaphoreTake(fade->ledc_fade_sem, portMAX_DELAY);` in `ledc.c` -> `ledc_set_fade_with_time` ->  `_ledc_fade_hw_acquire`
The issue doesn't occur when running for ESP32s3.
The code also runs fine for both esp32 and esp32s3 with v5.0.4 and master(v5.3-dev) but doesn't work for esp32 in v5.1.1 and v5.2-dev.

### Steps to reproduce.

Run the code snippet on IDF v5.1.1 for esp32.
The loop just gets stuck after few iterations when using `int target_duty[][3] = {{0, 255, 255}, {255, 0, 255}, {255, 255, 0}};` where as it works fine when using `int target_duty[][3] = {{255, 0, 0}, {0, 255, 0}, {0, 0, 255}}; `

```
/* LEDC (LED Controller) fade example*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "driver/ledc.h"
#include "esp_err.h"

#define LEDC_LS_TIMER          LEDC_TIMER_0
#define LEDC_LS_MODE           LEDC_LOW_SPEED_MODE
#define LEDC_LS_CH0_GPIO       (21)
#define LEDC_LS_CH0_CHANNEL    LEDC_CHANNEL_0
#define LEDC_LS_CH1_GPIO       (22)
#define LEDC_LS_CH1_CHANNEL    LEDC_CHANNEL_1
#define LEDC_LS_CH2_GPIO       (23)
#define LEDC_LS_CH2_CHANNEL    LEDC_CHANNEL_2
#define LEDC_TEST_CH_NUM       (3)
#define LEDC_TEST_FADE_TIME    (250)
#define INTR_FLAG_IRAM         ESP_INTR_FLAG_IRAM
#define INTR_FLAG_LOWMED       ESP_INTR_FLAG_LOWMED
#define INTR_FLAG_SHARED       ESP_INTR_FLAG_SHARED

void app_main(void)
{
    int ch;
    /*
     * Prepare and set configuration of timers
     * that will be used by LED Controller
     */
    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TIMER_8_BIT, // resolution of PWM duty
        .freq_hz = 5000,                     // frequency of PWM signal
        .speed_mode = LEDC_LS_MODE,          // timer mode
        .timer_num = LEDC_LS_TIMER,          // timer index
        .clk_cfg = LEDC_AUTO_CLK,            // Auto select the source clock
    };
    ledc_timer_config(&ledc_timer);
    /*
     * Prepare individual configuration for each  
     * channel of LED Controller by selecting:
     * - controller's channel number
     * - output duty cycle, set initially to 0
     * - GPIO number where LED is connected to
     * - speed mode, either high or low
     * - timer servicing selected channel
     */
    ledc_channel_config_t ledc_channel[LEDC_TEST_CH_NUM] = {
        {
            .channel    = LEDC_LS_CH0_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_LS_CH0_GPIO,
            .speed_mode = LEDC_LS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_LS_TIMER,
            .flags.output_invert = 0
        },
        {
            .channel    = LEDC_LS_CH1_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_LS_CH1_GPIO,
            .speed_mode = LEDC_LS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_LS_TIMER,
            .flags.output_invert = 0
        },
        {
            .channel    = LEDC_LS_CH2_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_LS_CH2_GPIO,
            .speed_mode = LEDC_LS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_LS_TIMER,
            .flags.output_invert = 0
        },
    };

    // Set LED Controller with previously prepared configuration
    for (ch = 0; ch < LEDC_TEST_CH_NUM; ch++) {
        ledc_channel_config(&ledc_channel[ch]);
    }

    // Initialize fade service.
    ledc_fade_func_install(INTR_FLAG_IRAM | INTR_FLAG_LOWMED | INTR_FLAG_SHARED);

    int target_duty[][3] = {{0, 255, 255}, {255, 0, 255}, {255, 255, 0}}; // this doesn't work
    // int target_duty[][3] = {{255, 0, 0}, {0, 255, 0}, {0, 0, 255}};         //    this Works

    while (1) {
        // SetColor(target_duty);
        for(int i=0; i<3; i++){
            ledc_set_fade_with_time(LEDC_LS_MODE, LEDC_LS_CH0_CHANNEL, target_duty[i][0], 250);
            ledc_set_fade_with_time(LEDC_LS_MODE, LEDC_LS_CH1_CHANNEL, target_duty[i][1], 250);
            ledc_set_fade_with_time(LEDC_LS_MODE, LEDC_LS_CH2_CHANNEL, target_duty[i][2], 250);

            // Call fade actions
            ledc_fade_start(LEDC_LS_MODE, LEDC_LS_CH0_CHANNEL, LEDC_FADE_NO_WAIT);
            ledc_fade_start(LEDC_LS_MODE, LEDC_LS_CH1_CHANNEL, LEDC_FADE_NO_WAIT);
            ledc_fade_start(LEDC_LS_MODE, LEDC_LS_CH2_CHANNEL, LEDC_FADE_NO_WAIT);

            // Wait for fade to end
            vTaskDelay(pdMS_TO_TICKS(250));
        }
    }
}
```

songruo, 2023-11-18:
Hi @shreyuk23, we recently fixed a related bug on master (821f7b04988d0721ff7dbb64af43d165e3124718),  the backports to the release versions are still on the way. I think this commit should also fix your problem, maybe you could try pick this commit (changes in `ledc.c` should be enough :)) .

shreyuk23, 2023-11-20:
Thanks, @songruo that fixed my problem. Will this issue be fixed in the v5.2 ?

songruo, 2023-11-20:
@shreyuk23 No, v5.2-dev is only a tag. We will publish release/v5.2 branch in the near future, the fix will be included there.

r3na, 2023-11-20:
@songruo  
but we definitely would need a 5.1.3 , before 5.2  :)

songruo, 2023-11-20:
@r3na I don't control the release schedule, either one of them will come first ;) Both will have the fix.

ZweiEuro, 2023-12-25:
@songruo is there a fix for this in 5.1.2 ? I tried applying ledc.c and .h  but am then gettign errors with other functions being undefined.

songruo, 2023-12-26:
@ZweiEuro The fix won't be in 5.1.2. You can wait for 5.1.3. Or you could cherry-pick [0329996](https://github.com/espressif/esp-idf/commit/03299962f67f1cb8f562749329f4b8fbf439b08d) to your local branch.

closed 2023-12-26 by songruo