espressif/esp-idf #12597: Does esp-dif really support wapi?

enric-lee, 2023-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

I fond esp-idf support wapi as bellow.
[image]

I expected it can scan wapi-psk ssid.

### What is the actual behavior?

but when I use scan example and add wapi psk item in print_auth_mode function, 
[image]

it print wep not wapi psk. 
[image]

If supports, please give me a example about scan and connect, thank you in advance.

### Steps to reproduce.

1. add wapi psk item in print_auth_mode function of wifi scan example.
2. build and flash to esp32-C3
3. reset and see the log of esp32C3.

enric-lee, 2023-11-20:
close

mhdong, 2023-11-22:
Hi @enric-lee 
please use idf.py menuconfig -> Component config -> WiFi -> Enable WAPI PSK support
open WAPI support.

enric-lee, 2023-11-27:
Thank you, I will try it.

closed 2023-11-27 by enric-lee