espressif/esp-idf #12598: Panic when installing SSD1306 panel driver with new I2C driver

tupyy, 2023-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'm using a little OLED display which is controlled via I2C. With ESP-IDF v5.2 a new I2C driver implementation has been added.
I would expect it to boot correctly. 

### What is the actual behavior?

This exception is raised at boot time `IntegerDivideByZero`

### Steps to reproduce.

This is my code to install the driver:
`
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ (100 * 1000)
#define EXAMPLE_PIN_NUM_SDA 21
#define EXAMPLE_PIN_NUM_SCL 22
#define EXAMPLE_PIN_NUM_RST -1
#define EXAMPLE_I2C_HW_ADDR 0x3C

// The pixel number in horizontal and vertical
#if CONFIG_EXAMPLE_LCD_CONTROLLER_SSD1306
#define EXAMPLE_LCD_H_RES 128
#define EXAMPLE_LCD_V_RES 64
#elif CONFIG_EXAMPLE_LCD_CONTROLLER_SH1107
#define EXAMPLE_LCD_H_RES 64
#define EXAMPLE_LCD_V_RES 128
#endif
// Bit number used to represent command and parameter
#define EXAMPLE_LCD_CMD_BITS 8
#define EXAMPLE_LCD_PARAM_BITS 8

extern void example_lvgl_demo_ui(lv_disp_t *disp);

/* The LVGL port component calls esp_lcd_panel_draw_bitmap API for send data to
the screen. There must be called lvgl_port_flush_ready(disp) after each
transaction to display. The best way is to use on_color_trans_done callback from
esp_lcd IO config structure. In IDF 5.1 and higher, it is solved inside LVGL
port component. */
static bool notify_lvgl_flush_ready(esp_lcd_panel_io_handle_t panel_io,
                                    esp_lcd_panel_io_event_data_t *edata,
                                    void *user_ctx) {
  lv_disp_t *disp = (lv_disp_t *)user_ctx;
  lvgl_port_flush_ready(disp);
  return false;
}

void app_main(void) {
  ESP_LOGI(TAG, "Initialize I2C bus");
[... 108 lines cut ...]
0x400ef3f4: i2c_hal_set_bus_timing at /home/cosmin/esp/esp-idf/components/hal/i2c_hal.c:28

0x400ec983: s_i2c_transaction_start at /home/cosmin/esp/esp-idf/components/driver/i2c/i2c_master.c:490 (discriminator 2)

0x400ecab3: s_i2c_synchronous_transaction at /home/cosmin/esp/esp-idf/components/driver/i2c/i2c_master.c:772

0x400ed3e8: i2c_master_transmit at /home/cosmin/esp/esp-idf/components/driver/i2c/i2c_master.c:982

0x400d66c7: panel_io_i2c_tx_buffer at /home/cosmin/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_io_i2c_v2.c:176

0x400d6739: panel_io_i2c_tx_param at /home/cosmin/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_io_i2c_v2.c:200

0x400d6556: esp_lcd_panel_io_tx_param at /home/cosmin/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_io.c:23 (discriminator 2)

0x400d6a32: panel_ssd1306_init at /home/cosmin/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_ssd1306.c:141

0x400d6e5d: esp_lcd_panel_init at /home/cosmin/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_ops.c:22 (discriminator 2)

0x400d5d49: app_main at /home/cosmin/esp/esp-idf/examples/peripherals/lcd/i2c_oled/main/i2c_oled_example_main.c:101 (discriminator 2)

0x400fcecb: main_task at /home/cosmin/esp/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x400868a1: vPortTaskWrapper at /home/cosmin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
`
```

suda-morris, 2023-11-17:
That's because you forgot set the SCL clock frequency:

https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test_apps/i2c_lcd/main/test_i2c_lcd_panel.c#L42

tupyy, 2023-11-17:
Thank you. This was the problem.

closed 2023-11-17 by tupyy