espressif/esp-idf #12601: How to fix Organic maps CMake android studio Error?

LaachirDeeper, 2023-11-16:
**The issue message as below:**

CMake Error at 3party/CMakeLists.txt:25 (add_subdirectory):
  The source directory
C:/Users/MyPC/Downloads/organicmaps-master/3party/jansson/jansson

  does not contain a CMakeLists.txt file.

igrr, 2023-11-16:
Please see https://github.com/organicmaps/organicmaps/issues/4724#issuecomment-1460542109 and https://github.com/organicmaps/organicmaps/issues/4724#issuecomment-1465034783, and for further help with that project please open issues in https://github.com/organicmaps/organicmaps/issues/.

closed 2023-11-16 by igrr

Yashbhosale00, 2024-05-12:
Hello LaachirDeeper, 
I'm encountering an issue while trying to build this project in Android Studio on my Linux system. Although all setups seem to be working correctly, I'm facing an error: "[CXX1429] error when building with CMake using /home/kalilinux/Desktop/organicmaps/CMakeLists.txt: -- The C compiler identification is Clang 17.0.2". Can you please guide me on how to fix this issue? I'd appreciate any assistance you can provide.

Yashbhosale00, 2024-05-12:
If anyone is familiar with this issue, could you please provide guidance on how to resolve it? Any assistance would be greatly appreciated.