espressif/esp-idf #12603: Improve wifi_scan_config_t documentation

llamasoft, 2023-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The documentation surrounding [wifi_scan_config_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/network/esp_wifi.html#_CPPv418wifi_scan_config_t) needs some additional clarification.

Specifically:

- The behavior of `.channel = 0` scanning all channels is undocumented.
- When scanning all channels, does it respect the current `wifi_country_t`?
- When using `.scan_type = WIFI_SCAN_TYPE_ACTIVE`:
  - How is the value of `.scan_time.passive` used?
  - Does probe request broadcast time count towards `.scan_time.active`?
  - How is the scan time determined using `.scan_time.active.min` and `.scan_time.active.max`?
- When using `.scan_type = WIFI_SCAN_TYPE_PASSIVE` is `.scan_time.active` ignored?
- What is the time unit of `.home_chan_dwell_time`?  Milliseconds?  FreeRTOS ticks?
- When `.channel = 0` and `.home_chan_dwell_time = 0`, does the scan visit the home channel between consecutive channel scans?

igrr, 2023-11-16:
> How is the scan time determined using .scan_time.active.min and .scan_time.active.max

Just a note, this one of the questions is answered in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#scan-configuration

mhdong, 2024-05-21:
Hi @llamasoft 

1. When scanning all channels, does it respect the current wifi_country_t?
Yes, the scanned channel must meet the requirements of the country code. For detailes, please refer to
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-country-code

closed 2024-06-19 by Sherry616