espressif/esp-idf #12604: iperf_run_tcp_client(361): Socket unable to connect: errno 113

arvind55555, 2023-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

In STA mode or AP the expected behaviour is that esp32, in client mode, should be able to send data packets with appropriate Mbits /sec to the computer.

### What is the actual behavior?

the only issue that i have been facing for a long time now is that i am unable to make the TCP throughput work. I tried every solution that i could get my hands on, but none of them works. i know that it should be something very very simple, that i missing out.

[image]

### Steps to reproduce.

1. Step 1:  got to ESP-IDF 5.1 CMD and go into examples/wifi/iperf
2. Step 2: once in iperf, run **idf.py build**
3. Step 3: Once that's over type **idf.py monitor -p <COM-PORT>**
4. step 4: Once the iperf tab opens, scan for your wifi router **scan <ssid>**
5. step 5: if your router got detected, connect it to iperf in sta mode, **sta <ssid> <pswrd>**
6. step 6. once connected, type iperf -c <ip>

### More Information.

the udp works for me irrespective of me connecting in sta mode or ap mode

[image]
[image]

hansw123, 2024-03-22:
hi @arvind55555 
Sorry for the late reply.
tcp is the protocol with connections，So you need two devices, a station and an ap, and then set up clinet and server on each, you can either connect the computer directly to the router, or use softap on the other board. You can refer to esp-idf/examples/wifi/iperf/README for details.

hansw123, 2024-03-22:
@arvind55555 
udp doesn't need a connection, it sends commands as soon as they are available, also your ip address looks like it is sent to yourself, it should be set to the ip address of another device under the same network

hansw123, 2024-03-22:
@arvind55555 
feel free to reopen

closed 2024-03-27 by Sherry616