espressif/esp-idf #12606: ESP32-C3 USB串口写64字节长度的问题

zdodo55, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

使用int usb_serial_jtag_write_bytes(const void* src, size_t size, TickType_t ticks_to_wait)函数发送任意长度数据主机应该能实时接收到

### What is the actual behavior?

结果发现使用int usb_serial_jtag_write_bytes(const void* src, size_t size, TickType_t ticks_to_wait)函数
在发送长度为64的数据时，主机无法接收到数据，
然后再发送非64长度的数据时，主机会一起收到两次发送的数据，
数据没丢，但是只写64字节的数据无法在主机上立即接收到

### Steps to reproduce.

```
/* Blink Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "led_strip.h"
#include "sdkconfig.h"
#include "driver/usb_serial_jtag.h"

static const char *TAG = "example";

/* Use project configuration menu (idf.py menuconfig) to choose the GPIO to blink,
   or you can edit the following line and set a number here.
*/
#define BLINK_GPIO CONFIG_BLINK_GPIO

static uint8_t s_led_state = 0;

#ifdef CONFIG_BLINK_LED_RMT

static led_strip_handle_t led_strip;

static void blink_led(void)
{
    /* If the addressable LED is enabled */
    if (s_led_state) {
        /* Set the LED pixel using RGB from 0 (0%) to 255 (100%) for each color */
        led_strip_set_pixel(led_strip, 0, 16, 16, 16);
        /* Refresh the strip to send data */
        led_strip_refresh(led_strip);
    } else {
        /* Set all LED off to clear all pixels */
        led_strip_clear(led_strip);
    }
}

static void configure_led(void)
{
    ESP_LOGI(TAG, "Example configured to blink addressable LED!");
    /* LED strip initialization with the GPIO and pixels number*/
    led_strip_config_t strip_config = {
        .strip_gpio_num = BLINK_GPIO,
        .max_leds = 1, // at least one LED on board
    };
    led_strip_rmt_config_t rmt_config = {
        .resolution_hz = 10 * 1000 * 1000, // 10MHz
    };
    ESP_ERROR_CHECK(led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip));
    /* Set all LED off to clear all pixels */
    led_strip_clear(led_strip);
}

#elif CONFIG_BLINK_LED_GPIO

static void blink_led(void)
{
    /* Set the GPIO level according to the state (LOW or HIGH)*/
    gpio_set_level(BLINK_GPIO, s_led_state);
}

static void configure_led(void)
{
    ESP_LOGI(TAG, "Example configured to blink GPIO LED!");
    gpio_reset_pin(BLINK_GPIO);
    /* Set the GPIO as a push/pull output */
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);
}

#endif

void usb_cdc_task(void *pvParameter)
{
    uint8_t *rxbuf;
    int sendcnt;
    rxbuf = (uint8_t *)malloc(256);
    int rxcnt;
    ESP_LOGI(TAG, "usb cdc task start!");
    while (1)
    {
        rxcnt = usb_serial_jtag_read_bytes((void *)rxbuf, 256, portMAX_DELAY); //阻塞当前任务，直到接收数据
        ESP_LOGI(TAG, "receive %d bytes", rxcnt);
        sendcnt = (int)usb_serial_jtag_write_bytes((const void *)rxbuf, rxcnt, 0);
        // ESP_LOGI(TAG, "send %d bytes", sendcnt);
    }
    free(rxbuf); //释放缓冲区
}

void usb_cdc_config(void)
{
    usb_serial_jtag_driver_config_t usb_cdc = USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT();
    TaskHandle_t usb_cdc_task_handler;
    ESP_ERROR_CHECK(usb_serial_jtag_driver_install(&usb_cdc)); //安装usb_serial驱动
    xTaskCreate(usb_cdc_task, "usb_cdc_task", 1024 * 5, NULL, 1, &usb_cdc_task_handler);
}

void app_main(void)
{

    /* Configure the peripheral according to the LED type */
    configure_led();
    usb_cdc_config();

    while (1) {
        ESP_LOGI(TAG, "Turning the LED %s!", s_led_state == true ? "ON" : "OFF");
        blink_led();
        /* Toggle the LED state */
        s_led_state = !s_led_state;
        vTaskDelay(CONFIG_BLINK_PERIOD / portTICK_PERIOD_MS);
    }
}

```

zhoumiaomiao, 2024-06-19:
这个问题在ESP-IDF-V5.2.1版本里面还是没有修复嘛？

tore-espressif, 2026-02-16:
Closing for inactivity, feel free to reopen if the problem persists after v5.2.1