espressif/esp-idf #12607: Can't find a nvs partition namespace

hgwc, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello
I was trying to use pre-encrypted nvs flash & inject initial data
I don't know what's wrong and causing the error.

Please advise

using esp-idf v4.3.1-3-g954d44c7b9-dirty
Chip is ESP32-D0WD-V3 (revision 3)
SecureBoot & Flash Encryption is enabled

print log
I (1577) nvs: nvs_key partition found
I (1587) nvs: stnv partition found
I (1587) nvs: Success to read nvs keys 
I (1597) nvs: Success to secure_init_partition 
Count: UsedEntries = (0), FreeEntries = (504),  AllEntries = (504),  NameSpaces = (0)
E (1607) nvs: nvs_open_from_partition Error (ESP_ERR_NVS_NOT_FOUND) opening NVS handle!

sdkconfig
CONFIG_NVS_ENCRYPTION=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y
CONFIG_SECURE_BOOT=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.2MB.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.2MB.csv"

esp_err_t nvs_secure_initialize() {
    static const char *nvs_tag = "nvs";
    esp_err_t err = ESP_OK;

    // 1. find partition with nvs_keys
    const esp_partition_t *key_partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
                                                                ESP_PARTITION_SUBTYPE_DATA_NVS_KEYS,
                                                                "nvs_key");
    if (key_partition == NULL) {
        ESP_LOGE(nvs_tag, "ERROR: nvs_key partition not found");
        return ESP_FAIL;
    }
    else {
        ESP_LOGI(nvs_tag, "nvs_key partition found");
    }
   const esp_partition_t *stnv_partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
                                                                ESP_PARTITION_SUBTYPE_DATA_NVS,
                                                                "stnv");
    if (stnv_partition == NULL) {
        ESP_LOGE(nvs_tag, "ERROR: stnv partition not found");
    }
    else{
        ESP_LOGI(nvs_tag, "stnv partition found");
    }
    // 2. read nvs_keys from key partition
    nvs_sec_cfg_t cfg;
    if (ESP_OK != (err = nvs_flash_read_security_cfg(key_partition, &cfg))) {
        ESP_LOGE(nvs_tag, "Failed to read nvs keys (rc=0x%x)", err);
        return err;
    }
    else{
        ESP_LOGI(nvs_tag, "Success to read nvs keys ");        
    }
    // 3. initialize nvs partition
    if (ESP_OK != (err = nvs_flash_secure_init(&cfg))) {
        ESP_LOGE(nvs_tag, "failed to initialize nvs partition (err=0x%x). Aborting.", err);
        return err;
    }

    err = nvs_flash_secure_init_partition("stnv", &cfg);
    if (err != ESP_OK)
    {
        ESP_LOGE(nvs_tag, "failed to initialize stnv partition (err=0x%x). Aborting.", err);
    }
    else{
        ESP_LOGI(nvs_tag, "Success to secure_init_partition ");        
    }
nvs_stats_t nvs_stats;
nvs_get_stats("stnv", &nvs_stats);
printf("Count: UsedEntries = (%u), FreeEntries = (%u),  AllEntries = (%u),  NameSpaces = (%u)\n",
       nvs_stats.used_entries, nvs_stats.free_entries, nvs_stats.total_entries, nvs_stats.namespace_count);
 
    nvs_handle_t cert_handle;
    //err = nvs_open("aaa", NVS_READONLY, &cert_handle);
    err = nvs_open_from_partition("stnv", "stdk", NVS_READONLY, &cert_handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(nvs_tag,"nvs_open_from_partition Error (%s) opening NVS handle!", esp_err_to_name(err));
    }
    else{
        ESP_LOGI(nvs_tag, "Success to nvs_open_from_partition ");        
    }
   size_t value_size;
    err = nvs_get_str(cert_handle, "PublicKey", NULL,  &value_size) ;
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG,"nvs_get_str(%s) ", esp_err_to_name(err));
        return 0;
    }
    char *value = malloc(value_size);
    if (nvs_get_str(cert_handle, "Publickey", *value, &value_size) != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to load key:");
        return 0;
    }
    server_cert_len = cert_get_str(cert_handle, "PublicKey", &server_cert);
    if (server_cert == NULL || server_cert_len==0)
    {
        ESP_LOGE(TAG,"Error! Unable to read server certificate");
    }
    priv_key_len = cert_get_str(cert_handle, "PrivateKey", &priv_key);
    if (priv_key == NULL || priv_key_len == 0)
    {
        ESP_LOGE(TAG,"Error, Unable to read private key");
    }
    return err;
}

Alvin1Zhang, 2023-11-17:
Thanks for reporting.

hgwc, 2023-11-17:
my csv file
1 partitions.2MB.csv
Name, Type, SubType, Offset, Size, Flags
stnv, data, nvs, , 0x4000,
nvs, data, nvs, , 0x4000,
info,data,nvs,,0x1000,
otadata, data, ota, , 0x2000,
phy_init, data, phy, , 0x1000,
nvs_key, data, nvs_keys, , 0x1000, encrypted
ota_0, app, ota_0, ,0x120000,
ota_1, app, ota_1, ,0x120000,

2 stnv.csv
key,type,encoding,value
stdk,namespace,,
PublicKey,data,string,qZoy040gvbmDWbQmZELQc+pAWliAkljLUCVSI95l1/M=
PrivateKey,data,string,fszLdY2HUVL1Iv2QZ00KTo3iB1TUo887gSX+JX4Q/v4=

rrtandler, 2024-09-12:
@hgwc