espressif/esp-idf #12608: esp_intr_free(): core could change after call to esp_cpu_get_core_id()

gralpli, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The function esp_intr_free() in components/esp_hw_support/intr_alloc.c on line 703 does this:

```c++
//Assign this routine to the core where this interrupt is allocated on.
if (handle->vector_desc->cpu != esp_cpu_get_core_id()) {
    esp_err_t ret = esp_ipc_call_blocking(handle->vector_desc->cpu, &esp_intr_free_cb, (void *)handle);
    return ret == ESP_OK ? ESP_OK : ESP_FAIL;
}
```

If this routine is run on a task that has no core affinity, the scheduler could move it to the other core after esp_cpu_get_core_id() got called. The call to esp_ipc_call_blocking() should therefore be made unconditionally, but at least in all cases in which the current task has no affinity.

ESP-Marius, 2023-11-17:
Thanks for reporting this, this indeed seems like a potential issue.

We'll investigate it.

KonstantinKondrashov, 2023-11-22:
Hi @gralpli!
Thanks for reporting it. I created a fix for it. I am wondering how this issue affects the system. Did you get an error message?
This would be useful to share an error log (or incorrect behavior that you observed) for people who encounter the same problem.

gralpli, 2023-11-22:
I did not get an error message, I spotted it by reading the source code.

closed 2024-02-28 by espressif-bot