espressif/esp-idf #12609: Error with Task watchdog got triggered

stellna, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I'm Trying to build my own esp32 project for camera deeplearning. I'm trying to run this file but I got a following Error code. I'm uploading my code line. As I understand the problem should have solved If I add delay line.

Any advice to fix this code would be appreciate. Thank you!

[image]

E (6449) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6449) task_wdt:  - IDLE (CPU 0)
E (6449) task_wdt: Tasks currently running:
E (6449) task_wdt: CPU 0: main
E (6449) task_wdt: CPU 1: IDLE
E (6449) task_wdt: Print CPU 0 (current core) backtrace

ESP-Marius, 2023-11-17:
Seems like whatever you are doing in the `test_decode_task` or `test_deep_learning` is running for too long without letting the idle task run. See more info [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer-twdt)

Cant see from your picture, but the backtrack in yellow should tell you where/which function was running the WDT triggered.

ESP-Marius, 2023-11-22:
Closing this as the issue does not seem to be a bug/feature request for esp-idf

For generic how-to questions I recommend also trying our forums [esp32.com](https://www.esp32.com/)

closed 2023-11-22 by ESP-Marius