espressif/esp-idf #12610: ot_rcp causes otbr-agent crash

thedevleon, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.1.1

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1-N8

### Power Supply used.

External 3.3V

### What is the expected behavior?

See also https://github.com/openthread/ot-br-posix/issues/2085. We are using ot_rcp together with otbr-posix and a Raspberry Pi 4 to build a Thread Border Router. 

We expect this setup to be stable and work. However, after around 3-7 days of uptime, the C6/ot_rcp seems to send a corrupted frame, which the otbr-agent failes to parse and ultimately crashes the otbr-agent, see:

```
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.455 [D] P-RadioSpinel-: Received spinel frame, flg:0x2, tid:0, cmd:PROP_VALUE_IS, key:STREAM_RAW, len:123, rssi:-81 ...
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.455 [D] P-RadioSpinel-: ... noise:-128, flags:0x0000, channel:16, lqi:9, timestamp:938961714397, rxerr:0
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.459 [D] P-RadioSpinel-: Received spinel frame, flg:0x2, tid:0, cmd:PROP_VALUE_IS, key:STREAM_RAW, len:49, rssi:-80 ...
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.459 [D] P-RadioSpinel-: ... noise:-128, flags:0x0000, channel:16, lqi:11, timestamp:938961720156, rxerr:0
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.459 [I] MeshForwarder-: Received IPv6 UDP msg, len:135, chksum:d068, ecn:no, from:0xd400, sec:yes, prio:normal, rss:-80.5
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.460 [I] MeshForwarder-:     src:[fd3a:c703:91ab:1:9ca8:2073:403a:f4bd]:34962
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.460 [I] MeshForwarder-:     dst:[fd3a:c703:91ab:2:0:0:34f:5ca3]:5683
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.466 [D] P-RadioSpinel-: Received spinel frame, flg:0x2, tid:9, cmd:PROP_VALUE_IS, key:LAST_STATUS, status:OK
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.466 [W] P-RadioSpinel-: Handle radio frame failed: Parse
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.466 [W] P-RadioSpinel-: Handle transmit done failed: Parse
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.467 [C] Platform------: ------------------ BEGINNING OF CRASH -------------
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.467 [C] Platform------: *** FATAL ERROR: Caught signal: 6 (Aborted)
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.469 [C] Platform------: # 0: /usr/sbin/otbr-agent(+0x2bc6e0) [0x555b07c6e0]
Nov  6 11:37:29 128f62455490 otbr-agent[200]: 2d.22:10:21.469 [C] Platform------: ------------------ END OF CRASH ------------------
```

As menionted in the linked issue
> So the root cause is that the Thread host receives a wrong Spinel frame from the esp32-c6.

### What is the actual behavior?

The C6 doesn´t send corrupted / invalid frames to otbr-agent.

### Steps to reproduce.

1. Flash `ot_rcp` to a C6
2. Hookup C6 via UART (i.e. via the UART USB on the devkit)
3. Start a docker container with OTBR 
```

zwx1995esp, 2023-11-22:
Hi @thedevleon Do you have some logs from ESP32C6 side? If not, I think I need to reproduce this issue on my local ENV. So, in your network topology, you only create some UDP sockets and transport massive datas via BR right?  I can believe the BR under the throughput test, can I?

thedevleon, 2023-11-22:
> Hi @thedevleon Do you have some logs from ESP32C6 side? If not, I think I need to reproduce this issue on my local ENV. So, in your network topology, you only create some UDP sockets and transport massive datas via BR right? I can believe the BR under the throughput test, can I?

Is there a way to get logs from the ESP32C6 side? AFAIK, the main UART is used for the RCP communication. But is it maybe possible to setup a second UART via some GPIO pins or the native USB/JTAG that would give me logging output?

In regards to setting up a reproducable environment, I will try to see if I can setup something for you.

zwx1995esp, 2023-11-22:
> Is there a way to get logs from the ESP32C6 side? AFAIK, the main UART is used for the RCP communication. But is it maybe possible to setup a second UART via some GPIO pins or the native USB/JTAG [...]

Hi, For logging on ESP32C6, you need a `USB to UART` module, something like this:
[image]

and a convenient way to enable an other uart is that, you can modify the file `ot_rcp/main/esp_ot_config.h`. Line 40: .port = 0, -> .port = 1,
and then set the UART pin using `iuudf.py menuconfig`
```
(Top) → OpenThread RCP Example 

[*] Configure RCP UART pin manually
```
like this:
[image]
And connect UART1 to ot-br-posix, use UART0 to print logs. Like this(One end of USB cable is connected at UART0(UART port on the ESP32C6 board), and the other end is connected to the host which is used for collecting logs; A `USB to UART` module is connect to ot-bt-posix from ESP32C6 UART1 via jumper wires):
[image]

zwx1995esp, 2023-11-22:
Also could you please enable IEEE802154 debug feature? You can enable it via `idf.py menuconfig`

[image]

thedevleon, 2023-11-22:
Thank you @zwx1995esp for the instructions, I will report back once I get a useful crash log for you.

zwx1995esp, 2023-12-26:
Hi @thedevleon any feedbacks? Do you have some process or you have already fix this issue?

chshu, 2024-03-01:
@thedevleon Please reopen if any follow up questions.

closed 2024-03-01 by chshu