espressif/esp-idf #12612: Oneshot ADC reading 500mV lower with automatic light sleep enabled

JVKran, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-S3 (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board with ESP32-S3-M0N16R8

### Power Supply used.

Battery

### What is the expected behavior?

I expect samples taken by the ADC using the oneshot API to be the same regardless of wether the automatic light-sleep feature is enabled.

### What is the actual behavior?

Samples taken with automatic light sleep enabled are approximately 500mV lower than those taken with automatic light sleep off.

### Steps to reproduce.

1. Compile this attached [modified_example](https://github.com/espressif/esp-idf/files/13391411/modified_example.zip).
2. Take note of a few of the samples.
3. Comment lines number 37 and 38 so power-management is not configured.
4. Observe that the samples (apart from the first one) are aproximately 500mV higher.

### Debug Logs.

```plain
### With automatic light sleep

I (350) EXAMPLE: calibration scheme version is Curve Fitting
I (350) EXAMPLE: Calibration Success
I (360) EXAMPLE: ADC1 Channel[5] Raw Data: 2227
I (360) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1818 mV
I (1370) EXAMPLE: ADC1 Channel[5] Raw Data: 1598
I (1370) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1310 mV
I (2370) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (2370) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (3370) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (3370) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (4370) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (4370) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (5380) EXAMPLE: ADC1 Channel[5] Raw Data: 1597
I (5380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1309 mV
I (6380) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (6380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (7380) EXAMPLE: ADC1 Channel[5] Raw Data: 1598
I (7380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1310 mV
I (8380) EXAMPLE: ADC1 Channel[5] Raw Data: 1598
I (8380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1310 mV
I (9380) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (9380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (10390) EXAMPLE: ADC1 Channel[5] Raw Data: 1598
I (10390) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1310 mV
I (11390) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (11390) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (12390) EXAMPLE: ADC1 Channel[5] Raw Data: 1598
I (12390) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1310 mV
I (13390) EXAMPLE: ADC1 Channel[5] Raw Data: 1598
I (13390) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1310 mV
I (14390) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (14390) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (15400) EXAMPLE: ADC1 Channel[5] Raw Data: 1597
I (15400) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1309 mV
I (16400) EXAMPLE: ADC1 Channel[5] Raw Data: 1599
I (16400) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1311 mV
I (17400) EXAMPLE: ADC1 Channel[5] Raw Data: 1597
I (17400) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1309 mV
```

### Without automatic light sleep
```
I (319) main_task: Calling app_main()
I (319) EXAMPLE: calibration scheme version is Curve Fitting
I (329) EXAMPLE: Calibration Success
I (329) EXAMPLE: ADC1 Channel[5] Raw Data: 2230
I (339) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1820 mV
I (1349) EXAMPLE: ADC1 Channel[5] Raw Data: 2251
I (1349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1837 mV
I (2349) EXAMPLE: ADC1 Channel[5] Raw Data: 2252
I (2349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1838 mV
I (3349) EXAMPLE: ADC1 Channel[5] Raw Data: 2249
I (3349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1837 mV
I (4349) EXAMPLE: ADC1 Channel[5] Raw Data: 2251
I (4349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1837 mV
I (5349) EXAMPLE: ADC1 Channel[5] Raw Data: 2249
I (5349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1837 mV
I (6349) EXAMPLE: ADC1 Channel[5] Raw Data: 2249
I (6349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1837 mV
I (7349) EXAMPLE: ADC1 Channel[5] Raw Data: 2251
I (7349) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1837 mV
```
```

### More Information.

I've taken the oneshot example as basis for this test, removed all that's not needed and added the power-management call.

Icarus113, 2023-11-23:
Weird, didn't reproduce from you uploaded program. I enabled `CONFIG_PM_ENABLE` and `CONFIG_FREERTOS_USE_TICKLESS_IDLE`.

By set or unset the `.light_sleep_enable` field, the results are:
```
I (374) EXAMPLE: ADC1 Channel[5] Raw Data: 1738
I (374) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
in sleep
out sleep
I (1384) EXAMPLE: ADC1 Channel[5] Raw Data: 1738
I (1384) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
in sleep
out sleep
```
and 
```
I (364) EXAMPLE: ADC1 Channel[5] Raw Data: 1737
I (364) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
I (1364) EXAMPLE: ADC1 Channel[5] Raw Data: 1739
I (1364) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1412 mV
I (2364) EXAMPLE: ADC1 Channel[5] Raw Data: 1739
I (2364) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1412 mV
```
I'm on `482a8fb2d7 (tag: v5.1.2, origin/test/coex_test_v5.1.2_20231113_karl) change(version): Update version to 5.1.2`

BTW I recall we fixed such issue on previous IDF versions so there's a test dedicated for this aim. 

Let me know the updates to see if there's any regression happens.

JVKran, 2023-11-23:
Hi @Icarus113,

I can't seem to find the test you're running [(this isn't the one) ](https://github.com/espressif/esp-idf/blob/release/v5.1/components/esp_adc/test_apps/adc/main/test_adc.c) and I can't really figure out from the logs if you're taking more than one single sample after leaving light-sleep.

If not, that corresponds to my logs. The first sample after leaving light sleep is always right, but the samples following after that aren't.

I'll update the example to run on an ESP32-S3-DevkitC-1 and automatically toggle between automatic light sleep or not.

Kind regards,

Jochem

JVKran, 2023-11-23:
Hi @Icarus113,

[See the newly modified example here.](https://github.com/espressif/esp-idf/files/13447332/auto_switch_example.zip) It automatically toggles automatic light sleep. This demonstrates the following things;

1. As long as automatic light sleep has never been enabled since boot, the one-shot ADC will return correct values.
2. As soon as automatic light sleep is enabled, one more sample will be correct.
3. After this one single sample, the one-shot ADC will never return a correct sample again. Even when automatic light sleep is disabled again, the samples remain wrong.

See the logs underneath.

Kind regards,

Jochem

```
I (330) main_task: Calling app_main()
I (330) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: DISABLED
I (340) EXAMPLE: calibration scheme version is Curve Fitting
I (340) EXAMPLE: Calibration Success
W (350) EXAMPLE: Automatic light sleep DISABLED!
I (350) EXAMPLE: ADC1 Channel[5] Raw Data: 2411
I (360) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1967 mV                                          // Good!
I (1360) EXAMPLE: ADC1 Channel[5] Raw Data: 2419
I (1360) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1973 mV                                        // Good!
I (2360) EXAMPLE: ADC1 Channel[5] Raw Data: 2420
I (2360) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1973 mV                                        // Good!
I (3360) EXAMPLE: ADC1 Channel[5] Raw Data: 2421
I (3360) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1974 mV                                        // Good!
I (4360) EXAMPLE: ADC1 Channel[5] Raw Data: 2416
I (4360) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1970 mV                                        // Good!
I (5360) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (5360) sleep: Code start at 0x42000020, total 114039, data start at 0x3c020020, total 47412 Bytes
0x42000020: _stext at ??:?

W (5360) EXAMPLE: Automatic light sleep ENABLED!
I (5370) EXAMPLE: ADC1 Channel[5] Raw Data: 2415
I (5370) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1969 mV                                        // Last good sample ever until reboot.
I (6380) EXAMPLE: ADC1 Channel[5] Raw Data: 1747
I (6380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1430 mV                                        // First wrong sample
I (7380) EXAMPLE: ADC1 Channel[5] Raw Data: 1747
I (7380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1430 mV                                        // Wrong!
I (8380) EXAMPLE: ADC1 Channel[5] Raw Data: 1749
I (8380) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1432 mV                                        // Wrong!
I (9390) EXAMPLE: ADC1 Channel[5] Raw Data: 1747
I (9390) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1430 mV                                        // Wrong!
I (10390) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: DISABLED
W (10390) EXAMPLE: Automatic light sleep DISABLED!
I (10390) EXAMPLE: ADC1 Channel[5] Raw Data: 1746
I (10400) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1429 mV                                        // Wrong!
I (11410) EXAMPLE: ADC1 Channel[5] Raw Data: 1747
I (11410) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1430 mV                                        // Wrong!
I (12410) EXAMPLE: ADC1 Channel[5] Raw Data: 1747
I (12410) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1430 mV                                        // Wrong!
I (13410) EXAMPLE: ADC1 Channel[5] Raw Data: 1748
I (13410) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1431 mV                                        // Wrong!
```

JVKran, 2023-11-23:
Just one more piece of information; when running this example on the ESP32-S3-DevkitC-1, the drop is even more drastic. It drops from 1970mV to around 1200mV.

Icarus113, 2023-11-29:
I'm still on `482a8fb2d7 (tag: v5.1.2, origin/test/coex_test_v5.1.2_20231113_karl) change(version): Update version to 5.1.2` and using the newer uploaded example, with the sdkconfig in your .zip.

The results are still stable.

```
I (326) main_task: Started on CPU0
I (336) main_task: Calling app_main()
I (336) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: DISABLED
I (346) EXAMPLE: calibration scheme version is Curve Fitting
I (346) EXAMPLE: Calibration Success
W (366) EXAMPLE: Automatic light sleep DISABLED!
I (366) EXAMPLE: ADC1 Channel[5] Raw Data: 1735
I (366) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1409 mV
I (1366) EXAMPLE: ADC1 Channel[5] Raw Data: 1736
I (1366) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1409 mV
I (2366) EXAMPLE: ADC1 Channel[5] Raw Data: 1737
I (2366) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
I (3366) EXAMPLE: ADC1 Channel[5] Raw Data: 1737
I (3366) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
I (4366) EXAMPLE: ADC1 Channel[5] Raw Data: 1738
I (4366) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
I (5366) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (5366) sleep: Code start at 0x42000020, total 114487, data start at 0x3c020020, total 47724 Bytes
0x42000020: _stext at ??:?

W (5366) EXAMPLE: Automatic light sleep ENABLED!
I (5376) EXAMPLE: ADC1 Channel[5] Raw Data: 1735
I (5376) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1409 mV
I (6386) EXAMPLE: ADC1 Channel[5] Raw Data: 1728
I (6386) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1404 mV
I (7386) EXAMPLE: ADC1 Channel[5] Raw Data: 1728
I (7386) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1404 mV
I (8386) EXAMPLE: ADC1 Channel[5] Raw Data: 1728
I (8386) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1404 mV
I (9396) EXAMPLE: ADC1 Channel[5] Raw Data: 1727
I (9396) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1402 mV
I (10396) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: DISABLED
W (10396) EXAMPLE: Automatic light sleep DISABLED!
I (10396) EXAMPLE: ADC1 Channel[5] Raw Data: 1728
I (10406) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1404 mV
I (11416) EXAMPLE: ADC1 Channel[5] Raw Data: 1735
I (11416) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1409 mV
I (12416) EXAMPLE: ADC1 Channel[5] Raw Data: 1735
I (12416) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1409 mV
I (13416) EXAMPLE: ADC1 Channel[5] Raw Data: 1737
I (13416) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
I (14416) EXAMPLE: ADC1 Channel[5] Raw Data: 1737
I (14416) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1411 mV
I (15416) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (15416) sleep: Code start at 0x42000020, total 114487, data start at 0x3c020020, total 47724 Bytes
0x42000020: _stext at ??:?

W (15416) EXAMPLE: Automatic light sleep ENABLED!
I (15426) EXAMPLE: ADC1 Channel[5] Raw Data: 1728
I (15426) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1404 mV
I (16436) EXAMPLE: ADC1 Channel[5] Raw Data: 1727
I (16436) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1402 mV
I (17436) EXAMPLE: ADC1 Channel[5] Raw Data: 1734
I (17436) EXAMPLE: ADC1 Channel[5] Cali Voltage: 1407 mV

Done
```

I'm thinking if it's related to hardware. I'm using ESP32-S3-WROOM-1 and ESP32-S3-USB-OTG board for testing. (currently no devkitC-1 in my side yet)

JVKran, 2023-11-29:
Are the design differences between the ESP32-S3-USB-OTG and ESP32-S3-DevKitC-1 so substantial that they justify a difference in behaviour of the ADC? I hope not. 

The input (a LiPo processed with a voltage divider and 100nF capacitor over it) is stable (as measured with a multimeter). I can check with an oscilloscope if there are no dips when sampling.

I don't see what the differences are between v5.1.2 and 482a8fb2d7, [because all files seem to be identical.](https://github.com/espressif/esp-idf/compare/v5.1.2...482a8fb2d7)

Could it be that I've applied patches to work with the ESP-ADF?

JVKran, 2023-11-29:
It appears that when connecting the multimeter directly to the GPIO-pin on which masurements are made, the readings are drastically affected when the multimeter is connected and light sleep is enabled while the readings aren't affected when it is disabled.

This makes me think maybe internally something is disconnected? I tried gpio_reset_pin and gpio_sleep_sel_dis before configuring the adc, but the behaviour remains unchanged. Is there anything else I could test that you know of?

JVKran, 2023-12-14:
I think [this issue](https://github.com/espressif/esp-idf/issues/12426) is related.

JVKran, 2024-07-19:
I've just repeated this test on V5.1.4 and it is still present. Is there by any change a newer chip revision?

lcolitti, 2025-01-20:
@JVKran did you ever find a workaround? I'm seeing the same thing on a custom ESP32H2 board (SOC v0.1, v5.4). Some testing indicates that "sometimes" re-initing and recalibrating the ADC helps, but not always.