espressif/esp-idf #12614: can't find a file

stellna, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

well, it maybe a dumb question, but I can't find a file?

The file path location is : C:\Users\whgPd\OneDrive\Desktop\esp-test\camera_test_5\src\coprocessor\main\image.hpp

ESP-IDF Version 5.1.1
Using CMake, Normal ESP32 board.

[image]

nopnop2002, 2023-11-18:
Files on the cloud such as OneDrive cannot be referenced from the ESP32.

stellna, 2023-11-18:
I want to create a file image.hpp in the embedded system. Since, ESP-IDF WHO examples use image.hpp in their embedded system. So, Is there anyway to create a files in the embedded system?

Thank you for your answer. I'm really appreciate it.

nopnop2002, 2023-11-18:
//const char *file_name = "/image.hpp";
const char *file_name = "/sdcard/image.hpp";

adokitkat, 2023-11-20:
Hi @stellna, I would advise you to use LittleFS (https://github.com/espressif/esp-idf/tree/master/examples/storage/littlefs) or FATFS (only if you know your device is going to be in an environment where power is stable and the device won't be shut down randomly as this can corrupt a FATFS partition https://github.com/espressif/esp-idf/tree/master/examples/storage/wear_levelling). SPIFFS is no longer maintained and has bugs in it. Personally I would use LittleFS.

adokitkat, 2023-11-23:
I am going to close this issue now, if you had any more questions feel free to reopen this or create a new issue.

closed 2023-11-23 by adokitkat