espressif/esp-idf #12615: ADC continuous read stop working after call `esp_ota_begin()`

guo-max, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2-dev-3775-gb4268c874a

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

LilyGo T-DISPLAY-S3

### Power Supply used.

USB

### What is the expected behavior?

ADC continuous read should work after call `esp_ota_begin()`. 
Calling `esp_ota_begin()` will erase the ota partition, which may take a while and block the adc reading task. But the adc reading should work after `esp_ota_begin()` finished. 

### What is the actual behavior?

ADC continuous read stop working after call `esp_ota_begin()`

### Steps to reproduce.

1. Tested with `esp-idf\examples\peripherals\adc\continuous_read\` 
2. Add a separate task that will call `esp_ota_begin()` after about 1s.
3. removed some unnecessary logs.
4. must build with an OTA partition. 
5. Modified example code:

```
/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <string.h>
#include <stdio.h>
#include "sdkconfig.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "esp_adc/adc_continuous.h"

#include <pthread.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
[... 176 lines cut ...]
            }
        }
    }

    ESP_LOGI("TASK","exit.");
    ESP_ERROR_CHECK(adc_continuous_stop(handle));
    ESP_ERROR_CHECK(adc_continuous_deinit(handle));
}

```

### Debug Logs.

```plain
ADC works fine before `ota begin`
Call `esp_ota_begin()` from another thread.
ADC stop working after `ota begin`.

![image](https://github.com/espressif/esp-idf/assets/27696259/e09c8ba5-318a-4f0a-a0e5-2d7476412fb6)
```

### More Information.

I think any task/function that may block other tasks could trigger this bug. Here I use `esp_ota_begin()` to demonstrate it. I think this bug is not related to `OTA` at all. 
I have experience that `writing large file to file system (esp_littlefs)` could trigger this bug.

p-ranav, 2023-11-20:
When `esp_ota_begin()` completes, it would be better if the ADC reader continued working OK like before? The component does not recover gracefully from the timeout error after the OTA function call has finished. Is there a technical reason why it does not recover?

soliis-jones, 2026-08-24:
Noting for others that I ran into a similar issue with the continuous ADC driver silently halting after long running higher priority tasks (reproduced via OTA) and after repeated testing it appears to have been fixed in the v6.1-rc1 branch.