espressif/esp-idf #12616: OOB Data generation on ESP32-MINI-1U and ESP32-PICO-V3 silently fails

finger563, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2-dev-3903-g66992aca7a-dirty

### Espressif SoC revision.

ESP32-U4WDH (revision v3.0)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board as well as QtPy ESP32 PICO

### Power Supply used.

USB

### What is the expected behavior?

When calling `esp_ble_create_sc_oob_data()` I expect to get the `ESP_GAP_BLE_SC_CR_LOC_OOB_EVT` event and be able to use the generated local OOB data.

### What is the actual behavior?

On ESP32 the event never fires.

After tracing through the stack, it seems that within `esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble.c`, specifically the function `UINT8 btm_proc_smp_cback(...)`, the `SMP_SC_LOC_OOB_DATA_UP_EVT` event is discarded within the branch `if (p_dev_rec != NULL)` and is only processed within the `else` condition which only occurs for an unknown device (according to the trace logs). I'm not sure why this condition check executes the `else` condition on ESP32s3 (where it does properly generate the OOB data), but on ESP32 it registers that `p_dev_rec != NULL` and therefore silently discards the `OOB_DATA_UP_EVT`. By simply adding a switch case for the `SMP_SC_LOC_OOB_DATA_UP_EVT` within the if branch, the OOB data is successfully generated.

### Steps to reproduce.

1. Call `esp_ble_create_sc_oob_data()`
2. Notice that `ESP_GAP_BLE_SC_CR_LOC_OOB_EVT` is never called

### More Information.

This is related to work I did for NFC OOB BLE pairing - see [this comment](https://github.com/finger563/esp-nfc-pairing-example/pull/1#issuecomment-1816751445).

zhp0406, 2023-11-20:
I'd like to ask whether this issue is related to the chip, and if only specific ESP32 models will encounter this problem. I haven't been able to reproduce the issue on my end. Below is the code I used to reproduce the problem; you can directly apply this patch.
[0001-feat-bt-bluedroid-Display-BLE-sc-oob-usage-for-blued.patch](https://github.com/espressif/esp-idf/files/13413677/0001-feat-bt-bluedroid-Display-BLE-sc-oob-usage-for-blued.patch)

finger563, 2023-11-20:
@zhp0406 the chips I've tested on:

- ESP32-PICO-D4 [works]
- ESP32-PICO-V3-02 (revision v3.0) [does not work]
- ESP32-U4WDH (revision v3.0) [does not work]
- ESP32-S3 (QFN56) (revision v0.1) [works]