espressif/esp-idf #12620: usb_serial_jtag_ll_write_txfifo() not working with ESP_LOGI

nikhil-robinson, 2023-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When wring a buffer through usb_serial_jtag_ll_write_txfifo() if ESP_LOGI is present it is not updating the buffer to serial.

**This works
```c
extern "C" void app_main()
{
	int cnt;
        int rxcnt;
	uint8_t rxbuf[64];
	while (true)
	{
		if (usb_serial_jtag_ll_rxfifo_data_available())
        {
            rxcnt = usb_serial_jtag_ll_read_rxfifo(rxbuf,64);
            // ESP_LOGI("TAG","Data %s size : %d\n",rxbuf,rxcnt);

			if (rxcnt > 1)
			{
				uint8_t buff[6];
				buff[0] = '\r';
				buff[1] = '\n';
				buff[2] = 'H';
				buff[3] = 'I';
				buff[4] = '\r';
				buff[5] = '\n';
				usb_serial_jtag_ll_write_txfifo(buff, sizeof(buff));
            	usb_serial_jtag_ll_txfifo_flush();
			}
			
        }
	}
	
}

```

**This Dont

```c

extern "C" void app_main()
{
	int cnt;
    int rxcnt;
	uint8_t rxbuf[64];
	while (true)
	{
		if (usb_serial_jtag_ll_rxfifo_data_available())
        {
            rxcnt = usb_serial_jtag_ll_read_rxfifo(rxbuf,64);
            ESP_LOGI("TAG","Data %s size : %d\n",rxbuf,rxcnt);

			if (rxcnt > 1)
			{
				uint8_t buff[6];
				buff[0] = '\r';
				buff[1] = '\n';
				buff[2] = 'H';
				buff[3] = 'I';
				buff[4] = '\r';
				buff[5] = '\n';
				usb_serial_jtag_ll_write_txfifo(buff, sizeof(buff));
            	usb_serial_jtag_ll_txfifo_flush();
			}
			
        }
	}
	
}

```

igrr, 2023-11-18:
Do you by any chance have CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG enabled? If yes, this would mean that you are trying to use usb_serial_jtag peripheral both via the low-level functions and at the same time via the higher-level console driver — which is a combination which is unlikely to work.

nikhil-robinson, 2023-11-18:
@igrr  nope i have not enable it.  this is my sdkconfig file.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13399862/sdkconfig.txt)

nikhil-robinson, 2023-11-18:
@igrr  For me the usb_serial_jtag_read_bytes(rxbuf,64,0); didint work it was not reaidng any input thats the reason i used the low level code with log level code everything works fine if i do esp_log_level_set("*",ESP_LOG_NONE);

igrr, 2023-11-18:
> For me the usb_serial_jtag_read_bytes(rxbuf,64,0); didint work it was not reaidng any input

Could you post the original code (with usb_serial_jtag driver, not the LL code) which didn't work? Also, please mention the IDF version (`git describe` output).

jetpax, 2023-11-18:
@nikhil-robinson  well the LL code works for me using ESP_LOGI, see below.
I am also using CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED 

However, rxcnt returned by `usb_serial_jtag_ll_read_rxfifo()` never goes above 1 ( although no chars are lost)

```
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "string.h"
#include "hal/usb_serial_jtag_ll.h"

#include "esp_log.h"

static const char *TAG = "echo";

void mytask1(void *pvParameter)
{
    uint8_t *rxbuf;
    int cnt;
    rxbuf = (uint8_t *)malloc(64);
    int rxcnt;

    while (1)
    {
        if (usb_serial_jtag_ll_rxfifo_data_available())
        {
            rxcnt = usb_serial_jtag_ll_read_rxfifo(rxbuf,64);
            ESP_LOGI(TAG, "got %d\n", rxcnt);
            vTaskDelay(pdMS_TO_TICKS(10));
            cnt = (int)usb_serial_jtag_ll_write_txfifo((const uint8_t *)rxbuf, rxcnt);
            usb_serial_jtag_ll_txfifo_flush();
            // printf("Send %d characters to host \n", cnt);
        }
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
    free(rxbuf);
    vTaskDelete(NULL);
}

void app_main(void)
{
    esp_log_level_set("*", ESP_LOG_INFO);
    ESP_LOGI(TAG, "App started!");
    xTaskCreate(mytask1, "mytask1", 1024 * 5, NULL, 1, NULL);
}

```

igrr, 2023-11-19:
> well the LL code works for me using ESP_LOGI, see below. I am also using CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED

@jetpax Sorry for not being clear. When I wrote that this is unlikely to work, I meant that you are doing something that is unsupported. Although it might end up working in some scenarios, but in general we don't guarantee it will.

Essentially you have two pieces of code in the system accessing the same underlying layer without any coordination:
- the console implementation, calling usb_serial_jtag driver, which in in turn calls usb_serial_jtag LL
- your mytask1 function, which also calls usb_serial_jtag LL.

As I wrote in https://github.com/espressif/esp-idf/issues/12605#issuecomment-1817449614 — you have to decide whether you want to control the USB_SERIAL_JTAG peripheral using the higher-level stdio functions or using the driver, and use just one of the two approaches.

Also it is highly recommended to use the usb_serial_jtag driver instead of the underlying LL functions.

nikhil-robinson, 2023-11-23:
My original usb_jtag_code.
```c

#include <stdio.h>
#include "driver/usb_serial_jtag.h"
#include "esp_vfs_usb_serial_jtag.h"
#include "esp_vfs_dev.h"
#include <fcntl.h>

static void initUart() {

    /* Disable buffering on stdin */
    setvbuf(stdin, NULL, _IONBF, 0);
    setvbuf(stdout, NULL, _IONBF, 0);

    /* Enable non-blocking mode on stdin and stdout */
    fcntl(fileno(stdout), F_SETFL, 0);
    fcntl(fileno(stdin), F_SETFL, 0);

    usb_serial_jtag_driver_config_t usb_serial_jtag_config;
    usb_serial_jtag_config.rx_buffer_size = 4096;
    usb_serial_jtag_config.tx_buffer_size = 4096;

    esp_err_t ret = ESP_OK;
    /* Install USB-SERIAL-JTAG driver for interrupt-driven reads and writes */
    ret = usb_serial_jtag_driver_install(&usb_serial_jtag_config);
    if (ret != ESP_OK) {
        return;
    }

    /* Tell vfs to use usb-serial-jtag driver */
    esp_vfs_usb_serial_jtag_use_driver();

}

static void Receive_callback(void *argument)
{

    uint8_t rxbuf[4096];
    size_t idx = 0;
    char ch;
    for (;;)
    {
        idx = usb_serial_jtag_read_bytes(rxbuf, 4096, 0);
        if(idx)
        {
            rxbuf[idx] ='\0';
            printf("RECIVED [%d] bytes: %s\n",idx, rxbuf);
        }
        vTaskDelay(1);
    }
}

void app_main(void)
{

    initUart();
    xTaskCreate(Receive_callback,"UART receive callback",4096 *2,NULL,5,NULL); // receiving commands from main uart
    
}

```
This was not working on this idf version :
idf version ESP-IDF v5.1.1-dirty

but i recently update my idf to latest release 5.1 (v5.1.2-89-g5c61c89308) and there it works

nikhil-robinson, 2023-11-23:
@igrr  regarding usb_jtag_serial i am also facing one issue with the esp32h2 module.

```
sleep_modes.c: In function 'light_sleep_uart_prepare':
/home/nikhi/idf5/esp-idf/components/esp_hw_support/sleep_modes.c:153:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  153 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13446807/sdkconfig.txt)

closed 2023-12-02 by espressif-bot