espressif/esp-idf #12621: ESP32 S2 Mini Clone crashing when using wifi

Aggeloz, 2023-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, i was researching the issue with the ESP32 S2 Mini clone boards and i found out that the wifi causing crashes only happens when im using CDC for communication, when i connect an external serial connection to see the console output then the board doesnt crash. For more info, ive tried all the fixes that the others have suggested either here on github or on the espressif forums. What could be causing this issue?

Dazza0, 2023-11-21:
Hi @Aggeloz, could you provide some more information what is happening in your application such as:

- Which ESP-IDF version/tag are you using?
- `sdkconfig` file
- Crash log and backtrace
- Any code that you can share to reproduce the crash?

Could you provide more details regarding the crash?

- Are you using the [USB OTG Console](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32s2/api-guides/usb-otg-console.html#usb-otg-console) for your serial communication (instead of UART), and when WiFi gets enabled, the USB OTG Console crashes?
  - Do you get a run-time crash with log (e.g., a failed `assert()` or `abort()`), or does the application simply continue running but the USB OTG console no longer produces any serial output?
  - Are you able to track down which part of WiFi operations is causing the crash? Does the USB OTG console crash as soon as WiFi is initialized? or does the crash happen randomly during WiFi communication?
- Are you using the internal USB PHY (e.g., GPIO 19 and 20) or have you connect an external USB PHY?

Aggeloz, 2023-11-21:
Im using ESP IDF versions 5.1.1 and 5.1.2 they behave the same.
When i use my board on a breadboard, try to connect something on the board or even get my fingers close to the board it just hangs for ever and does nothing until i reset the board or it resets itself and acts weird like booting with no working CDC output.
Ive used an external serial connection using UART to see the output messages when the board hangs but it just freezes without any output to the serial console. 
The crash happens after the wifi is initialized as soon as i get my fingers close to the board or i connect something to the pins, if the board is already on a breadboard, the whole boot sequence takes more time and some times the CDC console output freezes but the code seems to be running.
I think that the ESP32 S2 Mini board im using uses the internal USB PHY, since it has no serial connection chip onboard.

Im going to use pastebin for the biggest files if thats okay so the comment wont be too big.

[sdkconfig](https://pastebin.com/fd5BWxGy)
This is the only core dump i managed to get from the board [Core Dump](https://pastebin.com/fUk8zfyj)