espressif/esp-idf #12623: An idea to make secure binary delivery possible

zzzh, 2023-11-19:
### Is your feature request related to a problem?

In embedded system production process, it's hard to keep the firmware binary as a secret, since people have to provide it to producer to burn into chip, the danger of intellectual property leak always exists, it's safety solely depends on the producer's obey to the moral or law.

I did some reasearch on esp32 security system, and realized it's really powerful.
Though, what it can guarantee is only after the finishment of first boot, is it possbile to have a way so that developers can be always confident whenever they provide  the binary file to producers  are safe from being reverse-engineered?

### Describe the solution you'd like.

I think that's possible, just enlarge the flash encryption to binary encryption.

The key is make the binary encrypted when needed, add an option to the build system so that encryption can be done whenever needed.

At production buring time, just burn the encrypted binary to flash.

So, at running time, how to decrypt it? 
Same as flash encryption, just save the password in the chip fuse at chip production time, which can't be read by anyone.

Make sure the way to produce the password being a secret inside expressif, it can be any algorithm from inputs like chip type, id, produce date... etc, it can change every month, or every week, even every day, just make sure keep it same on chip building and firmware building process.

User don't need to know it, just being confident that their firmware binary can be safe in anytime they want.

### Additional context.

It's a small change, but very valueable, the safety of intellectual property can be guaranteed,  people don't have to have their own production team only for keeping the intellectual property safer.
If Expressif decide to do this, please make sure it's used on older esp-idf version too, thank you.

KaeLL, 2023-11-20:
Do you mean something like [this](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota)?

zzzh, 2023-11-20:
> Do you mean something like [this](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota)?

No，I mean the bootloader, since it's the base of whole firmware security and is most vulnerable in first boot.

ESP-Marius, 2023-11-21:
Sounds like you are talking about pre-encrypting binaries before the first flashing then? 

If so then this is already possible, we have a guide describing this workflow [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally)

zzzh, 2023-11-21:
> Sounds like you are talking about pre-encrypting binaries before the first flashing then?
>
> If so then this is already possible, we have a guide describing this workflow [...]

Thanks for the link, I used to read it and understand flash encryption should be done in this way, but it's not the binary encryption I was talking about.

Please think it as two roles of work, one is firmware developer, who have developed a board and a firmware to function as a device to produce, another is producer, who take order from developer, together with the board files and firmware to build some boards and burn them as the output.

In my binary encryption way, developer don't have to worry about exposing his binary to producers, while in flash encryption way, developer have to take the risk, that's a big difference.

mahavirj, 2023-11-21:
@zzzh 

Your concern regarding the security of artifacts (e.g., bootloader) to manufacturing site is correct, if one do not trust the manufacturing facility then it could be important to maintain the confidentiality of the artifacts to be delivered.

I have few questions regarding your proposal:

- How would the secret key which is generated using unique chip specific parameters be available during the build phase? 
- If there is a way to derive the secret key outside of the chip then wouldn't it be possible for the manufacturer to do so and decrypt the artifacts?

zzzh, 2023-11-21:
@mahavirj 

About your questions:

How would the secret key which is generated using unique chip specific parameters be available during the build phase?
It's not possible before Expressif accept this idea and put it in their building tools and chip production process.

If there is a way to derive the secret key outside of the chip then wouldn't it be possible for the manufacturer to do so and decrypt the artifacts?
I don't think so, in current flash encryption process, before the first boot finish, the boot loader have to be exposed to producer, if producer want, they can get public key from it and decrypt both partition table and app, no secret can be hidden in this way.

mahavirj, 2023-11-22:
@zzzh 

Sorry but I am still not quite clear on your proposal here.

For the current scenario, one way that could help to solve this problem is to divide the manufacturing workflow in 2 stages. For the first stage, which happens at the actual manufacturing site, the basic sanity of the hardware could be verified using test artifacts. If required the manufacturing site can also enable some security configuration, like flash encryption in development mode, keeping re-flashing possibility still open.  For the 2nd part, actual production artifacts can be flashed once the device is shipped back to the developer site. After locking the security configuration, the device can be finally shipped out.

Just to note that, the flash encryption (symmetric encryption scheme) key is randomly generated by the bootloader (in the default workflow) and its programmed with software read protection enabled. Please refer to the code [here](https://github.com/espressif/esp-idf/blob/a7fbf452fa181452f266b16c49063e2ff069f119/components/bootloader_support/src/flash_encryption/flash_encrypt.c#L184-L200).

Of course, there is an overhead in the actual manufacturing process here, but it helps to ensure that actual production artifacts are never shared with anyone in plaintext.

Side note: FWIW, Espressif offers [secure pre-provisioning ](https://www.espressif.com/en/support/services#manufacturing) service with customized security workflows, you may find more information by contacting our sales channel.

Hope this helps!

zzzh, 2023-11-22:
@mahavirj

Thanks for your understanding and thoughtful responding.

I realized that I did not describe the scenario I think of clear, in this scenario no ship back to developer is needed, the manufacturer would do programming, testing and deliver it to end user directly.

I understand that flash encryption (symmetric encryption scheme) key is randomly generated by the bootloader (in the default workflow) and its programmed with software read protection enabled.

But isn't it possible for a manufacturer to use a debug tool to debug the bootloader's process?
Then, they can see the flash encryption key and no secret could be hidden from them.

mahavirj, 2023-11-22:
> But, isn't it possible for a manufacturer to use a debug tool to debug the bootloader's process?

Yes, this is quite possible. In-fact, similar concern exist in your proposal too, if there is a mechanism to derive the key outside of chip. 

If the manufacturer is not trusted then it becomes difficult to ensure security in supply chain at multiple levels. For this scenario, it is best to not delegate the security configuration to mfg facility and handle it ourselves.

zzzh, 2023-11-22:
> Yes, this is quite possible. In-fact, similar concern exist in your proposal too, if there is a mechanism to derive the key outside of chip.
>
> If the manufacturer is not trusted then it becomes [...]

This is a feature request, so, will Expressif consider to adopt this binary encryption idea in the future?
Or, is it too hard to be implemented in chip building process?

zzzh, 2023-11-23:
> For the current scenario, one way that could help to solve this problem is to divide the manufacturing workflow in 2 stages. For the first stage, which happens at the actual manufacturing site, the [...]

Hi, @mahavirj 

About the usual scenario you mentioned, it have to have 2 stages to keep secret safe, and developers have to flash the chips again for actual production artifacts, which as I see suppose to be manufacturer's work.

With binary encryption, this overhead would disappear, the 'usual' workflow in the unique esp32 embedded world would become  more smoother and natural than others.

That's why I think this binary encryption idea should be adopted.

zzzh, 2023-11-24:
Anyone in charge here?

igrr, 2023-11-28:
@zzzh I think the main issue with this proposal is this part:

> Make sure this rule being a secret of Expressif, it can be any algorithm from inputs like chip type, produce date etc..
> User don't need to know the password, they just can be confident that their firmware binary are safe at anytime they want.

Essentially, there has to be a chip-specific secret which is:
- can be derived by Espressif
- can be derived by the application developer
- can _not_ be derived by the board manufacturer

If the secret can be derived from inputs such as chip type, production date and so on, then both the application developer and the board manufacturer will be able to derive the secret.

The only way this could work is if Espressif and the application developer have securely exchanged a secret. This is essentially what is already enabled by the [pre-provisioning service](https://www.espressif.com/en/support/services#manufacturing) mentioned by @mahavirj above.

zzzh, 2023-11-29:
@igrr 

Please read this part:
"Describe the solution you'd like.
...
At building time, make the binary encrypted when needed, add an option to the build system so that encryption can be done whenever needed.
At production burning time, burn the encrypted binary to flash, so it can be used when encryption needed.
As flash encryption, the password being saved in chip fuse, not readable.

Make sure the chip fuse password is same as that used in firmware encryption, by following same rule of producing password.
Make sure this rule being a secret of Expressif, it can be any algorithm from inputs like chip type, produce date etc..

User don't need to know the ruled, they just can be confident that their firmware binary are safe at anytime they want.
"

Pre-provisioning service only works for big quantity of chip order, it would cost time and money.. 
My binary encryption idea don't have these limits.

igrr, 2023-11-29:
@zzzh Yep, I read your initial post. As I mentioned in my previous reply, the problem with this idea is that if the secret can be derived at building time, the secret can also be derived by the hardware vendor.

The process of building the binary happens locally. There will have to be a program, running as part of the SDK, which would derive the key and encrypt the binary. If the program running locally on the computer can internally derive the key, it will always be possible to reverse engineer the program and extract the algorithm. This can be, at best, an obfuscation step, not a security mechanism.

zzzh, 2023-11-29:
@igrr , I see, it's not easy to hide this part since the SDK itself is open source.

What if Expressif put this part in a cloud server?  then no one can get it.

igrr, 2023-11-29:
Even this key derivation and encryption part was not open source, it would still be part of a binary distributed publicly — so the binary could be disassembled and the key derivation algorithm could be extracted.

Encrypting binaries in cloud is an option, but this would be challenging logistically. Each chip should have a unique encryption key, so the developer would need to generate a binary for every chip, and then the manufacturer would have to flash the correct binary to every chip.

Besides, the developer would have to know which chips to generate the binaries for, in advance. So the process would have to look like this:

1. Hardware manufacturer reserves a batch of chips for production.
2. The manufacturer prepares the list of chip IDs in that batch, and sends the IDs to the application developer.
3. The application developer encrypts binaries for all these chip IDs, and sends them to the manufacturer.
4. The manufacturer flashes the binaries to the correct chips.

All in all, this is not a simple process to set up, and the delay between steps 1 and 4 could pose a challenge for hardware production.

zzzh, 2023-11-29:
@igrr,

Why each chip should have a unique encryption key?  that looks unnecessary, one key for each chip type, or all using one same key looks acceptable..

I understand it's easier to be crypted for only one same key(let's say it have 256 bits in length), but the encryption algorithm behind it can be composite, so, it's still almost impossible without enormous effort for years.

As I see, no one would like to do that for a firmware.

igrr, 2023-11-29:
> Why each chip should have a unique encryption key? that looks unnecessary, one key for each chip type, or all using one same key looks acceptable.
>
> ... so, it's still almost impossible without enormous effort for years.

Key extraction from eFuses is not such an expensive process. It [can be done](https://raelize.com/blog/espressif-systems-esp32-breaking-hw-aes-with-power-analysis/) using DPA techniques. If the key is not unique for each chip, someone can extract the key used by the whole chip series and then easily sell "firmware decryption" services to the interested parties. Besides, even without this attack vector, storing this key securely becomes of huge importance. If there is a security breach on Espressif side and the unique key is leaked, that would allow decrypting firmware of every product.

So I have to disagree to you here, using the same key does not look acceptable at all.

zzzh, 2023-11-30:
I thought esp32 had overcome the Correlation Power Attacks (CPA) since  the chip version 3, sadly to see not.

As I see, they can do this probably because they knew the AES algorithm is used.
If this is true, then using an unknown composite algorithm can probably solve it.

The defeat of current flash encryption is an obvious sign of that some changes of it must happen.

I believe Expressif is able to keep a secret safe, as Coca cola did.

chipweinberger, 2023-11-30:
this is something i've thought a lot about as well.

I think it would ideally work like this.

1. every chip ships with a preprogrammed public/private key

2. the public keys can be found by serial number in an online espressif database

3. the application developer provides a unique encrypted image for each device, probably using some big hard drive. 

4. on first boot the device would re-encrypt using its aes key

It would be a lot of hassle, and as @igrr says, fuses are not fully secure anyway.

chipweinberger, 2023-12-05:
yes it would need to be hardware only re-encryption.

mahavirj, 2023-12-05:
Thanks for raising this issue and providing your thoughts, possible solutions. Any potential solution may involve changes in the hardware, ROM code, module or SoC manufacturing process and setting up additional infrastructure (cloud service route), associated cost etc. We will brainstorm on this topic more and see if we can come up with something that could benefit the community, end-users.

Given that the issue itself is not directly related to ESP-IDF, I will go ahead and close it.

closed 2023-12-05 by mahavirj