espressif/esp-idf #12625: Expose unload_partitions to allow for dynamic updates

angelnu, 2023-11-19:
### Is your feature request related to a problem?

I wrote an extension of the OTA component for esphome. There I am able to update the partition table OTA.

In order to allow the new partitions to be used without reboot I had to call esp-idf [unload_partitions](https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_partition/include/esp_private/partition_linux.h#L240). I have done this by [adding my own definition](https://github.com/esphome/esphome/blob/592e278925b9be014ba8c57700c04c9ff55254ec/esphome/components/ota/ota_backend_esp_idf.cpp#L19) as currently this function is not public.

I need to avoid the reboot since I also need to update the app partition in the same session - this allows changes in partition table sizes and migrations from different firmwares. More [info in the PR](https://github.com/esphome/esphome/pull/5535).

### Describe the solution you'd like.

Move 
https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_partition/include/esp_private/partition_linux.h#L240

to https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_partition/include/esp_partition.h

### Describe alternatives you've considered.

Rebooting after updating the partition table but this does not work when updating the app partition addresses/sizes

closed 2023-11-27 by espressif-bot

adokitkat, 2023-11-27:
Hi, it should now be available in `esp_partition.h`. The function name changed to `esp_partition_unload_all()`.

angelnu, 2023-11-27:
Many thanks @adokitkat - I will switch to use the new name once this commit gets into an official release of the ESP-IDF.

Currently I have

```
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 1, 0)
// private function in partition.c to unload partitions and free space allocated by them
extern "C" void unload_partitions(void);
#endif
```

adokitkat, 2023-11-27:
It will be available in the v5.3 version. I can try to backport it to v5.1 and v5.2 as well.

angelnu, 2023-11-27:
If you could backport @adokitkat it would be really useful!