espressif/esp-idf #12626: ESP-IDF windows build error

coderboyisongithub, 2023-11-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build my project and flash

### What is the actual behavior?

an error while building

### Steps to reproduce.

1. create any project
2. run cmd esp idf
3. run idf set-target esp32
...

### Build or installation Logs.

```plain
Determining if the include file sys/types.h exists failed with the following output:
Change Dir: F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp

Run Build Command(s):E:/Software/ESP32/tools/ninja/1.10.2/ninja.exe cmTC_1f413 && [1/2] Building C object CMakeFiles/cmTC_1f413.dir/CheckIncludeFile.c.obj
FAILED: CMakeFiles/cmTC_1f413.dir/CheckIncludeFile.c.obj 
E:\Software\ESP32\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_1f413.dir/CheckIncludeFile.c.obj -c F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: "/sys/types.h: Invalid argument
    1 | #include <sys/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining if the include file stdint.h exists failed with the following output:
Change Dir: F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp

Run Build Command(s):E:/Software/ESP32/tools/ninja/1.10.2/ninja.exe cmTC_51311 && [1/2] Building C object CMakeFiles/cmTC_51311.dir/CheckIncludeFile.c.obj
FAILED: CMakeFiles/cmTC_51311.dir/CheckIncludeFile.c.obj 
E:\Software\ESP32\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_51311.dir/CheckIncludeFile.c.obj -c F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: "/stdint.h: Invalid argument
    1 | #include <stdint.h>
      |          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining if the include file stddef.h exists failed with the following output:
Change Dir: F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp

Run Build Command(s):E:/Software/ESP32/tools/ninja/1.10.2/ninja.exe cmTC_eaac6 && [1/2] Building C object CMakeFiles/cmTC_eaac6.dir/CheckIncludeFile.c.obj
FAILED: CMakeFiles/cmTC_eaac6.dir/CheckIncludeFile.c.obj 
E:\Software\ESP32\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_eaac6.dir/CheckIncludeFile.c.obj -c F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: "/stddef.h: Invalid argument
    1 | #include <stddef.h>
      |          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining size of time_t failed with the following output:
Change Dir: F:/ESP_project/hello_world/build/CMakeFiles/CMakeTmp

Run Build Command(s):E:/Software/ESP32/tools/ninja/1.10.2/ninja.exe cmTC_2814e && [1/2] Building C object CMakeFiles/cmTC_2814e.dir/TIME_T_SIZE.c.obj
FAILED: CMakeFiles/cmTC_2814e.dir/TIME_T_SIZE.c.obj 
E:\Software\ESP32\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_2814e.dir/TIME_T_SIZE.c.obj -c F:/ESP_project/hello_world/build/CMakeFiles/CheckTypeSize/TIME_T_SIZE.c
F:/ESP_project/hello_world/build/CMakeFiles/CheckTypeSize/TIME_T_SIZE.c:24:22: error: 'time_t' undeclared here (not in a function)
   24 | #define SIZE (sizeof(time_t))
      |                      ^~~~~~
F:/ESP_project/hello_world/build/CMakeFiles/CheckTypeSize/TIME_T_SIZE.c:26:12: note: in expansion of macro 'SIZE'
   26 |   ('0' + ((SIZE / 10000)%10)),
      |            ^~~~
ninja: build stopped: subcommand failed.

F:/ESP_project/hello_world/build/CMakeFiles/CheckTypeSize/TIME_T_SIZE.c:

#undef KEY
#if defined(__i386)
# define KEY '_','_','i','3','8','6'
#elif defined(__x86_64)
# define KEY '_','_','x','8','6','_','6','4'
#elif defined(__PPC64__)
# define KEY '_','_','P','P','C','6','4','_','_'
#elif defined(__ppc64__)
# define KEY '_','_','p','p','c','6','4','_','_'
#elif defined(__PPC__)
# define KEY '_','_','P','P','C','_','_'
#elif defined(__ppc__)
# define KEY '_','_','p','p','c','_','_'
#elif defined(__aarch64__)
# define KEY '_','_','a','a','r','c','h','6','4','_','_'
#elif defined(__ARM_ARCH_7A__)
# define KEY '_','_','A','R','M','_','A','R','C','H','_','7','A','_','_'
#elif defined(__ARM_ARCH_7S__)
# define KEY '_','_','A','R','M','_','A','R','C','H','_','7','S','_','_'
#endif

#define SIZE (sizeof(time_t))
static char info_size[] =  {'I', 'N', 'F', 'O', ':', 's','i','z','e','[',
  ('0' + ((SIZE / 10000)%10)),
  ('0' + ((SIZE / 1000)%10)),
  ('0' + ((SIZE / 100)%10)),
  ('0' + ((SIZE / 10)%10)),
  ('0' +  (SIZE    % 10)),
  ']',
#ifdef KEY
  ' ','k','e','y','[', KEY, ']',
#endif
  '\0'};

#ifdef __CLASSIC_C__
int main(argc, argv) int argc; char *argv[];
#else
int main(int argc, char *argv[])
#endif
{
  int require = 0;
  require += info_size[argc];
  (void)argv;
  return require;
}
```

### More Information.

nothing

wanckl, 2023-11-20:
@coderboyisongithub  Hi, I have met similar issue resent times on windows, the Header files can't find are all in tool chain, they must be here. Root issue is almost from `RF/LF` line ending format of windows, so:        
- what you mean `command line`? windows CMD or powershell or wsl?
- what line-ending-format you have set for your `git`

Please don't mix using different terminal on windows.    

For my way, I deleted all files in `esp-idf` folder, then `git reset --hard HEAD~1` to let it Re-checkout all files, then is pass the build. because it must be some files is in different format, but we don't know which one, So we can simply refresh them all again. If not mind, you can also follow my step, then all git options are as normal

dobairoland, 2023-11-20:
How did you installed your ESP-IDF? Did you used the [Windows Installer](https://dl.espressif.com/dl/esp-idf/)?

> fatal error: "/sys/types.h:

`/` is put before the includes. It is like it would try to locate those includes in the root directory (which doesn't exist on Windows, of course). I don't think I've seen this before but maybe the the include paths are not properly set. So I would also check your first step, `create any project`. What kind of project is that? How did you make it? With `idf.py create-project` (https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/api-guides/tools/idf-py.html#start-a-new-project-create-project)? Or have you copied an example project (https://github.com/espressif/esp-idf/tree/release/v5.0/examples)? Does building any of these produce an issue?

ESP-Marius, 2023-12-20:
Closing this issue due to lack of response.

@coderboyisongithub if you are still experiencing this issue then please feel free to re-open the ticket.

closed 2023-12-20 by ESP-Marius

coderboyisongithub, 2024-04-14:
it was happening because I was using windows 11 preview beta(from insider program).It was unstable os not updating env variables while installation .I re installed the os with stable build and it works as according to documentation