espressif/esp-idf #12627: W5500 ETH - Host name doesn't get set properly

Flamabalistic, 2023-11-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom PCB

### Power Supply used.

USB

### What is the expected behavior?

- Running  `esp_netif_set_hostname` before initializing the interface, should set the hostname.

### What is the actual behavior?

- The hostname is not set.
  - Ran both nslookup `ip` and nbtstat -A `ip`, neither show up with any hostname
- The new hostname *can* however be seen when running `esp_netif_set_hostname`

### Steps to reproduce.

Short version:

- esp_netif_new(&cfg_spi)
- esp_netif_set_hostname(eth_netif_spi, hostname)
- gpio_install_isr_service(0)
- spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO)
- esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi)
- esp_eth_phy_new_w5500(&phy_config_spi)
- esp_eth_driver_install(&eth_config_spi, &eth_handle_spi)
- esp_read_mac(mac_addr, ESP_MAC_ETH)
- esp_eth_ioctl(eth_handle_spi, ETH_CMD_S_MAC_ADDR, mac_addr)
- esp_netif_attach(eth_netif_spi, esp_eth_new_netif_glue(eth_handle_spi))
- esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &EthernetHardwareEventHandler, NULL)
- esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &GetIpEventHandler, NULL)
- esp_eth_start(eth_handle_spi)

Long version:

Below is all of the relevant code, with `w5500_init` being ran on start-up
```
static const char *TAG = "W5500 eth";

#include "esp_log.h"
#include "esp_netif.h"
[... 456 lines cut ...]
D (4413) esp_netif_lwip: esp_netif_dhcpc_stop_api esp_netif:0x3fcad430
D (4423) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fcad430
D (4423) esp_netif_lwip: if0x3fcad430 start ip lost tmr: no need start because netif=0x3fcad4b4 interval=120 ip=0
D (4433) esp_netif_lwip: dhcp client stop successfully
D (4443) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (4443) esp_netif_lwip: check: remote, if=0x3fcad430 fn=0x42045b40

D (4453) esp_netif_lwip: esp_netif_set_ip_info_api esp_netif:0x3fcad430
D (4463) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3fcad4b4
D (4463) esp_netif_lwip: if0x3fcad430 ip unchanged
D (4473) esp_netif_lwip: if0x3fcad430 netif set static ip: ip changed=1
D (4473) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (4483) event: running post IP_EVENT:4 with handler 0x420486ec and context 0x3fcb2798 on loop 0x3fca30c8
D (4493) esp_eth.netif.netif_glue: eth_action_got_ip: 0x3fcb267c, 0x3c09a2a8, 4, 0x3fcb4df0, 0x3fcad430
D (4503) esp_netif_handlers: esp_netif action got_ip with netif0x3fcad430 from event_id=4
I (4513) esp_netif_handlers: eth0 ip: 192.168.0.93, mask: 255.255.254.0, gw: 192.168.1.253
D (4523) event: running post IP_EVENT:4 with handler 0x42017180 and context 0x3fcb27ec on loop 0x3fca30c8
I (4533) W5500 eth: Ethernet Got IP Address
I (4533) W5500 eth: ~~~~~~~~~~~
I (4543) W5500 eth: ETHIP:192.168.0.93
I (4543) W5500 eth: ETHMASK:255.255.254.0
I (4543) W5500 eth: ETHGW:192.168.1.253
I (4553) W5500 eth: ~~~~~~~~~~~
```
```

kostaond, 2023-11-20:
Please try to call `esp_netif_set_hostname` under `ETHERNET_EVENT_START` event.

Flamabalistic, 2023-11-21:
changed
```
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
```
to
```
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        ESP_ERROR_CHECK(esp_netif_set_hostname(eth_netif_spi, "esphost"));
        break;
```

Still got the same problem. Are there any other logs I can provide that will help diagnose the problem?

kostaond, 2023-11-21:
I see what is your problem, now. It's not that simple. You need to properly configure all the services to domain name to IP translation worked properly. Frankly speaking, I'm not expert at this domain, so I can provide you just some generic guidance. Either:

1) Use [mDNS](https://github.com/espressif/esp-protocols/tree/master/components/mdns).

2) or:
    * Configure your ESP32 device as DHCP Client and set the hostname under the Ethernet start event as I mentioned above. 
    * Configure DHCP Server on your PC. The ESP32 hostname will be exchanged between server and client during DHCP discovery process.
    * Configure local DNS server. Local DNS names can be defined by importing names from the DHCP subsystem.

Flamabalistic, 2023-11-30:
Sorry for the long wait - been busy with some other features

I can't seem to get mdns working right now, and the other options sadly won't work for my use case (it needs to work with a static ip, and without modifications to the server)

@kostaond I'm no expert in this either, so if `esp_netif_set_hostname` actually is behaving as expected, feel free to close this issue.

closed 2024-01-03 by kostaond

muratdemirtas, 2024-03-15:
with new version on v5.2.1 this issue resolved in our side, we have disabled w5500 int pin to do polling method work. After that the hostname is apperaing in modem control panel which we set. The ethernet speed is also improved x10 than old interrupt version of ethernet w5500 for OTA and TCP connections.

KaeLL, 2024-03-15:
> The ethernet speed is also improved x10 than old interrupt version of ethernet w5500 for OTA and TCP connections.

Unrelated, but interesting, especially considering the 100ms RX polling interval. Care to elaborate on this now that you mentioned it?

muratdemirtas, 2024-03-15:
> > The ethernet speed is also improved x10 than old interrupt version of ethernet w5500 for OTA and TCP connections.
> 
> Unrelated, but interesting, especially considering the 100ms RX polling interval. Care to elaborate on this now that you mentioned it?

We have multiple designs for w5500 on another projects using with STM32 based MCU's. For new project we decided to make it with ESP32 but we figured out that w5500 driver need interrupt pin to work. (SDK <5.2.1)..

As i mentioned on STM32 based projects, we used poll method to drive w5500 macraw with SOCK0 with 16 kb rx/tx buffer.
Until this time, we have no issue with w5500, mac address and hostname is everytime set correctly.

But when switched to ESP32, there was a bugs like exists( Hostname set failed, bad TCP performance, OTA takes 3 min for 1.15 mb bin file etc.). We configured w5500 spi pins, reset, irq and cs pin to drive with ESP32 gpios.

`#define CONFIG_EXAMPLE_USE_SPI_ETHERNET 1
#define CONFIG_EXAMPLE_SPI_ETHERNETS_NUM 1
#define CONFIG_EXAMPLE_USE_W5500 1
#define CONFIG_EXAMPLE_ETH_SPI_HOST 1
#define CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO 18
#define CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO 23
#define CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO 19
#define CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ 20
#define CONFIG_EXAMPLE_ETH_SPI_CS0_GPIO 5
#define CONFIG_EXAMPLE_ETH_SPI_INT0_GPIO 36
#define CONFIG_EXAMPLE_ETH_SPI_PHY_RST0_GPIO 25
#define CONFIG_EXAMPLE_ETH_SPI_PHY_ADDR0 1
`

SCK frequency tested on every frequency ( 20,10,5 Mhz) but TCP performance too bad on ethernet side. I can say that the OTA with ethernet takes 3 minute for 1.2 mb bin file but wifi takes only 30 seconds to perform OTA. I was know there is anything wrong with our side. 

 I have waited to poll method support for next SDK's, Today i have upgraded my SDK to v5.2.1 and configured irq pin to -1 and poll interval to 100 ms and ta ta...

OTA and TCP performance improved x10. OTA takes 28 second on ethernet which is perfect. TCP uploads for 1.2 mb takes 12 seconds( without SDK upgrade this was around 1.5 minute).

Hostname is not stable appeared the modem control panel when older SDK. But now is stable.
 

I want to share that who has that issue.

kostaond, 2024-03-19:
The symptoms sound suspicious. Are you sure the interrupt line was properly connected and configured when not using poll mode? The thing is, it could have work even with incorrectly configured INT line because there used to be 1 sec timeout when no INT hit. Therefore, it could explain your poor performance.

```c
    while (1) {
        /* check if the task receives any notification */
        if (ulTaskNotifyTake(pdTRUE, pdMS_TO_TICKS(1000)) == 0 &&    // if no notification ...
            gpio_get_level(emac->int_gpio_num) != 0) {               // ...and no interrupt asserted
            continue;                                                // -> just continue to check again
        }
```

kostaond, 2024-03-19:
> The ethernet speed is also improved x10 than old interrupt version 

It improved 10 times, 100 ms is 10 times shorter interval than `ulTaskNotifyTake` 1 sec timeout...

muratdemirtas, 2024-03-19:
> > The ethernet speed is also improved x10 than old interrupt version
> 
> It improved 10 times, 100 ms is 10 times shorter interval than `ulTaskNotifyTake` 1 sec timeout...

you are right, we found it, the int pin is our old daughter boards not connected with ESP32 in our side. thank you. we fixed.