espressif/esp-idf #12629: JTAG debugger wirking not correctly

maxkondor, 2023-11-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to use a JTAG debugger based on FTDI chip. IDF version is 5.1.0. When I'm using an `idf.py gdb` and `idf.py openocd` all work fine, breakpoint sets to my func etc. But when I'm trying to use my IDE with your toolchain (xtensa-esp32-elf-gdb, your version of openocd), It's stucks on `core exeption was unhandled, breaks in guru meditation error`. I was trying to use gdbinit commands from your programming guide, use generated `gdbinit` from `build` directory, but it always stucks on the unhandled exception... Please, give me working arguments to run It successfully like on my IDE. How I must start debug server and how I must setup your gdb ? I want the same result as `idf.py gdb` and `idf.py openocd`.

igrr, 2023-11-20:
@maxkondor Which IDE are you using?
- For VS Code, please report issues in https://github.com/espressif/vscode-esp-idf-extension/
- For Eclipse, please use https://github.com/espressif/idf-eclipse-plugin

Both repositories also contain guides on how to set up debugging.

(I'm going to close this issue since `idf.py ...` part — maintained in this repo — seems to work fine.)

closed 2023-11-20 by igrr

maxkondor, 2023-11-20:
I'm using Clion and don't want to use esp-idf extension for VS code. All I want, give me the trace info of idf.py openocd and gdb, I want to read parameters from your script. And when I'm run it with --verbose, It's only telling me 'Executing gdb target'. Make pls It visible, like: `running xtensa-esp-elf-gdb -x /path/to/gdbinit --abc --def`...