espressif/esp-idf #12633: Possible to put any logic in sdkconfig.defaults?

eme-bennytang, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I got these different settings in SDK config which I may want them differently in release/debug config.

Currently I am just commenting/uncommenting them manually everytime I want to switch over.

Is it possible that I can put some logic inside so that I can just flip the main switch instead of changing everytime?
e.g. 
IF RELEASE:
CONFIG_LOG_DEFAULT_LEVEL_NONE=y
ELSE:
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
...etc

Or maybe there is a better way to do this?

Thanks in advance

ESP-Marius, 2023-11-21:
Hi,

It is not possible to add any logic to the sdkconfig, and I think we want to keep them simple that way.

What you are looking for sounds like it could be achieved by having multiple sdkconfigs though, we have an example of how to do this [here](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config)

We also have [idf-build-apps](https://github.com/espressif/idf-build-apps) to help you manage building of apps with multiple configs if this is something you need.

eme-bennytang, 2023-11-21:
Understood, I will try with the different building sdk config one  

Thanks for the help

closed 2023-11-21 by eme-bennytang